ompl/ompl #828: What does MechanicalWorkOptimizationObjective do?

werner291, 2021-08-23:
Hello,

I'm currently looking into whether TRRT and BiTRRT could prove useful in the problem I'm working on, but I'm confused about how the `MechanicalWorkOptimizationObjective` is defined.

Namely: https://github.com/ompl/ompl/blob/main/src/ompl/base/objectives/src/MechanicalWorkOptimizationObjective.cpp#L59

which, since state cost is always 1 (https://github.com/ompl/ompl/blob/main/src/ompl/base/objectives/src/MechanicalWorkOptimizationObjective.cpp#L51)

appears to essentially just collapse into optimizing for path length?

Am I misunderstanding this?

werner291, 2021-08-23:
Never mind, I think my confusion stems from how these objectives are used in MoveIt; this implementation makes total sense if the objective gets subclassed.

closed 2021-08-23 by werner291

MarcM0, 2023-01-27:
Wait so is it actually any different? Does it minimize joint movement?