ompl/ompl #829: collect2: error: ld returned 1 exit status

emrecemal, 2021-08-27:
When I try to compile Constrained Planning demos

`g++ -I/ompl-1.5.2/src -std=c++14 -I/usr/local/include/ompl-1.5 -I/usr/include/eigen3 ompl-1.5.2/demos/constraint/ConstrainedPlanningSphere.cpp -o Sphere.out -L/ompl-1.5.2/build/Release/lib -lompl`

I get the following error:

/usr/bin/ld: /tmp/cc5pLNaM.o: in function `addSpaceOption(boost::program_options::options_description&, SPACE_TYPE*)':
ConstrainedPlanningSphere.cpp:(.text+0x174): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x1a4): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: /tmp/cc5pLNaM.o: in function `addPlannerOption(boost::program_options::options_description&, std::vector<PLANNER_TYPE, std::allocator<PLANNER_TYPE> >*)':
ConstrainedPlanningSphere.cpp:(.text+0x490): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x4c8): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: /tmp/cc5pLNaM.o: in function `addConstrainedOptions(boost::program_options::options_description&, ConstrainedOptions*)':
ConstrainedPlanningSphere.cpp:(.text+0x550): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x58f): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x59b): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x5de): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x5ea): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x62d): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x639): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x68c): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x698): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x6db): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x6e7): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x736): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: /tmp/cc5pLNaM.o: in function `addAtlasOptions(boost::program_options::options_description&, AtlasOptions*)':
ConstrainedPlanningSphere.cpp:(.text+0x7c9): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x808): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x814): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x867): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x873): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x8b6): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x8c2): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x905): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x911): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x925): undefined reference to `boost::program_options::bool_switch(bool*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x95b): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x967): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x97b): undefined reference to `boost::program_options::bool_switch(bool*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x9b1): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x9bd): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0xa00): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: /tmp/cc5pLNaM.o: in function `main':
ConstrainedPlanningSphere.cpp:(.text+0x186e): undefined reference to `boost::program_options::options_description::m_default_line_length'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x1879): undefined reference to `boost::program_options::options_description::m_default_line_length'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x18c0): undefined reference to `boost::program_options::options_description::options_description(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, unsigned int)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x18ed): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x1911): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, char const*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x1920): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x193d): undefined reference to `boost::program_options::bool_switch(bool*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x1975): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x1984): undefined reference to `boost::program_options::options_description::add_options()'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x19a1): undefined reference to `boost::program_options::bool_switch(bool*)'
/usr/bin/ld: ConstrainedPlanningSphere.cpp:(.text+0x19d9): undefined reference to `boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)'
[... cut ...] t'
/usr/bin/ld: /tmp/cc5pLNaM.o:(.data.rel.ro._ZTVN5boost16exception_detail19error_info_injectorINS_15program_options20invalid_option_valueEEE[_ZTVN5boost16exception_detail19error_info_injectorINS_15program_options20invalid_option_valueEEE]+0x30): undefined reference to `boost::program_options::error_with_option_name::substitute_placeholders(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
collect2: error: ld returned 1 exit status

What should I do?

GeorgiaKp, 2021-11-04:
You need to link boost program_options library, I solved the same problem by adding the flag -lboost_program_options to my command: 
g++ -I/usr/local/include/ompl-1.5 ConstrainedPlanningSphere.cpp -L/usr/local/lib -lompl -lboost_program_options

zkingston, 2021-11-05:
If you are using g++ directly to compile OMPL scripts, I would recommend using pkg-config to get the appropriate flags and libraries, e.g., `pkg-config --cflags --libs ompl`.

closed 2021-11-05 by zkingston