ompl/ompl #835: Error in omplConfig.cmake.in checking for opengl libraries

mahaarbo, 2021-09-30:
### Reproduction steps
1. Run `./install-ompl-ubuntu.sh --app`
2. `find_package(ompl REQUIRED)` in a CMake project
3. Get an error message

### Error message
```
CMake Error at /usr/local/share/ompl/cmake/omplConfig.cmake:69 (if):
  if given arguments:

    "/usr/lib/x86_64-linux-gnu/libOpenGL.so" "/usr/lib/x86_64-linux-gnu/libGLX.so" "/usr/lib/x86_64-linux-gnu/libGLU.so"

```

### Cause
https://github.com/ompl/ompl/blob/5477faccbb9dfafc7b97f4a6fad487f21f77a4c3/omplConfig.cmake.in#L55
Should this be a `foreach` with `EXISTS` checks?

mahaarbo, 2021-10-01:
Yup, that seems to work.

yurirocha15, 2021-12-20:
This issue is still present when installing the omplapp from source, as it references [this ompl commit](https://github.com/ompl/ompl/tree/f255ddd60c0ed278b7d7df2b3e37de5e872bac23), which is prior to [this fix](https://github.com/ompl/ompl/commit/d101202ebcb3fe9c7882c286615f00887e73e775). Would be good to have the submodule updated to include this fix there as well.

trebeggiani, 2022-02-04:
Please do update the commit on the source install. This has cost me many hours of debugging until I found this thread. Thanks.

PR #838 (2021-10-06): fix opengl_libraries check
Resolves #835.
Fix by @mamoll, see https://github.com/ompl/ompl/issues/835#issuecomment-931475324