ompl/ompl #836: declaration_not_found_t for ompl::base::ProjectionEvaluator and vector< std::string >

jonathanglima, 2021-10-04:
Hey, archlinux user here trying to build ompl with python bindings. While running `ninja update_bindings` I get the following errors.

I'm running python 3.9.7 with pyenv and virtualenvwrapper. Any suggestions?

I've installed the dependencies that show up in cmake and enabled `OMPL_BUILD_PYBINDINGS=On`.

```Traceback (most recent call last):
  File "/mnt/extra_disk/home/jonathan/.cache/yay/ompl/src/ompl-1.5.2/py-bindings/generate_bindings.py", line 950, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/mnt/extra_disk/home/jonathan/.cache/yay/ompl/src/ompl-1.5.2/py-bindings/generate_bindings.py", line 878, in __init__
    code_generator_t.__init__(self, 'util', None, replacement, 1)
  File "/mnt/extra_disk/home/jonathan/.cache/yay/ompl/src/ompl-1.5.2/py-bindings/ompl/bindings_generator.py", line 146, in __init__
    self.filter_declarations()
  File "/mnt/extra_disk/home/jonathan/.cache/yay/ompl/src/ompl-1.5.2/py-bindings/generate_bindings.py", line 893, in filter_declarations
    self.std_ns.class_('vector< std::string >').include()
  File "/home/jonathan/.virtualenvs/msc_simulation_env/lib/python3.9/site-packages/pygccxml/declarations/scopedef.py", line 546, in class_
    self._find_single(
  File "/home/jonathan/.virtualenvs/msc_simulation_env/lib/python3.9/site-packages/pygccxml/declarations/scopedef.py", line 464, in _find_single
    found = matcher.get_single(decl_matcher, decls, False)
  File "/home/jonathan/.virtualenvs/msc_simulation_env/lib/python3.9/site-packages/pygccxml/declarations/scopedef.py", line 90, in get_single
    raise runtime_errors.declaration_not_found_t(decl_matcher)
pygccxml.declarations.runtime_errors.declaration_not_found_t: Unable to find declaration. Matcher: [(decl type==class_t) and (name==vector< std::string >)]
```

```Traceback (most recent call last):
  File "/mnt/extra_disk/home/jonathan/.cache/yay/ompl/src/ompl-1.5.2/py-bindings/generate_bindings.py", line 950, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/mnt/extra_disk/home/jonathan/.cache/yay/ompl/src/ompl-1.5.2/py-bindings/generate_bindings.py", line 154, in __init__
    code_generator_t.__init__(self, 'base', ['bindings/util'], replacement)
  File "/mnt/extra_disk/home/jonathan/.cache/yay/ompl/src/ompl-1.5.2/py-bindings/ompl/bindings_generator.py", line 146, in __init__
    self.filter_declarations()
  File "/mnt/extra_disk/home/jonathan/.cache/yay/ompl/src/ompl-1.5.2/py-bindings/generate_bindings.py", line 163, in filter_declarations
    self.std_ns.class_(
  File "/home/jonathan/.virtualenvs/msc_simulation_env/lib/python3.9/site-packages/pygccxml/declarations/scopedef.py", line 546, in class_
    self._find_single(
  File "/home/jonathan/.virtualenvs/msc_simulation_env/lib/python3.9/site-packages/pygccxml/declarations/scopedef.py", line 464, in _find_single
    found = matcher.get_single(decl_matcher, decls, False)
  File "/home/jonathan/.virtualenvs/msc_simulation_env/lib/python3.9/site-packages/pygccxml/declarations/scopedef.py", line 90, in get_single
    raise runtime_errors.declaration_not_found_t(decl_matcher)
pygccxml.declarations.runtime_errors.declaration_not_found_t: Unable to find declaration. Matcher: [(decl type==class_t) and (name==map< std::string, std::shared_ptr< ompl::base::ProjectionEvaluator > >)]```

mamoll, 2021-10-04:
Can you check if [this PR](https://github.com/ompl/ompl/pull/837) fixes it?

jonathanglima, 2021-10-04:
It does not. Error went from generate_bindings.py, line 893 to generate_bindings.py", line 894. Basically the same thing but the declaration matcher changed:

`pygccxml.declarations.runtime_errors.declaration_not_found_t: Unable to find declaration. Matcher: [((decl type==class_t) and (name==vector< std::basic_string< char, std::char_traits< char >, std::allocator< char > > >)]` (sorry if I had typos, just wrote this by hand)

Before this PR I also tried to build using python3.7 from pyenv and the system's python @ 3.9.7. Same error as before.
I tried changing -march as well (suggestion from archlinux AUR). Znver2 (current), native and generic gave me the same results before the PR. I can test them in this PR as well if you'd like to.

Does it have something to do with pyplusplus/pygccxml? Any suggestions on code changes or environment changes/tests I could make to a better debugging? Maybe some package versions?

jonathanglima, 2021-10-06:
Just logged the installed versions. Maybe it's boost or eigen's fault?

```boost 1.76
gcc 9.4 (had 11 before, tried a downgrade and it didn't helped)
python 3.9.7
pyplusplus 1.8.4
cmake 3.21.3
eigen 3.4.0
flann 1.9.1
libspot 2.9.7
doxygen 1.9.2
pygccxml==2.2.1```

JianLi0-0, 2021-12-16:
I met the same problem. Is it solved?

jonathanglima, 2021-12-16:
Sorry for the delay. I'm using Manjaro.

I actually went with another approach and did not fix/retried this issue

UtkarshMishra04, 2022-10-09:
Hi is this problem solved for Ubuntu 20.04 and pygccxml version 2.2.1?

xinyazhang, 2022-11-01:
A temporary fix:
Replace `std::string` with `std::basic_string<char>` presented in `ompl_util_generator_t.filter_declarations`

zkingston, 2023-01-23:
Just checked Python binding build on archlinux, with #978 they should work now.

zkingston, 2023-02-06:
Closing - please follow up if this issue persists.

closed 2023-02-06 by zkingston

zkingston, 2023-02-10:
See #980 for a potential fix.

sleichu2, 2023-02-11:
Another error has occurred when build geometric. Ubuntu 20.04.

<pre>Traceback (most recent call last):
  File &quot;/home/rikka/Project/ompl/ompl-1.6.0/py-bindings/generate_bindings.py&quot;, line 966, in &lt;module&gt;
    globals()[&apos;ompl_&apos;+module+&apos;_generator_t&apos;]()
  File &quot;/home/rikka/Project/ompl/ompl-1.6.0/py-bindings/generate_bindings.py&quot;, line 625, in __init__
    replacement)
  File &quot;/home/rikka/Project/ompl/ompl-1.6.0/py-bindings/ompl/bindings_generator.py&quot;, line 156, in __init__
    self.mb.split_module(&apos;bindings/&apos; + name, use_files_sum_repository=True)
  File &quot;/usr/local/lib/python3.8/dist-packages/pyplusplus/module_builder/boost_python_builder.py&quot;, line 389, in split_module
    , encoding=self.encoding)
  File &quot;/usr/local/lib/python3.8/dist-packages/pyplusplus/file_writers/__init__.py&quot;, line 37, in write_multiple_files
    mfs.write()
  File &quot;/usr/local/lib/python3.8/dist-packages/pyplusplus/file_writers/multiple_files.py&quot;, line 395, in write
    self.split_values_traits()
  File &quot;/usr/local/lib/python3.8/dist-packages/pyplusplus/file_writers/multiple_files.py&quot;, line 321, in split_values_traits
    for trait in self.__value_traits: self.split_value_traits(trait)
  File &quot;/usr/local/lib/python3.8/dist-packages/pyplusplus/file_writers/multiple_files.py&quot;, line 317, in split_value_traits
    , value_traits.create() ) )
  File &quot;/usr/local/lib/python3.8/dist-packages/pyplusplus/file_writers/multiple_files.py&quot;, line 56, in write_file
    raise RuntimeError( os.linesep.join(msg) )
RuntimeError: `Py++` is going to write different content to the same file(bindings/geometric/_Edge__value_traits.pypp.hpp).
</pre>

zkingston, 2023-02-11:
Can you try to clean your build directory and rebuild? I was able to compile in a clean environment on Ubuntu 20.04.

zkingston, 2023-02-11:
For an operational fix, try removing the line `src/ompl/geometric/planners/informedtrees/EITstar.h` from `py-bindings/headers_geometric.txt`. It seems there is an unintended naming conflict between EIT* and the other informed planners.

zkingston, 2023-02-11:
See #979 for a potential fix.

sleichu2, 2023-02-12:
Worked!Thank you very much!

lydonchandra, 2023-07-19:
This breaks on main at the moment, when reverting main back to where #980 is committed, works OK

yufeiwang63, 2023-08-23:
> This breaks on main at the moment, when reverting main back to where #980 is committed, works OK

Still facing the issue with ubuntu 18.04, python3, and the command to install is `./install-ompl-ubuntu.sh --python --github`. 
One thing to notice during the installation process:
`${SUDO} pip${PYTHONV} install -vU https://github.com/CastXML/pygccxml/archive/develop.zip` this command fails and I replaced it with `pip3 install pygccxml`. 

@lydonchandra Can you elaborate a bit more how did you make it work? Thanks!

johnnynunez, 2025-02-01:
Error in sitecustomize; set PYTHONVERBOSE for traceback:
    AssertionError:
    /tmp/pip-build-env-lvw74kuq/overlay/lib/python3.10/site-packages/pyplusplus/binary_parsers/parsers.py:7: DeprecationWarning: invalid escape sequence 'm'
      """
    Traceback (most recent call last):
      File "/project/py-bindings/ompl/bindings_generator.py", line 145, in __init__
        self.std_ns = self.mb.namespace('std')
                      ^^^^^^^^^^^^^^^^^^^^^^^^
      File "/tmp/pip-build-env-lvw74kuq/overlay/lib/python3.10/site-packages/pyplusplus/module_builder/module_builder.py", line 265, in namespace
        return self.global_ns.namespace( name=name
      File "/tmp/pip-build-env-lvw74kuq/overlay/lib/python3.10/site-packages/pygccxml/declarations/namespace.py", line 117, in namespace
        self._find_single(
      File "/tmp/pip-build-env-lvw74kuq/overlay/lib/python3.10/site-packages/pygccxml/declarations/scopedef.py", line 464, in _find_single
        found = matcher.get_single(decl_matcher, decls, False)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "/tmp/pip-build-env-lvw74kuq/overlay/lib/python3.10/site-packages/pygccxml/declarations/scopedef.py", line 92, in get_single
        raise runtime_errors.multiple_declarations_found_t(decl_matcher)
    pygccxml.declarations.runtime_errors.multiple_declarations_found_t: Multiple declarations have been found. Matcher: [(decl type==namespace_t) and (name==std)]
    During handling of the above exception, another exception occurred:
    Traceback (most recent call last):
      File "/project/py-bindings/generate_bindings.py", line 1042, in <module>
        globals()['ompl_'+module+'_generator_t']()
      File "/project/py-bindings/generate_bindings.py", line 1002, in __init__
        code_generator_t.__init__(self, 'util', None, replacement, 1)
      File "/project/py-bindings/ompl/bindings_generator.py", line 147, in __init__
        std_decls = self.mb.global_ns.declarations('std')
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    TypeError: 'list' object is not callable
    -- Build type: Release
    warning: target 'Boost::system' is not absolute...
    warning: target 'Boost::system' does not exist...
    -- Checking for module 'libspot'
    --   Package 'libspot', required by 'virtual:world', not found
    -- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS)
    -- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
    -- Code for module util not found; type "make update_bindings"
    -- Code for module base not found; type "make update_bindings"
    -- Code for module control not found; type "make update_bindings"
    -- Code for module geometric not found; type "make update_bindings"