ompl/ompl #846: python: No dubins() function or DubinsPath class in DubinsStateSpace

rallen10, 2021-11-19:
I'm using the python bindings for OMPL and having problems accessing member functions and classes of [DubinsStateSpace](https://ompl.kavrakilab.org/classompl_1_1base_1_1DubinsStateSpace.html), specifically `dubins` function which returns a `DubinsPath` object. 

I am using python bindings generated from `ompl-1.5.2`. 

To reproduce:

```
from ompl import base as ob
ds = ob.DubinsStateSpace()
ds.isMetricSpace()    # this works and returns False, so other member functions are working
dir(ds)   # dubins and DubinsPath do not appear in this output
ds.dubins

AttributeError: 'DubinsStateSpace' object has no attribute 'dubins'
```

rallen10, 2021-11-29:
Any thoughts on how to even begin debugging this? I feel like I must be missing something obvious, but not even sure where to start looking since the python binding generation is a black box to me.

rallen10, 2021-11-30:
I tried rebuilding ompl from a fresh clone of main (instead of using version 1.5.2), but ran into the same issue.

rallen10, 2021-12-03:
I just tried building the python bindings from release 1.5.0 to see if it was a recently introduced problem. Unfortunately ran into the same issue.

rallen10, 2021-12-03:
@zkingston do you happen to have any thoughts on this issue? I'm just trying to bring some attention to this issue because I feel like it might be a simple fix that I'm just ignorant of.

nykabhishek, 2022-03-12:
Please check out this thread. Hope it helps :)

https://github.com/ompl/ompl/issues/240

zkingston, 2023-02-06:
The `DubinsPath` type and related classes are specifically excluded from being exposed in the Python bindings, due to issues with exposing arrays of enum types in the automatic generation process (see https://github.com/ompl/ompl/blob/main/py-bindings/generate_bindings.py#L253). As indicated in #240, using the interpolate function to retrieve points along this path is exposed in the bindings.

closed 2023-02-06 by zkingston