ompl/ompl #848: OMPL compilation from source (with SPOT) does not find C++17

marcos-pereira, 2021-11-24:
Dear OMPL developers,

Would you suggest any solution to the problem below?
I am trying to install OMPL with SPOT (spot-2.10.1) previously installed and it throws the following error because the current `__cplusplus` variable is not set to C++17:
```
In file included from /usr/local/include/spot/tl/formula.hh:46,
                 from /usr/local/include/spot/tl/parse.hh:25,
                 from /home/marcos/git/ompl/src/ompl/control/planners/ltl/src/Automaton.cpp:40:
/usr/local/include/spot/misc/common.hh:39:6: error: #error C++17 compiler required
   39 |     #error C++17 compiler required
      |      ^~~~~
make[2]: *** [src/ompl/CMakeFiles/ompl.dir/build.make:934: src/ompl/CMakeFiles/ompl.dir/control/planners/ltl/src/Automaton.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:368: src/ompl/CMakeFiles/ompl.dir/all] Error 2
make: *** [Makefile:163: all] Error 2

```
I tried installing both using the [installation script from source](https://ompl.kavrakilab.org/installation.html) and also by cloning the OMPL git repository: 
```
git clone https://github.com/ompl/ompl.git
mkdir -p build/Release
cd build/Release
cmake ../..
make -j 2
sudo make install
```
In both cases, it complains about the C++ compiler when running `make`.
I have tried setting some CMake variables, but the error continued:
```
cmake ../.. -DCMAKE_CXX_STANDARD=17 -DCMAKE_CXX_FLAGS=-std=c++17
# also tried this variation with ""
cmake ../.. -DCMAKE_CXX_STANDARD=17 -DCMAKE_CXX_FLAGS="-std=c++17"
```
I also checked that by running the following program from [this thread](https://stackoverflow.com/questions/2324658/how-to-determine-the-version-of-the-c-standard-used-by-the-compiler), I get C++14 as output:
```
#include<iostream>

int main() {
    if (__cplusplus == 201703L) std::cout << "C++17\n";
    else if (__cplusplus == 201402L) std::cout << "C++14\n";
    else if (__cplusplus == 201103L) std::cout << "C++11\n";
    else if (__cplusplus == 199711L) std::cout << "C++98\n";
    else std::cout << "pre-standard C++\n";
}
```
If I compile it with `-std=c++17`, then I get C++17.

I am currently using Ubuntu 20.04: `lsb_release -a`
```
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal
```
My g++ version is 9.3.0 and my gcc version is 9.3.0.

Last, my CMakeCache.txt is the following:
```
# This is the CMakeCache file.
# For build in directory: /home/marcos/git/ompl/build/Release
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

[... 800 lines cut ...]
__pkg_config_arguments_SPOT:INTERNAL=libspot
__pkg_config_arguments_YAMLCPP:INTERNAL=yaml-cpp
__pkg_config_checked_FLANN:INTERNAL=1
__pkg_config_checked_ODE:INTERNAL=1
__pkg_config_checked_SPOT:INTERNAL=1
__pkg_config_checked_YAMLCPP:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_FLANN_flann
pkgcfg_lib_FLANN_flann-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_FLANN_flann_cpp
pkgcfg_lib_FLANN_flann_cpp-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_FLANN_lz4
pkgcfg_lib_FLANN_lz4-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_ODE_ode
pkgcfg_lib_ODE_ode-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_SPOT_bddx
pkgcfg_lib_SPOT_bddx-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_SPOT_spot
pkgcfg_lib_SPOT_spot-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_YAMLCPP_yaml-cpp
pkgcfg_lib_YAMLCPP_yaml-cpp-ADVANCED:INTERNAL=1
prefix_result:INTERNAL=/usr//usr/lib/x86_64-linux-gnu

```

Would you have any idea on how to install OMPL with SPOT considering the problem described above?

mamoll, 2021-11-24:
If you run `make VERBOSE=1` you'll see exactly how the compiler is invoked. Can you verify that `--std=c++17` is used when compiling Automaton.cpp?

marcos-pereira, 2021-11-24:
When I run `make VERBOSE=1` it results in
```
/usr/bin/cmake -S/home/marcos/git/ompl -B/home/marcos/git/ompl/build/Release --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/marcos/git/ompl/build/Release/CMakeFiles /home/marcos/git/ompl/build/Release/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/marcos/git/ompl/build/Release'
make -f src/ompl/CMakeFiles/ompl.dir/build.make src/ompl/CMakeFiles/ompl.dir/depend
make[2]: Entering directory '/home/marcos/git/ompl/build/Release'
cd /home/marcos/git/ompl/build/Release && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/marcos/git/ompl /home/marcos/git/ompl/src/ompl /home/marcos/git/ompl/build/Release /home/marcos/git/ompl/build/Release/src/ompl /home/marcos/git/ompl/build/Release/src/ompl/CMakeFiles/ompl.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/marcos/git/ompl/build/Release'
make -f src/ompl/CMakeFiles/ompl.dir/build.make src/ompl/CMakeFiles/ompl.dir/build
make[2]: Entering directory '/home/marcos/git/ompl/build/Release'
[  0%] Building CXX object src/ompl/CMakeFiles/ompl.dir/control/planners/ltl/src/Automaton.cpp.o
cd /home/marcos/git/ompl/build/Release/src/ompl && /usr/bin/c++  -DBOOST_MATH_NO_LONG_DOUBLE_MATH_FUNCTIONS -DBOOST_TEST_DYN_LINK -Dompl_EXPORTS -I/home/marcos/git/ompl/src -I/home/marcos/git/ompl/build/Release/src -I/usr/include/eigen3 -I/usr/lib/python3/dist-packages/numpy/core/include  -std=c++17 -O3 -O3 -DNDEBUG -fPIC   -W -Wall -Wextra -Wno-system-headers -fPIC -std=c++14 -o CMakeFiles/ompl.dir/control/planners/ltl/src/Automaton.cpp.o -c /home/marcos/git/ompl/src/ompl/control/planners/ltl/src/Automaton.cpp
In file included from /usr/local/include/spot/tl/formula.hh:46,
                 from /usr/local/include/spot/tl/parse.hh:25,
                 from /home/marcos/git/ompl/src/ompl/control/planners/ltl/src/Automaton.cpp:40:
/usr/local/include/spot/misc/common.hh:39:6: error: #error C++17 compiler required
   39 |     #error C++17 compiler required
      |      ^~~~~
make[2]: *** [src/ompl/CMakeFiles/ompl.dir/build.make:934: src/ompl/CMakeFiles/ompl.dir/control/planners/ltl/src/Automaton.cpp.o] Error 1

```
It seems that there is `-std=c++17` but then there is also a `-std=c++14`.

mamoll, 2021-11-24:
Can you try making this change in the ompl source directory and see if that fixes things?:
```diff
diff --git a/CMakeModules/CompilerSettings.cmake b/CMakeModules/CompilerSettings.cmake
index acb5ac55..1c397050 100644
--- a/CMakeModules/CompilerSettings.cmake
+++ b/CMakeModules/CompilerSettings.cmake
@@ -1,4 +1,4 @@
-set(CMAKE_CXX_STANDARD 14)
+set(CMAKE_CXX_STANDARD 17)
 set(CMAKE_CXX_STANDARD_REQUIRED ON)
 set(CMAKE_CXX_EXTENSIONS OFF)
```

marcos-pereira, 2021-11-24:
It seems to have worked. The `make` executed without errors and also the `sudo make install`. 
I will still test the library includes and so on, but for now the issue is solved.

Maybe you could add some lines in the CompilerSettings.cmake to detect if SPOT is installed. If yes, then set C++17.

Thank you!

Best regards,
Marcos