ompl/ompl #851: I am installing ompl on my jupyter notebooks (anaconda) via conda install -c conda-forge ompl command but when i import ompl it is showing module not found

muk465, 2021-12-04:
how to solve this problem?

mamoll, 2021-12-05:
This sounds like an issue with conda, not ompl. Try to figure out where conda installed the ompl-related files and make sure the relevant directory is in your python path. It may be as simple as just running the correct python interpreter (in case you have multiple python versions installed).

closed 2021-12-05 by mamoll

ntorresalberto, 2021-12-07:
Could it be possible that python bindings are not packaged in conda?
I tried this:

```
conda create --name myenv
conda activate myenv
conda install ompl -c conda-forge
```

And I could not find the bindings in the installed files:
```
$ find /home/user/miniconda3/envs/myenv | grep ompl | grep '\.py'
/home/user/miniconda3/envs/myenv/lib/python3.10/site-packages/pip/_internal/cli/autocompletion.py
/home/user/miniconda3/envs/myenv/lib/python3.10/site-packages/pip/_internal/cli/__pycache__/autocompletion.cpython-310.pyc
/home/user/miniconda3/envs/myenv/lib/python3.10/site-packages/pip/_internal/commands/completion.py
/home/user/miniconda3/envs/myenv/lib/python3.10/site-packages/pip/_internal/commands/__pycache__/completion.cpython-310.pyc
/home/user/miniconda3/envs/myenv/lib/python3.10/idlelib/autocomplete_w.py
/home/user/miniconda3/envs/myenv/lib/python3.10/idlelib/idle_test/test_autocomplete.py
/home/user/miniconda3/envs/myenv/lib/python3.10/idlelib/idle_test/test_autocomplete_w.py
/home/user/miniconda3/envs/myenv/lib/python3.10/idlelib/idle_test/__pycache__/test_autocomplete.cpython-310.pyc
/home/user/miniconda3/envs/myenv/lib/python3.10/idlelib/idle_test/__pycache__/test_autocomplete_w.cpython-310.pyc
/home/user/miniconda3/envs/myenv/lib/python3.10/idlelib/autocomplete.py
/home/user/miniconda3/envs/myenv/lib/python3.10/idlelib/__pycache__/autocomplete_w.cpython-310.pyc
/home/user/miniconda3/envs/myenv/lib/python3.10/idlelib/__pycache__/autocomplete.cpython-310.pyc
/home/user/miniconda3/envs/myenv/lib/python3.10/rlcompleter.py
/home/user/miniconda3/envs/myenv/lib/python3.10/__pycache__/rlcompleter.cpython-310.pyc
/home/user/miniconda3/envs/myenv/bin/ompl_benchmark_statistics.py
/home/user/miniconda3/envs/myenv/share/ompl/demos/StateSampling.py
/home/user/miniconda3/envs/myenv/share/ompl/demos/OptimalPlanning.py
/home/user/miniconda3/envs/myenv/share/ompl/demos/RigidBodyPlanningWithControls.py
/home/user/miniconda3/envs/myenv/share/ompl/demos/RigidBodyPlanningWithODESolverAndControls.py
/home/user/miniconda3/envs/myenv/share/ompl/demos/RigidBodyPlanning.py
/home/user/miniconda3/envs/myenv/share/ompl/demos/VFRRT/plotConservative.py
/home/user/miniconda3/envs/myenv/share/ompl/demos/VFRRT/__pycache__/plotNonconservative.cpython-39.pyc
/home/user/miniconda3/envs/myenv/share/ompl/demos/VFRRT/__pycache__/plotConservative.cpython-39.pyc
/home/user/miniconda3/envs/myenv/share/ompl/demos/VFRRT/plotNonconservative.py
/home/user/miniconda3/envs/myenv/share/ompl/demos/KinematicChainPathPlot.py
/home/user/miniconda3/envs/myenv/share/ompl/demos/Koules/__pycache__/KoulesPlayback.cpython-39.pyc
/home/user/miniconda3/envs/myenv/share/ompl/demos/Koules/KoulesPlayback.py
/home/user/miniconda3/envs/myenv/share/ompl/demos/Point2DPlanning.py
/home/user/miniconda3/envs/myenv/share/ompl/demos/RandomWalkPlanner.py
/home/user/miniconda3/envs/myenv/share/ompl/demos/__pycache__/RigidBodyPlanningWithODESolverAndControls.cpython-39.pyc
/home/user/miniconda3/envs/myenv/share/ompl/demos/__pycache__/OptimalPlanning.cpython-39.pyc
/home/user/miniconda3/envs/myenv/share/ompl/demos/__pycache__/RandomWalkPlanner.cpython-39.pyc
/home/user/miniconda3/envs/myenv/share/ompl/demos/__pycache__/KinematicChainPathPlot.cpython-39.pyc
/home/user/miniconda3/envs/myenv/share/ompl/demos/__pycache__/Point2DPlanning.cpython-39.pyc
/home/user/miniconda3/envs/myenv/share/ompl/demos/__pycache__/PlannerData.cpython-39.pyc
/home/user/miniconda3/envs/myenv/share/ompl/demos/__pycache__/StateSampling.cpython-39.pyc
/home/user/miniconda3/envs/myenv/share/ompl/demos/__pycache__/RigidBodyPlanningWithControls.cpython-39.pyc
/home/user/miniconda3/envs/myenv/share/ompl/demos/__pycache__/RigidBodyPlanning.cpython-39.pyc
/home/user/miniconda3/envs/myenv/share/ompl/demos/PlannerData.py
```

traversaro, 2021-12-07:
If you want to install ompl via conda-forge, I guess that the right place to discuss this is https://github.com/conda-forge/ompl-feedstock . In particular this is a duplicate of https://github.com/conda-forge/ompl-feedstock/issues/3 .