ompl/ompl #860: No Python class registered for C++ class std::shared_ptr<ompl::base::MotionValidator>

rallen10, 2021-12-21:
I am trying to make use of a [MotionValidator](https://ompl.kavrakilab.org/classompl_1_1base_1_1MotionValidator.html) using the python bindings but running into an error I haven't been able to debug. Not sure if it's in my implementation or a problem with the actual python bindings.

To reproduce:

```
from ompl import base as ob
si = ob.SpaceInformation(ob.SE2StateSpace())
mvalidator = ob.MotionValidator(si)
si.setMotionValidator(mvalidator)
si.getMotionValidator()

TypeError: No Python class registered for C++ class std::shared_ptr<ompl::base::MotionValidator>
```

mamoll, 2021-12-21:
Boost.Python should convert `shared_ptr<T>` to `T` in Python, but it doesn't in this instance. I can confirm this bug. Even if that conversion worked, your code would still be broken. MotionValidator is an abstract base class. If you don't set one explicitly, an instance of `DiscreteMotionValidator` will be created.

rallen10, 2021-12-22:
@mamoll Is this something that I can quick fix and recompile the python bindings, or does it need a more extensive fix as part of a new OMPL version? Sorry I don't have any experience with Boost.Python.

I think I have workarounds that don't make direct use of MotionValidator, so it's not urgent, but if I can patch it myself then it would be better to use MotionValidator.

JojoSim0n, 2023-12-27:
did something happend here? Since i run into the same problem and would like to use the motion validator but the error still proceed

JojoSim0n, 2023-12-27:
i have the same problem and would like to know if a fix is present? Even if i register a custom spacetimeMotionvlidator and run si.setMotionValidator the getMotionValidator results in the same error as above