ompl/ompl #861: How to access quaternions in SE3 statespace ,I am getting error if i set bounds to 7

muk465, 2021-12-29:
space = ob.SE3StateSpace()
  
     # set lower and upper bounds
bounds = ob.RealVectorBounds(7)
bounds.setLow(-1)
bounds.setHigh(256)
space.setBounds(bounds)
I am using a control based rrt where i require use of quaternions ,can anyone tell how to define bounds for them?

muk465, 2021-12-29:
because if i use state[3] i.e 1st quaternion then kernel dies , so i need to define bounds for this but how to do that @mamoll

muk465, 2021-12-29:
bounds = ob.RealVectorBounds(7) also this seems not possible as it only takes 3 as argument

zkingston, 2021-12-29:
See [this demo](http://ompl.kavrakilab.org/RigidBodyPlanningWithControls_8cpp_source.html) for an example with SE2, which translates to SE3.

closed 2021-12-29 by zkingston