ompl/ompl #864: Multiprocessing with python bindings raises RunTimeError due to inability to pickle

rallen10, 2022-01-03:
Context: 
I am attempting to use OMPL in a machine learning context that leverages ML tools like [PyTorch Lightning](https://www.pytorchlightning.ai/) (thus the need to use OMPL's python bindings, not the underlying C++ API). For this workflow to be effective, I need to run some of OMPL's functions within parallel processes. For example, I create a [control.SpaceInformation object](https://ompl.kavrakilab.org/classompl_1_1control_1_1SpaceInformation.html) and then call the underlying [StatePropagator.propagate() function](https://ompl.kavrakilab.org/classompl_1_1control_1_1StatePropagator.html) many times in parallel using something like [multiprocessing](https://docs.python.org/3/library/multiprocessing.html), [multiprocess](https://pypi.org/project/multiprocess/), or [joblib](https://joblib.readthedocs.io/en/latest/).

No matter which multi-process library I use, I always end up getting an error to the effect of:
```bash
_pickle.PicklingError: Could not pickle the task to send it to the workers.
```

I boiled it down to the simplest example I could. This example only uses the [PPM module](https://ompl.kavrakilab.org/classompl_1_1PPM.html), not the entire SpaceInformation object, but I thought this was a more simplified example of the error and I need PPM to work for my use case, anyway.

```bash
from ompl import util as ou
import pickle
ppm_obj = ou.PPM()
pickle.dumps(ppm_obj)
    RuntimeError: Pickling of "ompl.util._util.PPM" instances is not enabled (http://www.boost.org/libs/python/doc/v2/pickle.html)
```

I also tried using [`dill`](https://pypi.org/project/dill/) in place of `pickle` because it can serialize objects that `pickle` cannot, but I got the exact same error.

Any thoughts on a workaround?

Thank you.

zkingston, 2022-01-05:
Unfortunately I believe that we would need to change the Python binding creation process in order to support pickling - as the bindings are wrappers around underlying C++ objects that do not necessarily have the supported serialization routines. I would recommend writing your own initialization routine / factory pattern that can generate the OMPL objects based on some serializable specification rather than sending the OMPL objects themselves.

closed 2022-01-05 by zkingston

rallen10, 2022-01-07:
Thank you. Based on this response---as well as stackoverflow threads on pickling un-pickleable objects (see [here](https://stackoverflow.com/a/7309686/4055705) and [here](https://stackoverflow.com/a/50308545/4055705))---I think I've found some workarounds. I can use [`copyreg`](https://docs.python.org/3/library/copyreg.html) and wrapper classes that define  [`__getstate__`](https://docs.python.org/3/library/pickle.html#object.__getstate__), [`__setstate__`](https://docs.python.org/3/library/pickle.html#object.__setstate__), and [`__reduce__`](https://docs.python.org/3/library/pickle.html#object.__reduce__) functions. It might be pretty inefficient, and potentially error-prone, but I thought I would share it for posterity.

Here is a minimal viable workaround for [my previous example](https://github.com/ompl/ompl/issues/864#issue-1092891794). It makes use of the [floor.ppm](https://github.com/ompl/ompl/blob/1.5.2/tests/resources/ppm/floor.ppm) file from ompl demos.

```pickle_ppm.py
import pickle
import copyreg
from ompl import base as ob
from ompl import util as ou

# Wrap ou.PPM in a pickleable class
class PickleablePPM(ou.PPM):
    def __init__(self):
        super().__init__()

    def __getstate__(self):
        state = dict()
        state['width'] = self.getWidth()
        state['height'] = self.getHeight()
        state['pixels'] = self.getPixels()
        return state

    def __setstate__(self, state):
        self.setWidth(state['width'])
        self.setHeight(state['height'])
        pix = self.getPixels()
        pix = state['pixels']

    def __reduce__(self):
        return (PickleablePPM, (), self.__getstate__())

# Since PPM objects contain vectorPPMColor objects, we also need to
# define pickling function for vectorPPMColor
def pickle_vectorPPMColor(pixels):
    rgb = [(pix.red, pix.green, pix.blue) for pix in pixels]
    return unpickle_vectorPPMColor, (rgb,)

# define un-pickling function for vectorPPMColor
def unpickle_vectorPPMColor(rgb):
    pixels = ou.vectorPPMColor()
    for (r,g,b) in rgb:
        color = ou.PPM.Color()
        color.red = r
        color.green = g
        color.blue = b
        pixels.append(color)
    return pixels

# tell pickle how to pickle vectorPPMColor objects
copyreg.pickle(ou.vectorPPMColor, pickle_vectorPPMColor, unpickle_vectorPPMColor)  

# ensure that PPM object can now be pickled
ppm = PickleablePPM()
assert pickle.dumps(ppm)

# ensure that you can copy and reproduce an exact PPM object
ppm.loadFile('floor.ppm')
ppm_copy = pickle.loads(pickle.dumps(ppm))
assert ppm.getWidth() == ppm_copy.getWidth()
assert ppm.getHeight() == ppm_copy.getHeight()
for i, p1, p2 in enumerate(zip(ppm.getPixels(), ppm_copy.getPixels())):
    assert p1.red == p2.red
    assert p1.green == p2.green
    assert p1.blue == p2.blue
```

Even though this example is very narrow in scope---only working with the `PPM` class---the patterns are reusable for pickling and multi-processing other OMPL objects in Python.

Further note that this is pretty clunky; you need to define 2 functions and a class with 3 methods just to pickle one OMPL object that contains one other OMPL object. This looks untenable if you are wanting to pickle OMPL objects that contain many other OMPL objects. To work around this, it is much easier if you use wrapper classes that instantiate OMPL objects during `__init__`, and then never modify those underlying OMPL objects (at least don't modify them during multiprocessing when the pickling must occur). In this case you can greatly simplify the above example and avoid the need to define pickling functions for underlying objects such as `vectorPPMColor`

```pickle_ppm2.py
import pickle
from ompl import base as ob
from ompl import util as ou

# Wrap ou.PPM in a pickleable class
class PickleablePPM(ou.PPM):
    def __init__(self, load_file):
        super().__init__()
        self.load_file = load_file
        self.loadFile(self.load_file)

    def __reduce__(self):
        return (PickleablePPM, (self.load_file,))

# ensure that PPM object can now be pickled
ppm = PickleablePPM('floor.ppm')
assert pickle.dumps(ppm)

# ensure that you can copy and reproduce an exact PPM object
ppm_copy = pickle.loads(pickle.dumps(ppm))
assert ppm.getWidth() == ppm_copy.getWidth()
assert ppm.getHeight() == ppm_copy.getHeight()
assert len(ppm.getPixels()) == len(ppm_copy.getPixels())
for i, pix in enumerate(ppm.getPixels()):
    pix.red == ppm_copy.getPixels()[i].red
    pix.green == ppm_copy.getPixels()[i].green
    pix.blue == ppm_copy.getPixels()[i].blue
```