ompl/ompl #866: Planner Converging to Something Other Than Goal

fishbotics, 2022-01-12:
Hi, 

I am using the AIT* Planner from Python and the planner seems to often converge to a configuration that's completely different from the goal I gave to the planner.

I've put in how I use the API below. Apologies that this is kind of long. I basically look at the final state in the path and check if it's close to the goal. That assert fails very often. 

Am I doing something wrong?

This error comes up in OMPL 1.5.2.      
```                                              
# Define the state space                                                
space = ob.RealVectorStateSpace(FrankaRobot.DOF)                        
                                                                                
# Set the boundaries on the state space via the joint limits            
bounds = ob.RealVectorBounds(FrankaRobot.DOF)                           
for ii in range(FrankaRobot.DOF):                                       
    low, high = FrankaRobot.JOINT_LIMITS[ii]                            
    bounds.setLow(ii, low)                                              
    bounds.setHigh(ii, high)                                            
space.setBounds(bounds)                                                 
                                                                                
# Space information is an object that wraps the planning space itself, as well as
# other properties about the space. Most notably, it has a reference to 
# a collision checker able to verify a specific configuration           
space_information = ob.SpaceInformation(space)                          
                                                                                
# Sets the validity checker as a function. This can also be a class if there are
# additional methods that need to be defined, for example to use in an optimizing object
# This uses an external collision check (from Pybullet)
def collision_checker(q):                                               
        state = np.zeros(7)                                                 
        for ii in range(7):                                                 
            state[ii] = q[ii]                                               
        return self._not_in_collision(state)                                
                                                                                
space_information.setStateValidityChecker(                              
    ob.StateValidityCheckerFn(collision_checker)                        
)                                                                       
space_information.setup()                                               
                                                                                
# Define a planning problem on the planning space with the given collision checker
pdef = ob.ProblemDefinition(space_information)

# Copy the start and goal states into the OMPL representation           
# start and goal come in as np arrays
start_state = ob.State(space)                                           
for ii in range(FrankaRobot.DOF):                                       
    start_state[ii] = start[ii]                                         
pdef.addStartState(start_state)                                         
                                                                        
goal_state = ob.GoalState(space_information)                            
gstate = ob.State(space)                                                
for ii in range(FrankaRobot.DOF):                                       
    gstate[ii] = goal[ii]                                               
goal_state.setState(gstate)                                             
pdef.setGoal(goal_state)                                                
                                                                        
# The planning problem needs to know what it's optimizing for           
pdef.setOptimizationObjective(                                          
    ob.PathLengthOptimizationObjective(space_information)               
)                                                                       
                                                                        
# Set up the actual planner and give it the problem                     
optimizing_planner = og.AITstar(space_information)                      
                                                                        
optimizing_planner.setProblemDefinition(pdef)                           
optimizing_planner.setup()                                              
                                                                        
start_time = time.time()                                                

solved = optimizing_planner.solve(max_runtime)
if not solved:                                                          
    return None                                                         
path = pdef.getSolutionPath()                                                                                            
path_length = path.getStateCount()                                          
path = [[path.getState(ii)[j] for j in range(dof)] for ii in range(path_length)]
assert np.allclose(goal, path[-1], atol=0.01)
```

gammell, 2022-01-12:
Hi @fishbotics,

Have you checked if the solution found by the planner is exact or only approximate?

See `hasSolution`, `hasExactSolution`, and `hasApproximateSolution` here:
http://ompl.kavrakilab.org/classompl_1_1base_1_1ProblemDefinition.html

fishbotics, 2022-01-12:
Thanks @gammell! It does appear that the solution is only approximate. How is an approximate solution defined? Is this a parameter I can set?

Also, is there a way to use an exact solution as the terminating condition instead of time? Or perhaps in addition to time?

fishbotics, 2022-01-12:
FWIW, I tried replacing the call to solve with
```
solved = optimizing_planner.solve(                       
    ob.plannerOrTerminationCondition(                    
        ob.timedPlannerTerminationCondition(max_runtime),
        ob.exactSolnPlannerTerminationCondition(pdef),   
    )                                                    
)                                                        
```
and this produces a segfault

In fact, when I even just do 
```
solved = optimizing_planner.solve(ob.timedPlannerTerminationCondition(max_runtime))
```
I also get a segfault

gammell, 2022-01-13:
I think maybe you should open a new issue about the segfault in the `exactSolnPlannerTerminationCondition`, I'm not familiar with that code.

Another way to tell optimizing planners to end when a solution is found is to set the cost threshold, `OptimizationObjective::setCostThreshold()`, to infinity, ` OptimizationObjective::infiniteCost()`, see:
http://ompl.kavrakilab.org/classompl_1_1base_1_1OptimizationObjective.html

That will cause a (properly written) optimizing planner to return as soon as the cost of the solution is finite (which should be any exact solution). But if you ultimately don't care about solution cost, you probably want to try nonoptimizing planners too.

fishbotics, 2022-01-13:
Thanks @gammell. This is very helpful. I'll try setting the cost to infinity and report back.