ompl/ompl #867: AITStar Planner Segfault With Custom Termination Condition

fishbotics, 2022-01-13:
Hi all,

As I mentioned in #866, when I run AITStar in version 1.5.2 with a custom termination condition (such as converging to an exact solution), I get a segfault. I'm using the Python bindings. Am I using the custom termination conditions correctly?

Thanks!

Here is a sample of the code:
```
# Define the state space                                                
space = ob.RealVectorStateSpace(FrankaRobot.DOF)                        
                                                                                
# Set the boundaries on the state space via the joint limits            
bounds = ob.RealVectorBounds(FrankaRobot.DOF)                           
for ii in range(FrankaRobot.DOF):                                       
    low, high = FrankaRobot.JOINT_LIMITS[ii]                            
    bounds.setLow(ii, low)                                              
    bounds.setHigh(ii, high)                                            
space.setBounds(bounds)                                                 
                                                                                
# Space information is an object that wraps the planning space itself, as well as
# other properties about the space. Most notably, it has a reference to 
# a collision checker able to verify a specific configuration           
space_information = ob.SpaceInformation(space)                          
                                                                                
# Sets the validity checker as a function. This can also be a class if there are
# additional methods that need to be defined, for example to use in an optimizing object
# This uses an external collision check (from Pybullet)
def collision_checker(q):                                               
        state = np.zeros(7)                                                 
        for ii in range(7):                                                 
            state[ii] = q[ii]                                               
        return self._not_in_collision(state)                                
                                                                                
space_information.setStateValidityChecker(                              
    ob.StateValidityCheckerFn(collision_checker)                        
)                                                                       
space_information.setup()                                               
                                                                                
# Define a planning problem on the planning space with the given collision checker
pdef = ob.ProblemDefinition(space_information)

# Copy the start and goal states into the OMPL representation           
# start and goal come in as np arrays
start_state = ob.State(space)                                           
for ii in range(FrankaRobot.DOF):                                       
    start_state[ii] = start[ii]                                         
pdef.addStartState(start_state)                                         
                                                                        
goal_state = ob.GoalState(space_information)                            
gstate = ob.State(space)                                                
for ii in range(FrankaRobot.DOF):                                       
    gstate[ii] = goal[ii]                                               
goal_state.setState(gstate)                                             
pdef.setGoal(goal_state)                                                
                                                                        
# The planning problem needs to know what it's optimizing for           
pdef.setOptimizationObjective(                                          
    ob.PathLengthOptimizationObjective(space_information)               
)                                                                       
                                                                        
# Set up the actual planner and give it the problem                     
optimizing_planner = og.AITstar(space_information)                      
                                                                        
optimizing_planner.setProblemDefinition(pdef)                           
optimizing_planner.setup()                                              
                                                                        
start_time = time.time()                                                

solved = optimizing_planner.solve(                       
    ob.plannerOrTerminationCondition(                    
        ob.timedPlannerTerminationCondition(max_runtime),
        ob.exactSolnPlannerTerminationCondition(pdef),   
    )                                                    
)        
if not solved:                                                          
    return None                                                         
path = pdef.getSolutionPath()                                                                                            
path_length = path.getStateCount()                                          
path = [[path.getState(ii)[j] for j in range(dof)] for ii in range(path_length)]
assert np.allclose(goal, path[-1], atol=0.01)
```

fishbotics, 2022-01-13:
Alternatively, when I use this, I also get a segfault

```
solved = optimizing_planner.solve(ob.timedPlannerTerminationCondition(max_runtime))
```

marlinstrub, 2022-01-18:
Hi Adam,

Thanks for reporting this. To me it looks like you're using the termination conditions correctly but I'm curious about your collision checker. I never code in Python but I thought Python's `self` was an analogue to C++'s `this` and it looks like you're using it in a free-standing function? Anyway, trying to narrow down the issue -- Can you please:

- Try another planner to see whether the segfault only happens with AIT*
- Provide an [MWE](https://en.wikipedia.org/wiki/Minimal_working_example) (preferably in C++ if at all possible)

Thanks!

fishbotics, 2022-02-19:
Hi @marlinstrub, sorry for the slow reply. I think I may have fixed the issue on my end and I can't repro this anymore (the problem of coding in a RSS-deadline-haze for three weeks). If I stumble into the issue again, I will make a C++ MWE.

Also, fwiw you are right about the `self` variable. This code typically lives in a little wrapper library I have written around OMPL and I forgot to strip that out before posting the demo code.

closed 2022-02-19 by fishbotics

marlinstrub, 2022-02-20:
No problem at all, @fishbotics, I'm glad it's not an issue anymore.