ompl/ompl #868: New release?

simonschmeisser, 2022-01-13:
Some important bugs have been fixed and it would be nice to have a new (patch) release. Both for ros distribution as well as for getting those bug fixes included with Ubuntu 22.04

@mamoll @zkingston 

E.g.: https://github.com/ros-planning/moveit/issues/1920 via #739

mamoll, 2022-01-17:
We're planning to do this soon. @gammell are there any last-minute things you'd want to include in a patch-level release?

gammell, 2022-01-18:
I don't think we are sitting on any patch-level changes at the moment. @marlinstrub would know best if we have anything, but he's very busy at the moment, so you shouldn't necessarily wait on his response.

k2shah, 2022-09-21:
bump!
any chance of a new release?

simonschmeisser, 2022-11-01:
Friendly ping

mamoll, 2022-11-01:
@zkingston?

zkingston, 2022-11-01:
Added to the todo list - hopefully will push out within a week or so.

zkingston, 2023-01-18:
1.6.0 is out now!

closed 2023-01-18 by zkingston