ompl/ompl #871: Seeding StateSampler::rng_

alimaiem, 2022-01-17:
Hello,

Is there a way to use your own seed for StateSampler::rng_? This is particularly critical in our application, where we need to have the same result each time we solve the same problem...

Thanks,

mamoll, 2022-01-17:
See the documentation for the `RNG::setLocalSeed` and `RNG::setSeed` methods [here](http://ompl.kavrakilab.org/classompl_1_1RNG.html).

closed 2022-01-17 by mamoll

alimaiem, 2022-01-17:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<http://aka.ms/LearnAboutSenderIdentification>
> This email is not from Hexagon's Office 365 instance. Please be careful while clicking links, opening attachments, or replying to this email.

Thanks for your feedback.
I know about the documentation for RNG. My question is more about how to access StateSampler::rng_ in order to seed it.
Looking more at the doc, I think I just need to use a deterministic sate sampler?