ompl/ompl #888: failed to add new files in ompl/base

rofgmd, 2022-03-22:
Hello, I have a problem. I added  new .h file called InformedGaussianValidStateSampler.h and new .cpp file  InformedGaussianValidStateSampler.cpp. Both files are just copied UniformValidStateSampler.h and UniformValidStateSampler.cpp. And I used this new class in StateSampling.cpp.

I compiled successfully. The result is as follows.
xxx@ubuntu:~/ompl-1.5.2/build/Release$ cmake ../..
-- Build type: Release
-- Checking for module 'libspot'
--   No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS) 
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * pypy, <https://pypy.org>
   Used to speed up the generation of python bindings.
 * ODE, <http://ode.org>
   The Open Dynamics Engine can be used as a "black box" for kinodynamic planning.
 * flann (required version >= 1.8.3), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * Python, <https://www.python.org>
   Used for python bindings.
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.

-- The following REQUIRED packages have been found:

 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org).

-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/xxx/ompl-1.5.2/build/Release
xxx@ubuntu:~/ompl-1.5.2/build/Release$ make
[ 59%] Built target ompl
[ 60%] Built target test_ptc
[ 61%] Built target test_planner_data_control
[ 62%] Built target test_halton_sampling
[ 63%] Built target test_state_spaces
[ 63%] Built target test_2dcircles_opt_geometric
[ 64%] Built target test_pdf
[ 64%] Built target test_state_storage
[ 64%] Built target test_gridb
[ 65%] Built target test_grid
[ 67%] Built target test_nearestneighbors
[ 67%] Built target test_heap
[ 67%] Built target test_planner_data
[ 68%] Built target test_2dmap_ik
[ 69%] Built target test_2dpath_simplifying
[ 69%] Built target test_adjacency_list
[ 70%] Built target test_2denvs_geometric
[ 70%] Built target test_2dmap_geometric_simple
[ 71%] Built target test_state_operations
[ 72%] Built target test_constraint_sphere
[ 73%] Built target py_std_function
[ 74%] Built target test_2dmap_control
[ 74%] Built target test_random
[ 75%] Built target regression_test
[ 76%] Built target demo_HypercubeBenchmark
[ 76%] Built target demo_GeometricCarPlanning
[ 77%] Built target demo_StateSampling
[ 78%] Built target demo_HybridSystemPlanning
[ 78%] Built target demo_RigidBodyPlanningWithIntegrationAndControls
[ 78%] Built target demo_ConstrainedPlanningImplicitChain
[ 78%] Built target demo_RigidBodyPlanningWithControls
[ 81%] Built target demo_Koules
[ 81%] Built target demo_ConstrainedPlanningSphere
[ 82%] Built target demo_RigidBodyPlanningWithODESolverAndControls
[ 83%] Built target demo_VectorFieldNonconservative
[ 84%] Built target demo_QuotientSpacePlanningHyperCubeBenchmark
[ 84%] Built target demo_KinematicChainBenchmark
[ 85%] Built target demo_QuotientSpacePlanningKinematicChain
[ 86%] Built target demo_RigidBodyPlanningWithIK
[ 87%] Built target demo_OpenDERigidBodyPlanning
[ 88%] Built target demo_PlannerProgressProperties
[ 88%] Built target demo_PlannerData
[ 89%] Built target demo_RigidBodyPlanning
[ 90%] Built target demo_CForestCircleGridBenchmark
[ 90%] Built target demo_QuotientSpacePlanningRigidBody2D
[ 91%] Built target demo_Diagonal
[ 92%] Built target demo_VectorFieldConservative
[ 93%] Built target demo_ConstrainedPlanningTorus
[ 94%] Built target demo_QuotientSpacePlanningRigidBody3D
[ 95%] Built target demo_Point2DPlanning
[ 96%] Built target demo_ConstrainedPlanningImplicitParallel
[ 98%] Built target demo_PlanarManipulator
[100%] Built target demo_ConstrainedPlanningKinematicChain
[100%] Built target demo_OptimalPlanning
[100%] Built target demo_ThunderLightning
[100%] Built target demo_QuotientSpacePlanningHyperCube

But when I run demo_StateSampling, I get the following error
xxx@ubuntu:~/ompl-1.5.2/build/Release/bin$ ./demo_StateSampling 
./demo_StateSampling: symbol lookup error: ./demo_StateSampling: undefined symbol: _ZTVN4ompl4base33InformedGaussianValidStateSamplerE

I thought the problem should be with the DLL, so I used the ldd command to see the problem
xxx@ubuntu:~/ompl-1.5.2/build/Release/lib$ ldd -r libompl.so.16
        linux-vdso.so.1 (0x00007fff367e6000)
        libode.so.8 => /lib/x86_64-linux-gnu/libode.so.8 (0x00007fd2bd28b000)
        libboost_serialization.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_serialization.so.1.71.0 (0x00007fd2bd247000)
        libboost_filesystem.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007fd2bd229000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd2bd206000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd2bd024000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd2bced5000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd2bceb8000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd2bccc6000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fd2bdb5b000)
        libccd.so.2 => /lib/x86_64-linux-gnu/libccd.so.2 (0x00007fd2bccba000)
xxx@ubuntu:~/ompl-1.5.2/build/Release/lib$ ls
libompl.so  libompl.so.1.5.2  libompl.so.16  py_std_function.so

How should I do. I think problem is libompl

rofgmd, 2022-03-22:
[CMakeCache.txt](https://github.com/ompl/ompl/files/8325040/CMakeCache.txt)

rofgmd, 2022-03-22:
xxx@ubuntu:~/ompl-1.5.2/build/Release$ make VERBOSE=1
/usr/bin/cmake -S/home/xxx/ompl-1.5.2 -B/home/xxx/ompl-1.5.2/build/Release --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/xxx/ompl-1.5.2/build/Release/CMakeFiles /home/xxx/ompl-1.5.2/build/Release/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/xxx/ompl-1.5.2/build/Release'
make -f src/ompl/CMakeFiles/ompl.dir/build.make src/ompl/CMakeFiles/ompl.dir/depend
make[2]: Entering directory '/home/xxx/ompl-1.5.2/build/Release'
cd /home/xxx/ompl-1.5.2/build/Release && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/xxx/ompl-1.5.2 /home/xxx/ompl-1.5.2/src/ompl /home/xxx/ompl-1.5.2/build/Release /home/xxx/ompl-1.5.2/build/Release/src/ompl /home/xxx/ompl-1.5.2/build/Release/src/ompl/CMakeFiles/ompl.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/xxx/ompl-1.5.2/build/Release'
make -f src/ompl/CMakeFiles/ompl.dir/build.make src/ompl/CMakeFiles/ompl.dir/build
make[2]: Entering directory '/home/xxx/ompl-1.5.2/build/Release'
make[2]: Nothing to be done for 'src/ompl/CMakeFiles/ompl.dir/build'.
make[2]: Leaving directory '/home/xxx/ompl-1.5.2/build/Release'
[ 59%] Built target ompl
make -f tests/CMakeFiles/test_ptc.dir/build.make tests/CMakeFiles/test_ptc.dir/depend
make[2]: Entering directory '/home/xxx/ompl-1.5.2/build/Release'
cd /home/xxx/ompl-1.5.2/build/Release && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/xxx/ompl-1.5.2 /home/xxx/ompl-1.5.2/tests /home/xxx/ompl-1.5.2/build/Release /home/xxx/ompl-1.5.2/build/Release/tests /home/xxx/ompl-1.5.2/build/Release/tests/CMakeFiles/test_ptc.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/xxx/ompl-1.5.2/build/Release'
make -f tests/CMakeFiles/test_ptc.dir/build.make tests/CMakeFiles/test_ptc.dir/build
make[2]: Entering directory '/home/xxx/ompl-1.5.2/build/Release'
make[2]: Nothing to be done for 'tests/CMakeFiles/test_ptc.dir/build'.
make[2]: Leaving directory '/home/xxx/ompl-1.5.2/build/Release'
[ 60%] Built target test_ptc
make -f tests/CMakeFiles/test_planner_data_control.dir/build.make tests/CMakeFiles/test_planner_data_control.dir/depend
make[2]: Entering directory '/home/xxx/ompl-1.5.2/build/Release'
cd /home/xxx/ompl-1.5.2/build/Release && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/xxx/ompl-1.5.2 /home/xxx/ompl-1.5.2/tests /home/xxx/ompl-1.5.2/build/Release /home/xxx/ompl-1.5.2/build/Release/tests /home/xxx/ompl-1.5.2/build/Release/tests/CMakeFiles/test_planner_data_control.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/xxx/ompl-1.5.2/build/Release'
make -f tests/CMakeFiles/test_planner_data_control.dir/build.make tests/CMakeFiles/test_planner_data_control.dir/build
make[2]: Entering directory '/home/xxx/ompl-1.5.2/build/Release'
make[2]: Nothing to be done for 'tests/CMakeFiles/test_planner_data_control.dir/build'.
make[2]: Leaving directory '/home/xxx/ompl-1.5.2/build/Release'
[ 61%] Built target test_planner_data_control
make -f tests/CMakeFiles/test_halton_sampling.dir/build.make tests/CMakeFiles/test_halton_sampling.dir/depend
make[2]: Entering directory '/home/xxx/ompl-1.5.2/build/Release'
cd /home/xxx/ompl-1.5.2/build/Release && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/xxx/ompl-1.5.2 /home/xxx/ompl-1.5.2/tests /home/xxx/ompl-1.5.2/build/Release /home/xxx/ompl-1.5.2/build/Release/tests /home/xxx/ompl-1.5.2/build/Release/tests/CMakeFiles/test_halton_sampling.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/xxx/ompl-1.5.2/build/Release'
make -f tests/CMakeFiles/test_halton_sampling.dir/build.make tests/CMakeFiles/test_halton_sampling.dir/build
make[2]: Entering directory '/home/xxx/ompl-1.5.2/build/Release'
make[2]: Nothing to be done for 'tests/CMakeFiles/test_halton_sampling.dir/build'.
make[2]: Leaving directory '/home/xxx/ompl-1.5.2/build/Release'
[ 62%] Built target test_halton_sampling
make -f tests/CMakeFiles/test_state_spaces.dir/build.make tests/CMakeFiles/test_state_spaces.dir/depend
make[2]: Entering directory '/home/xxx/ompl-1.5.2/build/Release'
cd /home/xxx/ompl-1.5.2/build/Release && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/xxx/ompl-1.5.2 /home/xxx/ompl-1.5.2/tests /home/xxx/ompl-1.5.2/build/Release /home/xxx/ompl-1.5.2/build/Release/tests /home/xxx/ompl-1.5.2/build/Release/tests/CMakeFiles/test_state_spaces.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/xxx/ompl-1.5.2/build/Release'
make -f tests/CMakeFiles/test_state_spaces.dir/build.make tests/CMakeFiles/test_state_spaces.dir/build
make[2]: Entering directory '/home/xxx/ompl-1.5.2/build/Release'
make[2]: Nothing to be done for 'tests/CMakeFiles/test_state_spaces.dir/build'.
make[2]: Leaving directory '/home/xxx/ompl-1.5.2/build/Release'
[ 63%] Built target test_state_spaces
make -f tests/CMakeFiles/test_2dcircles_opt_geometric.dir/build.make tests/CMakeFiles/test_2dcircles_opt_geometric.dir/depend
make[2]: Entering directory '/home/xxx/ompl-1.5.2/build/Release'
cd /home/xxx/ompl-1.5.2/build/Release && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/xxx/ompl-1.5.2 /home/xxx/ompl-1.5.2/tests /home/xxx/ompl-1.5.2/build/Release /home/xxx/ompl-1.5.2/build/Release/tests /home/xxx/ompl-1.5.2/build/Release/tests/CMakeFiles/test_2dcircles_opt_geometric.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/xxx/ompl-1.5.2/build/Release'
[... cut ...]
make -f demos/CMakeFiles/demo_QuotientSpacePlanningHyperCube.dir/build.make demos/CMakeFiles/demo_QuotientSpacePlanningHyperCube.dir/build
make[2]: Entering directory '/home/xxx/ompl-1.5.2/build/Release'
make[2]: Nothing to be done for 'demos/CMakeFiles/demo_QuotientSpacePlanningHyperCube.dir/build'.
make[2]: Leaving directory '/home/xxx/ompl-1.5.2/build/Release'
[100%] Built target demo_QuotientSpacePlanningHyperCube
make[1]: Leaving directory '/home/xxx/ompl-1.5.2/build/Release'
/usr/bin/cmake -E cmake_progress_start /home/xxx/ompl-1.5.2/build/Release/CMakeFiles 0

zkingston, 2023-01-19:
It might be because in your source file the implementation of InformedGaussianValidStateSampler is not namespaced correctly, so despite compiling it is not being linked properly. Make sure your source file is implementing methods with the correct signature.

closed 2023-01-19 by zkingston