ompl/ompl #890: RRTConnect: Created 2 states (1 start + 1 goal)

muzerobot, 2022-03-26:
`planner` returns just two states, the start and the goal. Does this mean that the start and the goal are too close to one another? Is there a parameter which controls step size, which I could reduce? Or any other ideas as to what could be going on here?

Any help will be much appreciated. Thanks in advance!

ChamzasKonstantinos, 2022-03-27:
Hi @muzerobot . Trying setting the range greater strictly greater than 0. This is an important parameter that needs to be tuned depending on your application but 0 will break RRTConnect. Let me know if this fixes your problem.

muzerobot, 2022-03-27:
Hi @ChamzasKonstantinos, thanks for your response! What is the interpretation of `range`, and what are good values to try? 

And sorry, I think there was a bug in my jacobian computation which was causing the planner to return just two states -- now, I'm able to get more than two states, but `pdef.getSolutionPath() == None` when `range > 0.0` (but works as expected when `range == 0.0`). Is this expected?

Thanks for your help!

ChamzasKonstantinos, 2022-03-28:
Glad to hear, that. Actually I was mistaken, 0 range makes ompl to find a good range heuristically. This is done during the setup step [here](https://github.com/ompl/ompl/blob/main/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp#L66), where the [selfConfigure](https://ompl.kavrakilab.org/classompl_1_1tools_1_1SelfConfig.html#a65bff53ea4bc6f158342a856175ab9a6) function is called. The range parameter is used slightly differently depending on the planner. Specifically, for RRTConnect it controls how fast or slow the search tree expands in the configuration space. A low range will make it expand very slowly and a big range faster. Slow expansion is beneficial when narrow passages exist. Probably the values you chose are very small that is why it did not find a solution. If you search more you could find better values that lead to faster solution times. But if the default range works for you that is also fine.
Since you resolved your problem I will close this issue.

closed 2022-03-28 by ChamzasKonstantinos