ompl/ompl #894: ompl::base::AtlasStateSpace::anchorChart(): Initial chart creation failed. Cannot proceed.

arjung128, 2022-04-03:
Hello, I am trying to do constrained motion planning using the `TangentBundleStateSpace`. When I try to add my start state with `constraint_space.anchorChart(sstart())`, I am getting `ompl::base::AtlasStateSpace::anchorChart(): Initial chart creation failed. Cannot proceed.` I am also getting:

```
Error:   ompl::base::AtlasStateSpace::newChart(): Failed because manifold looks degenerate here.
         at line 277 in ~/ompl/base/spaces/constraint/src/AtlasStateSpace.cpp
```

My constraint function calculates a target rotation for the end-effector (given the current end-effector position), and calculates the angular distance between this target rotation and the current rotation -- when the target rotation is a fixed matrix (i.e. does not depend on the end-effector position), the motion planning seems to work, but not when I try to make it variable...

Any ideas why this might be the case? Thanks!

zkingston, 2022-04-03:
You can debug this by looking at what the Jacobian of the constraint function is at the starting state. If the Jacobian is not of full rank, you will get this error.

closed 2022-04-03 by zkingston

jih189, 2023-12-18:
In some cases, the Jacobian matrix is not full rank, how should we handle it?