ompl/ompl #895: ob.GoalSampleableRegion does not solve with RRTConnect

arjung128, 2022-04-05:
I have a toy example. When I solve it by providing the start and goal state as follows, RRTConnect is perfectly able to solve the problem:

```
# set the start and goal states
start, goal = [0, 0, 0, 0], [0, 0, np.pi/2, 0]
sstart, sgoal = ob.State(space), ob.State(space)
for i in range(space.getDimension()):
    sstart[i] = start[i]
    sgoal[i] = goal[i]
pdef.setStartAndGoalStates(sstart, sgoal)
```

However, when I extend this to use the `ob.GoalSampleableRegion` as follows (based on #615), while `planner.solve()` claims to return an 'Exact solution', the solution is does not solve the problem:

```
# set the start and goal states
start = [0, 0, 0, 0]
sstart = ob.State(space)
for i in range(space.getDimension()):
    sstart[i] = start[i]
pdef.addStartState(sstart)

class MyArbitraryGoal(ob.GoalSampleableRegion):
    def __init__(self, si):
        super(MyArbitraryGoal, self).__init__(si)
        self.setThreshold(1)
        self.L1 = 3
        self.L2 = 3
    def distanceGoal(self, state):
        print("foo")
        x, y, t1, t2 = state
        x_coord = x + self.L1 * np.cos(t1) + self.L2*np.cos(t1+t2)
        return x_coord
    def sampleGoal(self, state):
        state = [0, 0, np.pi/2, 0]
    def maxSampleCount(self):
        return 10

goal = MyArbitraryGoal(si)
pdef.setGoal(goal)
```

The solution often is something of the form `[starting_state, random_state, starting_state]`, and neither of the three states are within the threshold. Also, It turns out that `distanceGoal` does not seem to be called (`foo` is never printed). Any ideas what might be going on here?

Any help will be much appreciated, thanks!

arjung128, 2022-04-13:
I see, thanks! That was helpful.

So I was able to set a goal class as the goal state when using `SimpleSetup` as follows:

```
ss = og.SimpleSetup(space)
goal = MyArbitraryGoal(si)
ss.setGoal(goal)
```

However, when I try to do the same thing using `ProblemDefinition`, I am running into errors:

```
pdef = ob.ProblemDefinition(si)
goal = MyArbitraryGoal(si)
# pdef.setGoal(goal) # print(solved) prints 'Unrecognized goal type'
# pdef.setGoalState(goal) # get the following error:
```

```
ArgumentError: Python argument types in
    ProblemDefinition.setGoalState(ProblemDefinition, MyArbitraryGoal)
did not match C++ signature:
    setGoalState(ompl::base::ProblemDefinition {lvalue}, ompl::base::ScopedState<ompl::base::StateSpace> goal, double threshold=2.220446049250313e-16)
    setGoalState(ompl::base::ProblemDefinition {lvalue}, ompl::base::State const* goal, double threshold=2.220446049250313e-16)
```

I would like to use `ProblemDefinition` so that I can use the `RRTConnect` planner (with constraints). Any ideas how to set an arbitrary goal class as the goal when using `ProblemDefinition`?

Any help will be much appreciated, thanks!

zkingston, 2023-05-05:
Here is an example of using `ProblemDefinition` and planning without `SimpleSetup` and a custom goal:
```py
import ompl.base._base as ob
import ompl.geometric._geometric as og

class AnythingGoal(ob.GoalSampleableRegion):

    def __init__(self, si: ob.SpaceInformation):
        super().__init__(si)
        self.sampler = si.getStateSpace().allocStateSampler()

    def distanceGoal(self, state):
        return 0.

    def sampleGoal(self, state):
        self.sampler.sampleUniform(state)

    def maxSampleCount(self):
        return 1

so3 = ob.SO3StateSpace()
si = ob.SpaceInformation(so3)
si.setStateValidityChecker(ob.AllValidStateValidityChecker(si))
si.setup()

pdef = ob.ProblemDefinition(si)

start = ob.State(so3)
sampler = so3.allocStateSampler()
sampler.sampleUniform(start())
pdef.addStartState(start)
pdef.setGoal(AnythingGoal(si))

planner = og.RRTConnect(si)
planner.setProblemDefinition(pdef)
planner.setup()

solved = planner.solve(1.)
print(solved)
```

closed 2023-05-05 by zkingston