ompl/ompl #897: Failing to obtain optimal solution for small 2D paths in large maps with BIT*

VicenteMovai, 2022-04-06:
Hi, I am relatively new to OMPL and recently have been facing the following problem.

I cannot consistently obtain optimal small 2D paths with OMPL in a large map. I am running BIT* and using the default discrete motion validator, with setStateValidityCheckingResolution(1./si->getMaximumExtent()) in order to be able to plan around thin obstacles.

The costmap used represents an area of (320*130)m and has a resolution of 0,15m. The OMPL planning time is set to 1.0s.
The planning performs very well in generating long plans from one corner of the map to the other, in an environment with multiple corridors.
[image]

However, the trouble lies when I try to generate plans locally, e.g. planning around a corridor.
See for example this plan obtained within the walls on the top-right corner of the first figure.
[image]

While most of the time the plans are acceptable in such a small region, occasionally I get a plan like the one in the figure above, which does not really represent the shortest path possible. It also seems that in these cases, the planner exceeds the planning time, which I thought was not meant to happen with OMPL. In the case above it took over 5s.

I ran some comparisons with A*, which manages to consistently return the shortest path.

So I was wondering what possible setting I might be missing in order to achieve similar consistency in performance from OMPL. Any help is appreciated, thanks in advance!

zkingston, 2022-05-09:
The algorithms in OMPL are sampling-based, and thus have variance in their performance. It is concerning that BIT* took longer than the termination condition specified - can you inspect and see where the algorithm is getting stuck (e.g., by stopping and inspecting with gdb)? This usually is indicative of somewhere in the inner loop taking too long.

To get more consistent performance from sampling-based algorithms, I'd recommend using a 
[deterministic state sampler](https://ompl.kavrakilab.org/classompl_1_1base_1_1DeterministicStateSampler.html). I'd also recommend trying out using [path hybridiziation](https://ompl.kavrakilab.org/classompl_1_1geometric_1_1PathHybridization.html) with a generally fast planner (e.g., RRTConnect) instead of an AO planner. [Parallel Plan](https://ompl.kavrakilab.org/classompl_1_1tools_1_1ParallelPlan.html) is a useful tool for using this as well.

VicenteMovai, 2022-06-07:
Dear @zkingston ,

Once again thanks a lot for your feedback and sorry for the delay in getting back to you.
I finally managed to do some debugging to try and understand why the ompl solver occasionally exceeds the planning time.
I found that the problem occurs in the pruning function (BITstar::ImplicitGraph::pruneSamples()) when it has to loop over the samples in order to prune them:
_https://ompl.kavrakilab.org/bitstar_2src_2ImplicitGraph_8cpp_source.html_   l.1205-1213

`         for (const auto &sample : samples)
                 {
                     if (!sample->isInTree())
                     {
                         this->pruneSample(sample);
                         ++numPruned.second;
                         ++numFreeStatesPruned;
                     }
                 }`
                 
It isn't the pruneSample() function itself that takes too long to process, but rather the thousands of iterations called by the for-loop.
When I set the pruning to false, I no longer observe solver times higher than the planning time. The plans themselves look much better, and I no longer obtain plans like the ones shown in the 2nd figure above. However, since the solver does not perform any pruning, the final plans aren't always as optimal as they could be.
I have not yet tried any of the ompl tools you have suggested to try and improve the consistency of the ompl performance (deterministic sampler, path hybridisation and parallel plan). Once I get the opportunity to run some tests with those tools, I will report back to you here.

Many thanks for your help!

closed 2022-06-21 by zkingston

VicenteMovai, 2022-06-21:
Setting pruning to false solved my issues. But when it's true, the planning time is exceeded sometimes. 
And this is due to the loop above. Any plans of looking into this in the future?

By the way, I tried your recommendations and found that using a Gaussian Valid State Sampler seems to improve my plans.

Thanks a lot!

marlinstrub, 2022-06-21:
@VicenteMovai I will look into this, thanks for reporting it.