ompl/ompl #899: [Python] Accessing all states as part of the the current considered solution using BITStar

Nikolov-Georgi, 2022-04-14:
Hey, I am using BITStar as a planner and I have tried setting up both a MotionValidator and a StateValidityChecker for determining the validity of the considered states.

Looking at the image below:
The general idea is to move from Start towards the Goal with minimizing both the path length and the distance from Sub-Goal1. That part is setup and working.

However, when we reach a State, where the Sub-Goal is within the magenta rectangle (we'll call this "the trigger"), the validity of the forthcoming States needs to be determined by the distance travelled after the triger. Hence, i need to know the chain of previous states.

Looking at the documentation, I would assume that I need checkMotion (2/2) described in the link below:[https://ompl.kavrakilab.org/classompl_1_1base_1_1SpaceInformation.html](url)

However, when try to define a checkMotion function in my MotionValidator class that has more than two inputs i get:
`checkMotion() missing 1 required positional argument` when calling solve() for the BITStar planner.

Is it possible to obtain that information using the BITStar planner? Or am I just misunderstanding how the library works.

[image]

zkingston, 2022-04-18:
I'm assuming from the error output you provided that you are implementing this using the Python bindings. The internals of OMPL are C++ and expect a `checkMotion` function that takes only two arguments - you cannot define a function that takes in more.

zkingston, 2022-06-21:
Currently this is a bit difficult. I would probably start with a `checkMotion` and `isValid` implementation that hold a reference to the problem definition used by the planner, and then use the [`setIntermediateSolutionCallback`](https://ompl.kavrakilab.org/classompl_1_1base_1_1ProblemDefinition.html#ad5d02593c2c26824ecaf0e29ef40e596) function to have it set the current solution path in a way that is accessible to the validator.

zkingston, 2023-01-25:
Closing this for now.

closed 2023-01-25 by zkingston