ompl/ompl #904: Python argument types in ControlType did not match C++ signature

nnsriram97, 2022-05-19:
Hi, 

I am trying to run a few demo codes after installing ompl along with python bindings and I am facing this error when I ran `RigidBodyPlanningWithControls.py`. The problem only occurs in files where there's control involved and other files like `RigidBodyPlanning.py` work fine.

```
Warning: Assuming propagation will always have between 1 and 10 steps
         at line 77 in /net/ca-home1/home/ma/sriram/path_local/helper_code/ompl/ompl-713be56e5b3ba5742d480160aee1d8d05c9551e0/src/ompl/control/src/SpaceInformation.cpp
Info:    SyclopEST: Starting planning with 1 states already in datastructure
Traceback (most recent call last):
  File "RigidBodyPlanningWithControls.py", line 137, in <module>
    plan()
  File "RigidBodyPlanningWithControls.py", line 130, in plan
    solved = ss.solve(20.0)
  File "RigidBodyPlanningWithControls.py", line 72, in propagate
    state.setX(start.getX() + control[0] * duration * cos(start.getYaw()))
Boost.Python.ArgumentError: Python argument types in
    ControlType.__getitem__(ControlType, int)
did not match C++ signature:
    __getitem__(ompl::control::RealVectorControlSpace::ControlType*, unsigned int)
```

I get an empty dict when I do `print(control.__dict__)` above, indicating the object has no attributes inside. Any thoughts on how to solve this issue?

krzacz0r, 2022-07-25:
Hello,
I've the same problem. 
I would appreciate any working solution in Python.

zkingston, 2023-02-15:
This appears to be working with the latest OMPL - both the minimal example and demo script. Closing unless there is further interest.

closed 2023-02-15 by zkingston