ompl/ompl #907: How to use "modern" PlannerTerminationCoditions?

simonschmeisser, 2022-05-27:
I am experimenting around a bit with how the results obtained from OMPL could be improved. Currently we just use RRTConnect as it has a very high rate of finding a solution within a tolerable time. However the quality of solutions can be quite bad occasionally when it flips the robot to some different configuration. Those cases will then also fail to be simplified significantly.

I would like to have something similar to STOMP's "iterations after collision free". Ie once a solution has been found, invest some finite amount of future iterations in hope of better alternatives and then return the result.

The various PlannerTerminationConditions sound promising but unfortunately I could not get any of the optimizing planners to obey to them. I think it was @henningkayser implementing them, which planner did you use? I looked a bit at RRT* and it seems that it would need to call `addSolutionPath` whenever it calls `intermediateSolutionCallback` for the termination conditions to work? Also it seems that the OptimizationObjective is used in a similar manner here?

henningkayser, 2022-06-24:
just saw this now... I implemented part of it, @mamoll has completed the work and got it merged. If I recall correctly, we have used this with PRM* and AnytimePathShortening successfully. If you look at the CostConvergenceTerminationCondition, you'll see that the callback is being [registered on construction](https://github.com/ompl/ompl/blob/main/src/ompl/base/terminationconditions/src/CostConvergenceTerminationCondition.cpp#L46), and the condition itself doesn't have to depend on the solution path. If `addSolutionPath()` is missing, I'd expect the planner to still terminate while it may not provide the last intermediate solution.

DanyResasco, 2022-10-31:
Hi,
I have the same problem. I can trigger the processNewSolution(). Can you provide an example on how to use CostConvergenceTerminationCondition with prm* ? 

I was trying to use it with 
```
auto convergenceCondition = aidtr_ompl::CostConvergenceTerminationCondition(pdef, 1, 0.1);
 auto terminationCondition = ompl::base::plannerOrTerminationCondition(timeElapsedCondition,
    convergenceCondition); 
ompl::base::PlannerStatus solved = m_omplSimpleSetup->solve(terminationCondition);
```
    
    but the callback is never called even if I use windows = 1

DanyResasco, 2022-10-31:
I think is missing something like this 

```
                    std::vector<const ompl::base::State *> dummy;
                    pdef_->getIntermediateSolutionCallback()(nullptr,dummy,  pathCost);
```
in maybeConstructSolution()

simonschmeisser, 2022-12-11:
I had some time yesterday to play around with this again and found a similar bug to the one fixed in #906 for `IterationTerminationCondition`. With this fix and setting an enormously large timeout (50s) I can get `CostConvergenceTermination` to "work" with `RRT*`. There seems to be no other planner using the callback function `IntermediateSolutionCallback`. For `RRT*` I had to set a relatively small (4-5) `solutionsWindow` as it was computing a long time before finding improved solutions. Often when running 4 threads in parallel via MoveIt, the one that converged was by far not the one giving the best solution as convergence is determined independently in threads.