ompl/ompl #913: Python bindings generation issue in ubuntu 22.04

gabrielpeixoto-cvai, 2022-06-29:
Hello, I am trying to install OMPL with the script **install-ompl-ubuntu.sh **, I can import the library running `import ompl` but if I import utils I get the error `ModuleNotFoundError: No module named 'ompl.util._util'`. I executed the build script with the following parameters:

```
./install-ompl-ubuntu.sh --python
./install-ompl-ubuntu.sh --github --python
```

None of these worked. I also tried the fixes on #609, but it did not work. I also tried to clone and install pyplusplus and pygccxml and reinstalling:

```
cd ompl/build/Release
rm CmakeCache.txt
make ../..
make update_bindings
```

I will upload the output of the **update_bindings**

```
[  0%] Preparing C++ header file for Python binding generation for module tools
[  0%] Built target tools.h
[  0%] Preparing C++ header file for Python binding generation for module control
[  0%] Built target control.h
[  0%] Preparing C++ header file for Python binding generation for module base
[  0%] Built target base.h
[ 50%] Preparing C++ header file for Python binding generation for module util
[ 50%] Built target util.h
[ 50%] Creating C++ code for Python module util (see pyplusplus_util.log)
Traceback (most recent call last):
  File "/home/gabriel/Downloads/ompl/py-bindings/generate_bindings.py", line 950, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/home/gabriel/Downloads/ompl/py-bindings/generate_bindings.py", line 878, in __init__
    code_generator_t.__init__(self, 'util', None, replacement, 1)
  File "/home/gabriel/Downloads/ompl/py-bindings/ompl/bindings_generator.py", line 146, in __init__
    self.filter_declarations()
  File "/home/gabriel/Downloads/ompl/py-bindings/generate_bindings.py", line 893, in filter_declarations
    self.std_ns.class_('vector< std::string >').include()
  File "/home/gabriel/.local/lib/python3.10/site-packages/pygccxml/declarations/scopedef.py", line 554, in class_
    recursive=recursive)
  File "/home/gabriel/.local/lib/python3.10/site-packages/pygccxml/declarations/scopedef.py", line 464, in _find_single
    found = matcher.get_single(decl_matcher, decls, False)
  File "/home/gabriel/.local/lib/python3.10/site-packages/pygccxml/declarations/scopedef.py", line 90, in get_single
    raise runtime_errors.declaration_not_found_t(decl_matcher)
pygccxml.declarations.runtime_errors.declaration_not_found_t: Unable to find declaration. Matcher: [(decl type==class_t) and (name==vector< std::string >)]
-- Build type: Release
-- Checking for module 'libspot'
--   No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS) 
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Code for module base not found; type "make update_bindings"
-- Code for module control not found; type "make update_bindings"
-- Code for module geometric not found; type "make update_bindings"
-- Code for module tools not found; type "make update_bindings"
-- Code for module util not found; type "make update_bindings"
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * pypy, <https://pypy.org>
   Used to speed up the generation of python bindings.
[... 473 lines cut ...]

 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org).

-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/gabriel/Downloads/ompl/build/Release
[100%] Built target update_tools_bindings
[100%] Built target update_bindings

```

Is there any problem with ubuntu 22.04? I hope anyone can help me with this.

Thanks in advance!

gabrielpeixoto-cvai, 2022-06-29:
[UPDATE] I run: `make update_util_bindings VERBOSE=1` from #609 and maybe it has more information on solving my issue.

```
/usr/bin/cmake -S/home/gabriel/Downloads/ompl -B/home/gabriel/Downloads/ompl/build/Release --check-build-system CMakeFiles/Makefile.cmake 0
make  -f CMakeFiles/Makefile2 update_util_bindings
make[1]: Entering directory '/home/gabriel/Downloads/ompl/build/Release'
/usr/bin/cmake -S/home/gabriel/Downloads/ompl -B/home/gabriel/Downloads/ompl/build/Release --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/gabriel/Downloads/ompl/build/Release/CMakeFiles 1
make  -f CMakeFiles/Makefile2 py-bindings/CMakeFiles/update_util_bindings.dir/all
make[2]: Entering directory '/home/gabriel/Downloads/ompl/build/Release'
make  -f py-bindings/CMakeFiles/util.h.dir/build.make py-bindings/CMakeFiles/util.h.dir/depend
make[3]: Entering directory '/home/gabriel/Downloads/ompl/build/Release'
cd /home/gabriel/Downloads/ompl/build/Release && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/gabriel/Downloads/ompl /home/gabriel/Downloads/ompl/py-bindings /home/gabriel/Downloads/ompl/build/Release /home/gabriel/Downloads/ompl/build/Release/py-bindings /home/gabriel/Downloads/ompl/build/Release/py-bindings/CMakeFiles/util.h.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/gabriel/Downloads/ompl/build/Release'
make  -f py-bindings/CMakeFiles/util.h.dir/build.make py-bindings/CMakeFiles/util.h.dir/build
make[3]: Entering directory '/home/gabriel/Downloads/ompl/build/Release'
[100%] Preparing C++ header file for Python binding generation for module util
cd /home/gabriel/Downloads/ompl/build/Release/py-bindings && /usr/bin/cmake -D dir="/home/gabriel/Downloads/ompl/py-bindings/.." -D module=util -D exclude= -P /home/gabriel/Downloads/ompl/CMakeModules/generate_header.cmake
make[3]: Leaving directory '/home/gabriel/Downloads/ompl/build/Release'
[100%] Built target util.h
make  -f py-bindings/CMakeFiles/update_util_bindings.dir/build.make py-bindings/CMakeFiles/update_util_bindings.dir/depend
make[3]: Entering directory '/home/gabriel/Downloads/ompl/build/Release'
cd /home/gabriel/Downloads/ompl/build/Release && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/gabriel/Downloads/ompl /home/gabriel/Downloads/ompl/py-bindings /home/gabriel/Downloads/ompl/build/Release /home/gabriel/Downloads/ompl/build/Release/py-bindings /home/gabriel/Downloads/ompl/build/Release/py-bindings/CMakeFiles/update_util_bindings.dir/DependInfo.cmake --color=
make[3]: Leaving directory '/home/gabriel/Downloads/ompl/build/Release'
make  -f py-bindings/CMakeFiles/update_util_bindings.dir/build.make py-bindings/CMakeFiles/update_util_bindings.dir/build
make[3]: Entering directory '/home/gabriel/Downloads/ompl/build/Release'
[100%] Creating C++ code for Python module util (see pyplusplus_util.log)
cd /home/gabriel/Downloads/ompl/build/Release/py-bindings && env PYTHONPATH="/home/gabriel/.local/lib/python3.10/site-packages:/home/gabriel/.local/lib/python3.10/site-packages:" /usr/bin/pypy3 /home/gabriel/Downloads/ompl/py-bindings/generate_bindings.py util 2>&1 | tee /home/gabriel/Downloads/ompl/build/Release/pyplusplus_util.log
Traceback (most recent call last):
  File "/home/gabriel/Downloads/ompl/py-bindings/generate_bindings.py", line 950, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/home/gabriel/Downloads/ompl/py-bindings/generate_bindings.py", line 878, in __init__
    code_generator_t.__init__(self, 'util', None, replacement, 1)
  File "/home/gabriel/Downloads/ompl/py-bindings/ompl/bindings_generator.py", line 146, in __init__
    self.filter_declarations()
  File "/home/gabriel/Downloads/ompl/py-bindings/generate_bindings.py", line 893, in filter_declarations
    self.std_ns.class_('vector< std::string >').include()
  File "/home/gabriel/.local/lib/python3.10/site-packages/pygccxml/declarations/scopedef.py", line 554, in class_
    recursive=recursive)
  File "/home/gabriel/.local/lib/python3.10/site-packages/pygccxml/declarations/scopedef.py", line 464, in _find_single
    found = matcher.get_single(decl_matcher, decls, False)
  File "/home/gabriel/.local/lib/python3.10/site-packages/pygccxml/declarations/scopedef.py", line 90, in get_single
    raise runtime_errors.declaration_not_found_t(decl_matcher)
pygccxml.declarations.runtime_errors.declaration_not_found_t: Unable to find declaration. Matcher: [(decl type==class_t) and (name==vector< std::string >)]
cd /home/gabriel/Downloads/ompl/build/Release/py-bindings && /usr/bin/cmake /home/gabriel/Downloads/ompl/build/Release
Re-run cmake no build system arguments
-- Build type: Release
<string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
-- Checking for module 'libspot'
--   No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS) 
<string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
-- Code for module base not found; type "make update_bindings"
-- Code for module control not found; type "make update_bindings"
[... cut ...]
 * numpy pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/gabriel/Downloads/ompl/build/Release
make[3]: Leaving directory '/home/gabriel/Downloads/ompl/build/Release'
[100%] Built target update_util_bindings
make[2]: Leaving directory '/home/gabriel/Downloads/ompl/build/Release'
/usr/bin/cmake -E cmake_progress_start /home/gabriel/Downloads/ompl/build/Release/CMakeFiles 0
make[1]: Leaving directory '/home/gabriel/Downloads/ompl/build/Release'

```

zkingston, 2022-06-29:
From a very cursory inspection, it seems that pygccxml (used under the hood to generate bindings) has some incompatibility with Python 3.10 - but appears to work up to Python 3.8. So for now (until a fix or otherwise is done), Python bindings will not work on Ubuntu 22.04

UtkarshMishra04, 2022-10-24:
@zkingston 

Hi

I am facing the same issue on Ubuntu 20.04 and Python 3.8.
How should I proceed?

mamoll, 2022-10-24:
Can you check my fixes in PR #940 (and use PR #941 to create a clean docker build that includes Python bindings on 22.04)?

amberxie88, 2022-11-02:
Following up:
Running into many issues with installing python bindings in my conda env. 

**Attempt 1**: When I try running `install-ompl-ubuntu.sh`, I'm able to call `make update_bindings`, but I get errors like 
```sh 
pygccxml.declarations.runtime_errors.declaration_not_found_t: Unable to find declaration. Matcher: [(decl type==class_t) and (name==ve
ctor< std::string >)]     
 ``` 
in `pyplusplus_util.log` 

and thus later errors in `pyplusplus_geometric.log` : `FileNotFoundError: [Errno 2] No such file or directory: 'bindings/util'`

To try and address this issue, which I think is due to python3.10, I tried changing a few flags for a second attempt:

**Attempt 2**: `cmake ../.. -DOMPL_BUILD_PYBINDINGS=TRUE -DCMAKE_PREFIX_PATH=$CONDA_PREFIX  -DPYTHON_EXEC=/path/to/anaconda3/envs/envname/bin/python3.8`

leads to this error when I try to make bindings `make: *** No rule to make target 'update_bindings'.  Stop.`

Seems like it cannot find Boost for python3.8. 

Is there another flag I should use or another way I need to reinstall boost? I have attached both CMakeCache files.

Attempt 1 CMakeCache.txt:
```sh 
# This is the CMakeCache file.
# For build in directory: /home/usern/ompl-1.5.2/build/Release
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Boost filesystem library (debug)
Boost_FILESYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Boost filesystem library (release)
Boost_FILESYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/include

//Boost library directory DEBUG
Boost_LIBRARY_DIR_DEBUG:PATH=/usr/lib/x86_64-linux-gnu

//Boost library directory RELEASE
Boost_LIBRARY_DIR_RELEASE:PATH=/usr/lib/x86_64-linux-gnu

//Boost numpy310 library (debug)
Boost_NUMPY310_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_numpy310.so

//Boost numpy310 library (release)
Boost_NUMPY310_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_numpy310.so

//Boost numpy38 library (debug)
Boost_NUMPY38_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY38_LIBRARY_DEBUG-NOTFOUND

//Boost numpy38 library (release)
Boost_NUMPY38_LIBRARY_RELEASE:FILEPATH=Boost_NUMPY38_LIBRARY_RELEASE-NOTFOUND

//Boost numpy3 library (debug)
Boost_NUMPY3_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY3_LIBRARY_DEBUG-NOTFOUND

[... 811 lines cut ...]
_Boost_COMPONENTS_SEARCHED:INTERNAL=filesystem;numpy;numpy3;numpy310;numpy38;program_options;python;python-py310;python-py38;python3;python310;python38;serialization;system;unit_test_framework
//Last used Boost_INCLUDE_DIR value.
_Boost_INCLUDE_DIR_LAST:INTERNAL=/usr/include
//Last used Boost_LIBRARY_DIR_DEBUG value.
_Boost_LIBRARY_DIR_DEBUG_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_LIBRARY_DIR_RELEASE value.
_Boost_LIBRARY_DIR_RELEASE_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_NAMESPACE value.
_Boost_NAMESPACE_LAST:INTERNAL=boost
//Last used Boost_USE_MULTITHREADED value.
_Boost_USE_MULTITHREADED_LAST:INTERNAL=ON
//CMAKE_INSTALL_PREFIX during last run
_GNUInstallDirs_LAST_CMAKE_INSTALL_PREFIX:INTERNAL=/usr/local
__pkg_config_arguments_ODE:INTERNAL=ode
__pkg_config_arguments_YAMLCPP:INTERNAL=yaml-cpp
__pkg_config_checked_FLANN:INTERNAL=1
__pkg_config_checked_ODE:INTERNAL=1
__pkg_config_checked_SPOT:INTERNAL=1
__pkg_config_checked_YAMLCPP:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_ODE_ode
pkgcfg_lib_ODE_ode-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_YAMLCPP_yaml-cpp
pkgcfg_lib_YAMLCPP_yaml-cpp-ADVANCED:INTERNAL=1
prefix_result:INTERNAL=/usr/lib/x86_64-linux-gnu
```

amberxie88, 2022-11-02:
Attempt 2 CMakeCache.txt:
```sh # This is the CMakeCache file.
# For build in directory: /home/usern/ompl-1.5.2/build/Release
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Boost filesystem library (debug)
Boost_FILESYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Boost filesystem library (release)
Boost_FILESYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/include

//Boost library directory DEBUG
Boost_LIBRARY_DIR_DEBUG:PATH=/usr/lib/x86_64-linux-gnu

//Boost library directory RELEASE
Boost_LIBRARY_DIR_RELEASE:PATH=/usr/lib/x86_64-linux-gnu

//Boost numpy38 library (debug)
Boost_NUMPY38_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY38_LIBRARY_DEBUG-NOTFOUND

//Boost numpy38 library (release)
Boost_NUMPY38_LIBRARY_RELEASE:FILEPATH=Boost_NUMPY38_LIBRARY_RELEASE-NOTFOUND

//Boost numpy3 library (debug)
Boost_NUMPY3_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY3_LIBRARY_DEBUG-NOTFOUND

//Boost numpy3 library (release)
Boost_NUMPY3_LIBRARY_RELEASE:FILEPATH=Boost_NUMPY3_LIBRARY_RELEASE-NOTFOUND

//Boost numpy library (debug)
Boost_NUMPY_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY_LIBRARY_DEBUG-NOTFOUND

//Boost numpy library (release)
Boost_NUMPY_LIBRARY_RELEASE:FILEPATH=Boost_NUMPY_LIBRARY_RELEASE-NOTFOUND

//Boost program_options library (debug)
Boost_PROGRAM_OPTIONS_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost program_options library (release)
Boost_PROGRAM_OPTIONS_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost python-py38 library (debug)
Boost_PYTHON-PY38_LIBRARY_DEBUG:FILEPATH=Boost_PYTHON-PY38_LIBRARY_DEBUG-NOTFOUND

//Boost python-py38 library (release)
Boost_PYTHON-PY38_LIBRARY_RELEASE:FILEPATH=Boost_PYTHON-PY38_LIBRARY_RELEASE-NOTFOUND

//Boost python38 library (debug)
Boost_PYTHON38_LIBRARY_DEBUG:FILEPATH=Boost_PYTHON38_LIBRARY_DEBUG-NOTFOUND

//Boost python38 library (release)
Boost_PYTHON38_LIBRARY_RELEASE:FILEPATH=Boost_PYTHON38_LIBRARY_RELEASE-NOTFOUND

//Boost python3 library (debug)
Boost_PYTHON3_LIBRARY_DEBUG:FILEPATH=Boost_PYTHON3_LIBRARY_DEBUG-NOTFOUND
[... 744 lines cut ...]
//Components requested for this build tree.
_Boost_COMPONENTS_SEARCHED:INTERNAL=filesystem;numpy;numpy3;numpy38;program_options;python;python-py38;python3;python38;serialization;system;unit_test_framework
//Last used Boost_INCLUDE_DIR value.
_Boost_INCLUDE_DIR_LAST:INTERNAL=/usr/include
//Last used Boost_LIBRARY_DIR_DEBUG value.
_Boost_LIBRARY_DIR_DEBUG_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_LIBRARY_DIR_RELEASE value.
_Boost_LIBRARY_DIR_RELEASE_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_NAMESPACE value.
_Boost_NAMESPACE_LAST:INTERNAL=boost
//Last used Boost_USE_MULTITHREADED value.
_Boost_USE_MULTITHREADED_LAST:INTERNAL=ON
//CMAKE_INSTALL_PREFIX during last run
_GNUInstallDirs_LAST_CMAKE_INSTALL_PREFIX:INTERNAL=/usr/local
__pkg_config_arguments_ODE:INTERNAL=ode
__pkg_config_arguments_YAMLCPP:INTERNAL=yaml-cpp
__pkg_config_checked_FLANN:INTERNAL=1
__pkg_config_checked_ODE:INTERNAL=1
__pkg_config_checked_SPOT:INTERNAL=1
__pkg_config_checked_YAMLCPP:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_ODE_ode
pkgcfg_lib_ODE_ode-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_YAMLCPP_yaml-cpp
pkgcfg_lib_YAMLCPP_yaml-cpp-ADVANCED:INTERNAL=1
prefix_result:INTERNAL=/usr/lib/x86_64-linux-gnu```

zkingston, 2022-11-08:
Please take a look and see if the merge of #940 works

closed 2022-11-08 by zkingston

babbatem, 2022-11-10:
@zkingston the latest commit is still not working on Ubuntu 20.04 with python3.8 (with gcc, clang, etc.)

pygccxml version 2.2.1
pyplusplus version 1.8.5

Any ideas?

`cmake ../.. -DPYTHON_EXEC=/usr/bin/python3
-- The CXX compiler identification is Clang 10.0.0
-- Check for working CXX compiler: /usr/bin/clang
-- Check for working CXX compiler: /usr/bin/clang -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type: Release
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'ode'
--   Found ode, version 0.16
-- Checking for module 'flann'
--   No package 'flann' found
-- Checking for module 'libspot'
--   No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS) 
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Found PY_numpy: /home/abba/.local/lib/python3.8/site-packages/numpy  
-- Found PY_pyplusplus: /home/abba/.local/lib/python3.8/site-packages/pyplusplus  
-- Found PY_pygccxml: /home/abba/.local/lib/python3.8/site-packages/pygccxml  
-- Found castxml: /usr/bin/castxml  
-- Checking for module 'yaml-cpp'
--   Found yaml-cpp, version 0.6.2
-- Found yaml-cpp: yaml-cpp  
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * pypy, <https://pypy.org>
   Used to speed up the generation of python bindings.
 * ODE, <http://ode.org>
   The Open Dynamics Engine can be used as a "black box" for kinodynamic planning.
 * Python, <https://www.python.org>
   Used for python bindings.
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.

-- The following REQUIRED packages have been found:

 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
[... 926 lines cut ...]
 * numpy pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/abba/installations/ompl/build/Release
[ 50%] Built target update_util_bindings
Scanning dependencies of target update_base_bindings
[ 50%] Creating C++ code for Python module base (see pyplusplus_base.log)
Traceback (most recent call last):
  File "/home/abba/installations/ompl/py-bindings/generate_bindings.py", line 963, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/home/abba/installations/ompl/py-bindings/generate_bindings.py", line 154, in __init__
    code_generator_t.__init__(self, 'base', ['bindings/util'], replacement)
  File "/home/abba/installations/ompl/py-bindings/ompl/bindings_generator.py", line 151, in __init__
    self.filter_declarations()
  File "/home/abba/installations/ompl/py-bindings/generate_bindings.py", line 194, in filter_declarations
    self.ompl_ns.class_(f'SpecificParam< {self.string_decl} >').rename('SpecificParamString')
  File "/home/abba/.local/lib/python3.8/site-packages/pygccxml/declarations/scopedef.py", line 554, in class_
    recursive=recursive)
  File "/home/abba/.local/lib/python3.8/site-packages/pygccxml/declarations/scopedef.py", line 464, in _find_single
    found = matcher.get_single(decl_matcher, decls, False)
  File "/home/abba/.local/lib/python3.8/site-packages/pygccxml/declarations/scopedef.py", line 90, in get_single
    raise runtime_errors.declaration_not_found_t(decl_matcher)
pygccxml.declarations.runtime_errors.declaration_not_found_t: Unable to find declaration. Matcher: [(decl type==class_t) and (name==SpecificParam< std::string >)]
`

amberxie88, 2023-02-01:
Still same issue. It doesn't find Boost

zkingston, 2023-02-01:
See if #980 resolves your problem.

amberxie88, 2023-02-01:
The issue is that because of Conda, my boost versions don't match up. My system Python version is 3.10, whereas Conda uses 3.8. I installed boost from source, but I don't know how to to make it so that the CMakeFile finds the correct version (the manually installed) version of boost. I tried manually moving some of the files, but there still seems to be a version mismatch, and I'm not sure what I'm missing.

zkingston, 2023-02-01:
See if you can edit line 52 in the top-level CMakeLists.txt to specify the precise version of boost you have installed, or specify the precise path (see https://stackoverflow.com/a/49816516)

zkingston, 2023-02-01:
Is that an error or just a warning? If it is a warning, it seems to be indicating that it is doing what you want, which is having the local boost overshadow the system boost. 

Also, could you take a look at [this post](https://scaron.info/blog/cmake-can-use-a-local-boost.html) and see if it helps your case?

amberxie88, 2023-02-01:
Thanks! That's a helpful link. I tried some of their attempts but it still seems to be linking to the wrong Boost. I edited my CMakeCache.txt to change all the boost paths to my local paths, and I've reran cmake, but it still doesn't work. It's looking for 1.74.0, an older version I apt installed (and have since removed), whereas my local boost is 1.81.0.

```
$ ldd /usr/local/lib/libompl.so
        linux-vdso.so.1 (0x00007ffea0fe5000)
        libode.so.8 => /lib/x86_64-linux-gnu/libode.so.8 (0x00007f23724bd000)
        libboost_serialization.so.1.74.0 => not found
        libboost_filesystem.so.1.74.0 => not found
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f2372293000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f23721ac000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f237218a000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f2371f62000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f2372e17000)
        libccd.so.2 => /lib/x86_64-linux-gnu/libccd.so.2 (0x00007f2371f56000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2371f51000)
```

My previous message was just a warning, and I'm now getting this warning, which seems related to my present issue:
```
CMake Warning at /usr/share/cmake-3.22/Modules/FindBoost.cmake:1369 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindBoost.cmake:1492 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.22/Modules/FindBoost.cmake:2102 (_Boost_MISSING_DEPENDENCIES)
  CMakeModules/FindPython.cmake:175 (find_package)
  CMakeLists.txt:110 (find_boost_python)
```