ompl/ompl #914: Specifying heuristics in Python

Kchour, 2022-06-29:
By default, the trivial or euclidean distance between two points is used as a heuristic in OMPL and is not very accurate. I want to be able to specify a custom heuristic function that could be called by any of the planners. This seems doable in C++, but I am currently using Python bindings.

Specifically, is there a way to bind a python function to either `costToGoHeuristic` and `motionCostHeuristic`?

zkingston, 2022-06-29:
Take a look at [this example code](http://ompl.kavrakilab.org/OptimalPlanning_8py_source.html) - it has examples of custom cost functions.

closed 2022-06-29 by zkingston

Kchour, 2022-06-29:
I've taken a look at the mentioned page before, but it doesn't answer the second part of my question regarding `motionCostHeuristic`. Additionally, I have a MWE where most of the planners (PRM* or BIT* ) don't end up using a custom heuristic function. In the code below, I defined `customHeuristic` and set it up as follows:
```
def customHeuristic(state, goal):
    print("calling custom heuristic")
    return 0

def setup_example():
    # set up stuff here
 
    # try to bind customHeuristic function
    obj = ob.PathLengthOptimizationObjective(si)
    obj.setCostToGoHeuristic(ob.CostToGoHeuristic(customHeuristic))

    # finish setting up the problem
   
```

Here is the MWE; As a sanity check, I separately tried setting things up using a `SimpleSetup`  and `ProblemDefinition` object:

```
"""Try to use a custom costToGoHeuristic"""
from ompl import base as ob
from ompl import geometric as og
from math import fabs
from time import sleep

def customHeuristic(state, goal):
    print("calling custom heuristic")
    return 0

def isStateValid(state):
     # Let's pretend that the validity check is computationally relatively
     # expensive to emphasize the benefit of explicitly generating valid
     # samples
     sleep(.001)
     # Valid states satisfy the following constraints:
     # -1<= x,y,z <=1
     # if .25 <= z <= .5, then |x|>.8 and |y|>.8
     return not (fabs(state[0] < .8) and fabs(state[1] < .8) and \
         state[2] > .25 and state[2] < .5)

def plan_the_hard_way():
    # construct the state space we are planning in
    space = ob.RealVectorStateSpace(3)

    # get space information
    si = ob.SpaceInformation(space)

    # set the bounds
    bounds = ob.RealVectorBounds(3)
    bounds.setLow(-1)
    bounds.setHigh(1)
    space.setBounds(bounds)

    # set state validity checking for this space
    si.setStateValidityChecker(ob.StateValidityCheckerFn(isStateValid))

    # create a start state
    start = ob.State(space)
    start[0] = 0
    start[1] = 0
    start[2] = 0

    # create a goal state
    goal = ob.State(space)
    goal[0] = 0
    goal[1] = 0
    goal[2] = 1

    pdef = ob.ProblemDefinition(si)
    pdef.setStartAndGoalStates(start, goal)
    

    # try to set an objective with heuristic
    obj = ob.PathLengthOptimizationObjective(si)
    obj.setCostToGoHeuristic(ob.CostToGoHeuristic(customHeuristic))
    pdef.setOptimizationObjective(obj)

    # set planner (uncomment one)
    planner = og.PRMstar(si)
    # planner = og.InformedRRTstar(si)    # only one that calls custom heuristic
    # planner = og.BITstar(si)
    planner.setProblemDefinition(pdef)

    planner.setup()

    # attempt to solve the problem within ten seconds of planning time
    solved = planner.solve(10.0)
    if solved:
        print("Found solution:")
        # print the path to screen
        print(pdef.getSolutionPath())
        print(pdef.getSolutionPath().length())
    else:
        print("No solution found")

def plan_the_easy_way():
    # construct the state space we are planning in
    space = ob.RealVectorStateSpace(3)

    # set the bounds
    bounds = ob.RealVectorBounds(3)
    bounds.setLow(-1)
    bounds.setHigh(1)
    space.setBounds(bounds)

    # define a simple setup class
    ss = og.SimpleSetup(space)

    # t space information
    si = ss.getSpaceInformation()
    # 
    # try to set up a custom heuristic (works with RRTstar)
    obj = ob.PathLengthOptimizationObjective(si)
    obj.setCostToGoHeuristic(ob.CostToGoHeuristic(customHeuristic))
    ss.setOptimizationObjective(obj)

    # set state validity checking for this space
    ss.setStateValidityChecker(ob.StateValidityCheckerFn(isStateValid))

    # create a start state
    start = ob.State(space)
    start[0] = 0
    start[1] = 0
    start[2] = 0

    # create a goal state
    goal = ob.State(space)
    goal[0] = 0
    goal[1] = 0
    goal[2] = 1

    # set the start and goal states;
    ss.setStartAndGoalStates(start, goal)

    # set planner (uncomment one)
    planner = og.PRMstar(si)
    # planner = og.InformedRRTstar(si)    # only one that calls custom heuristic
    # planner = og.BITstar(si)
    ss.setPlanner(planner)

    # attempt to solve the problem within ten seconds of planning time
    solved = ss.solve(10.0)
    if solved:
        print("Found solution:")
        # print the path to screen
        print(ss.getSolutionPath())
        print(ss.getSolutionPath().length())
    else:
        print("No solution found")

plan_the_easy_way()
plan_the_hard_way()
```

marlinstrub, 2022-06-29:
Try defining your own optimization objective that derives from the optimization objective base class. In your objective, override the `motionCostHeuristic` method. This should get picked up by PRM* and BIT*.

Kchour, 2022-07-13:
I was able to get my own `motionCostHeuristic` method working thanks to your suggestion. However, I now find BIT* to converge to the wrong result. Just wanted to confirm if this method is the right one to override; my gut feeling says yes since the `CostHelper` class defines `CostToGoHeuristic` using the `motionCostHeuristic` internally (as seen here, Lines 201-202). But maybe there is some unintended side-effect of doing this, which may be why I'm getting wrong results. Any suggestions?