ompl/ompl #916: undefined reference to EITstart

yw1055, 2022-07-01:
CMakeFiles/demo_OptimalPlanning.dir/OptimalPlanning.cpp.o: In function `allocatePlanner(std::shared_ptr<ompl::base::SpaceInformation>, optimalPlanner)':
OptimalPlanning.cpp:(.text+0x4b4): undefined reference to `ompl::geometric::EITstar::EITstar(std::shared_ptr<ompl::base::SpaceInformation> const&)'
collect2: error: ld returned 1 exit status
demos/CMakeFiles/demo_OptimalPlanning.dir/build.make:101: recipe for target 'bin/demo_OptimalPlanning' failed
make[2]: *** [bin/demo_OptimalPlanning] Error 1
CMakeFiles/Makefile2:1883: recipe for target 'demos/CMakeFiles/demo_OptimalPlanning.dir/all' failed
make[1]: *** [demos/CMakeFiles/demo_OptimalPlanning.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
Hi whom may concerns,
   Do you have any suggestions to solve this problem?

marlinstrub, 2022-07-02:
Just to confirm you are on the `robotic-esp:planner/eitstar` branch, right? EIT* hasn't been merged into `main` yet.

yw1055, 2022-07-02:
Hi Marlin,
  Thanks for your reply. I was not on the robotic-esp:planner/eitstar acutally. I tried to merge the EIT* to OMPL. But  everything works well on robotic-esp:planner.
Thank you,
Yi