ompl/ompl #918: BVH Error! endModel() called on model with no triangles and vertices.

MohiniMohiniMohini, 2022-07-04:
Getting this error for RRT planner setup() method. I'm using KinematicCarPlanning object. After setPlanner(); I'm calling setup().

Detailed output:
BVH Error! endModel() called on model with no triangles and vertices.
Info:    Empty environment loaded
Warning: Assuming propagation will always have between 1 and 10 steps
         at line 77 in /home/admin1/Downloads/ompl-1.5.2/src/ompl/control/src/SpaceInformation.cpp
Warning: The propagation step size is assumed to be 20.104516

ChamzasKonstantinos, 2022-07-04:
As the error indicates you probably did not load an environment. You could try using some of OMPL.app environments

closed 2022-07-04 by ChamzasKonstantinos

MohiniMohiniMohini, 2022-07-05:
Can you share an example link to do that?

currently I'm referring to demos/SE2RigidBodyPlanning/KinematicCarPlanning.cpp example. I am not able to understand where is the environment loaded in this example.

Also, if i set the setCellSizes(), as per the example, the error seems to away. but I need to call registerDefaultProjector(). So a link to relevant example will help.

My aim is to move a car from one point to another with RRT*. 
1. I have implemented RRT* with OMPL (geometric namespace). That works but it does not consider car dimensions, is that correct? If no, can you help me setting car dimensions in the geometric namespace with RRT*?
2. I noticed RRT* is not implemented for control namespace. Is that correct?
3. So I am planning to implement RRT with car dimensions with control namespace, currently stuck with loading the environment issue.

ChamzasKonstantinos, 2022-07-05:
Hi @MohiniMohiniMohini  you should be able to load an environment and a mesh as shown in this python example: https://github.com/ompl/omplapp/blob/main/demos/SE2RigidBodyPlanning/KinematicCarPlanning.py

The same interface exist in cpp so it should work as well. 

However, if you are not using a projectionBasedPlanner such as Syclop or KPIECE you don't need to call the registerDefaultProjection. If you do want to have dimentions of the car though you should ate least load a robotMesh albeit in an empty environment. 

1. If you do not want to consider non-holonomic constraints, e.g. SE2 planning you can have "dimentions" by overloading the collision function 
2. RRT* cannot be implemented for the control space, since you would have to solve to 2 point boundary problem. Only SST* works for control statespaces. In other words you need a steering function, and a kinematicCar does not have one
3. I would suggest you go over the omplApp examples in python https://github.com/ompl/omplapp to experiment a bit with different planners and understand what works where. If you load the graphical interface you should be able to run a kinematicCar example

MohiniMohiniMohini, 2022-07-05:
Hello @ChamzasKonstantinos, Thanks for the reply.

I followed [](https://github.com/ompl/omplapp/blob/main/demos/SE2RigidBodyPlanning/KinematicCarPlanning.cpp) this link which is equivalent to the py file you shared. I do follow the .py implementation, but it is not the same in .cpp. that's why I was stuck. Why are the following methods not called from .ccp file?
setup.setRobotMesh("2D/car2_planar_robot.dae")
setup.setEnvironmentMesh("2D/Maze_planar_env.dae")

1. Let me try overload the collision method.
2.  That makes sense
3. I did not try the graphical interface. Getting installation issues even with Docker. (that's another topic)