ompl/ompl #919: Compile a demo cpp file on Ubuntu using Visual Studio

PhilWe-01, 2022-07-05:
Hello guys,

I have installed OMPL and OMPL.app with python bindings from Source on Ubuntu 18.04.6 using the VMware Workstation 16 Player. [https://ompl.kavrakilab.org/installation.html](url)
Visual Studio Code version 1.68.1 is used as IDE.
After the installation I opened the folder with VS and I tried to generate and start a banchmark by following the instructions on the homepage.  [https://ompl.kavrakilab.org/benchmark.html](url)
After that I could launch the gui with compiling ompl_app.py. I could also start the demo RigidBodyPlanning by starting the executable with the Terminal or by compiling the python script RigidBodyPlanning.py.
The problem is, that I couldn't compile the cpp codes with VS. I suppose there is a problem with the compiler or linker, but i don't know where.

I attached the error message text file, the tasks.json file and the RigidBodyPlanning.cpp file below.

Thanks in advance for your help.

#275
[error_message.txt](https://github.com/ompl/ompl/files/9045850/error_message.txt)
[tasks.json.txt](https://github.com/ompl/ompl/files/9045869/tasks.json.txt)
[RigidBodyPlanning.cpp.txt](https://github.com/ompl/ompl/files/9045876/RigidBodyPlanning.cpp.txt)

mamoll, 2022-07-19:
Do you happen to have more than one version of OMPL installed?

PhilWe-01, 2022-07-25:
I could solve the issue.
The problem was the linker. It used the wrong shared library.

closed 2022-07-25 by PhilWe-01