ompl/ompl #920: how to set bounds for kinematic car planning

MohiniMohiniMohini, 2022-07-07:
Hello,

I'm using [KinematicCarPlanningClass](http://ompl.kavrakilab.org/classompl_1_1app_1_1KinematicCarPlanning.html) in CPP language. I need to use RRT algorithm with turning radius (assuming I can use the vehicle length as turning radius). If the above assumption is correct, I am not able to set bounds for the environment. Here is the code for the same:

    // create an instance of a car.
    app::KinematicCarPlanning l_regularCar;

    // plan for kinematic car in SE(2)
    ob::StateSpacePtr l_SE2_state_space(l_regularCar.getStateSpace());

    // set the bounds for the R^2 part of SE(2)
    ob::RealVectorBounds l_ego_bounds(2);
    l_ego_bounds.setLow(0, 0);     // set min  x
    l_ego_bounds.setHigh(0, 199); // set max x
    l_ego_bounds.setLow(1, 0);     // set min  y
    l_ego_bounds.setHigh(1, 99);  // set max y
    l_SE2_state_space->as<ob::SE2StateSpace>()->setBounds(l_ego_bounds);

the last line of code does not seem to have any effect as I get an error:
Info: Empty environment loaded

referring this file as [sample](https://github.com/ompl/omplapp/blob/main/demos/SE2RigidBodyPlanning/KinematicCarPlanning.cpp)

zkingston, 2022-07-08:
That message is reporting that no collision environment is loaded. See [SE2RigidBodyPlanning](https://github.com/ompl/omplapp/blob/main/demos/SE2RigidBodyPlanning/SE2RigidBodyPlanning.cpp) for an example of how to set environment and robot meshes.

closed 2022-07-08 by zkingston

MohiniMohiniMohini, 2022-07-11:
I have my own environment and obstacles defined within the bounds (0,199 and 0,99). I do not need the predefined/given environment. I want the car to navigate on the custom environment defined by me. How do I do that?