ompl/ompl #922: RRTConnect with setCostThreshold

israfujta, 2022-07-19:
Hello,

i try to solve a pdef with RRTConnect as Planner.
I set the threshold for the solution as can be seen here: https://ompl.kavrakilab.org/optimalPlanningTutorial.html
with "getThresholdPathLengthObj".
Unfortunately it does not work, i can find solutions but they do not fullfill the threshold. 
Even when they fullfill the threshold by random "pdef->hasOptimizedSolution()" says that it has not.
Can someone help with this issue?
Thanks in advance

israfujta, 2022-07-19:
ob::OptimizationObjectivePtr getThresholdPathLengthObj(const ob::SpaceInformationPtr& si)
{
    ob::OptimizationObjectivePtr obj(new ob::PathLengthOptimizationObjective(si));
    obj->setCostThreshold(ob::Cost(20.0));
    return obj;
};

 // Create a problem instance
    ob::ProblemDefinitionPtr pdef(new ob::ProblemDefinition(si));

    // Set the start and goal states
    pdef->setStartAndGoalStates(start, goal);

    // set optimization objective : Pathlength Objective
    pdef->setOptimizationObjective(getThresholdPathLengthObj(si));

    ob::PlannerPtr optimizingPlanner(new og::RRTConnect(si));

    // Set the problem instance for our planner to solve
    optimizingPlanner->setProblemDefinition(pdef);
    optimizingPlanner->setup();

    // attempt to solve the planning problem within one second of
    // planning time

    ob::PlannerStatus solved = optimizingPlanner->solve(1.0);

    ob::PathPtr path = pdef->getSolutionPath();

    std::cout << "\npdef solution cost threshold is " << pdef->hasOptimizedSolution() << "\n";
    std::cout << "\nlength of solution path is " << path->length() << "\n";

mamoll, 2022-07-19:
RRTConnect is not an optimizing planner, so it will ignore the optimization objective. It _should_ closely obey the time limit passed to `solve()`. If it doesn't find a solution, the call to `solve()` will be just slightly longer than that limit. If it exceeds the limit by more than, say, .1 seconds, please create a new issue.