ompl/ompl #929: If PRM* fails, then we plan again --> segfault

AndyZe, 2022-08-02:
I'm using MoveIt to make a simple motion plan including orientation constraints. Usually the first motion plan fails (which is fine). On the second call to `plan()` I get a segfault.

To reproduce this issue with MoveIt, you can check out these 2 branches (ROS2 Rolling or Humble):

moveit2_tutorials:  https://github.com/AndyZe/moveit2_tutorials/tree/andyz/prm_twice_bug
moveit_resources:  https://github.com/AndyZe/moveit_resources/tree/andyz/prm_twice_bug

Run it with:  `ros2 launch moveit2_tutorials ompl_constrained_planning.launch.py`

@stephanie-eng knows a lot more about the details.

The stack trace is very long so I'll attach it.

stephanie-eng, 2022-08-02:
Related PR: #928

stephanie-eng, 2022-08-03:
I tried testing @AndyZe's branches and was able to reproduce the issue. After trying with #928 I no longer get the segfault!

AndyZe, 2022-08-08:
Closed by #928

closed 2022-08-08 by AndyZe