ompl/ompl #930: If PRM* fails, then we plan again --> segfault in StateValidityChecker

stephanie-eng, 2022-08-02:
Similar to #929, when using MoveIt to plan using orientation constraints. Upon replanning, I get a different segfault than #929.

Stack trace:
```
#0  ompl_interface::ConstrainedPlanningStateValidityChecker::isValid (
    this=0x7fffc4272630, wrapped_state=0x7fffc5713d70, verbose=false)
    at /workspaces/miso-ros2-ws2/src/moveit2/moveit_planners/ompl/ompl_interface/src/detail/state_validity_checker.cpp:236
#1  0x00007fffccea31c5 in ompl_interface::StateValidityChecker::isValid (
    this=0x7fffc4272630, state=0x7fffc5713d70)
    at /workspaces/miso-ros2-ws2/src/moveit2/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/detail/state_validity_checker.h:67
#2  0x00007fffce6a9dcd in ompl::base::ProjectedStateSpace::discreteGeodesic (
    this=0x7fffc570fe40, from=0x7fffc56f2230, to=0x7fffc571cf30, 
    interpolate=false, geodesic=0x0)
    at /workspaces/miso-ros2-ws2/src/ompl/src/ompl/base/spaces/constraint/src/ProjectedStateSpace.cpp:106
#3  0x00007fffce6a5886 in ompl::base::ConstrainedMotionValidator::checkMotion (
    this=0x7fffc42729e0, s1=0x7fffc56f2230, s2=0x7fffc571cf30)
    at /workspaces/miso-ros2-ws2/src/ompl/src/ompl/base/spaces/constraint/src/ConstrainedStateSpace.cpp:57
#4  0x00007fffce79ffe3 in ompl::base::SpaceInformation::checkMotion (
    this=0x5555557da240, s1=0x7fffc56f2230, s2=0x7fffc571cf30)
    at /workspaces/miso-ros2-ws2/src/ompl/src/ompl/base/SpaceInformation.h:354
#5  0x00007fffceaabdaa in ompl::geometric::PRM::addMilestone (
    this=0x7fffc571afd0, state=0x7fffc571cf30)
    at /workspaces/miso-ros2-ws2/src/ompl/src/ompl/geometric/planners/prm/src/PRM.cpp:582
#6  0x00007fffceaaaf67 in ompl::geometric::PRM::solve (this=0x7fffc571afd0, ptc=...)
    at /workspaces/miso-ros2-ws2/src/ompl/src/ompl/geometric/planners/prm/src/PRM.cpp:459
#7  0x00007fffced130bc in ompl::geometric::SimpleSetup::solve (this=0x7fffc41f7330, ptc=...)
    at /workspaces/miso-ros2-ws2/src/ompl/src/ompl/geometric/src/SimpleSetup.cpp:142
#8  0x00007fffcce1a443 in ompl_interface::ModelBasedPlanningContext::solve (this=0x7fffc42aff10, timeout=2, count=1)
    at /workspaces/miso-ros2-ws2/src/moveit2/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:853
#9  0x00007fffcce196cc in ompl_interface::ModelBasedPlanningContext::solve (this=0x7fffc42aff10, res=...)
    at /workspaces/miso-ros2-ws2/src/moveit2/moveit_planners/ompl/ompl_interface/src/model_based_planning_context.cpp:762
#10 0x00007ffff4e88397 in planning_request_adapter::(anonymous namespace)::callPlannerInterfaceSolve (planner=..., 
    planning_scene=std::shared_ptr<const planning_scene::PlanningScene> (use count 6, weak count 1) = {...}, req=..., res=...)
    #10 0x00007ffff4e88397 in planning_request_adapter::(anonymous namespace)::callPlannerInterfaceSolve (planner=..., 
    planning_scene=std::shared_ptr<const planning_scene::PlanningScene> (use count 6, weak count 1) = {...}, req=..., res=...)
    at /workspaces/ros2-ws2/src/moveit2/moveit_core/planning_request_adapter/src/planning_request_adapter.cpp:55
```

Tracing through with GDB, I noticed that StateValidityChecker->planning_context_ seemed to be invalid. Checking the addresses, the planning_context_ pointer does not seem to be updated between the first planning attempt and the second planning attempt.

stephanie-eng, 2022-08-02:
I will try to provide a minimal setup based off of the branches in #929

AndyZe, 2022-08-07:
Here's an easy way to reproduce this if you have MoveIt2 installed (from source, on ROS2 Rolling):

Check out these branches:
- moveit2_tutorials: https://github.com/AndyZe/moveit2_tutorials/tree/andyz/prm_twice_bug
- moveit_resources: https://github.com/AndyZe/moveit_resources/tree/andyz/prm_twice_bug
- Clone [this branch](https://github.com/tylerjw/ompl/tree/fix-segfault) of OMPL into your workspace. This patches a _different_ bug so this one can manifest itself.

Run it with: `ros2 launch moveit2_tutorials ompl_constrained_planning.launch.py`

Then click Next in RViz to step through the demo, until the crash happens.

tylerjw, 2022-08-15:
To make this more clear.

On this line in ompl_interface: https://github.com/ros-planning/moveit2/blob/1d67b519e6ef9ca1ebba494743791da998b72950/moveit_planners/ompl/ompl_interface/src/detail/state_validity_checker.cpp#L236

A segfault accurs because the `planning_context_` pointer is not valid (points to memory that is not a `ModelBasedPlanningContext`. Here is the line in the header where that raw pointer is defined: https://github.com/ros-planning/moveit2/blob/main/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/detail/state_validity_checker.h#L90

That pointer is set by OMPL to point at an object inside OMPL itself when it constructs the StateValidityChecker object itself: https://github.com/ros-planning/moveit2/blob/1d67b519e6ef9ca1ebba494743791da998b72950/moveit_planners/ompl/ompl_interface/src/detail/state_validity_checker.cpp#L49

In the most basic terms the problem is **the lifetime of planning_context_ which is owned by a `ompl_interface::StateValidityChecker` is shorter than the `ompl_interface::StateValidityChecker` itself.

tylerjw, 2022-08-15:
I believe this to be a class of problem that can be attributed to an underspecified interface. What I mean by that is nothing in the code links the lifetime of `ModelBasedPlanningContext` to `ompl_interface::StateValidityChecker` even though their lifetimes are dependent on each other. If OMPL is to be trusted by users it should not expose underspecified interfaces.

zkingston, 2022-08-23:
I do agree that there are many sharp corners on the OMPL interface as is that lead to these kinds of problems, but this appears to be an issue with the implementation of the OMPL interface in MoveIt, and not necessarily an issue with OMPL itself. Please open an issue on the MoveIt repository instead.

closed 2022-08-23 by zkingston

tylerjw, 2022-08-23:
@zkingston Sharp corners that invite seg-faults are an issue with OMPL itself.