ompl/ompl #931: How to set seed for Random number generator in RealVectorStateSpace

samarthjain96, 2022-08-03:
Hi, I am trying to use RealVectorStateSpace to define configration space of my robot. I need to set the seed for random samples as I plan to integrate it with PRM. Is there a way to access the RNG object and change seed?

dt1729, 2022-08-22:
Hi Samarth,
Have a look at [this](https://ompl.kavrakilab.org/OptimalPlanning_8cpp_source.html) example, it uses realVectorStateSpace and PRM*, maybe this will help you in your pursuit. As for RNG, OMPL does provide a Random number generator [class](https://ompl.kavrakilab.org/classompl_1_1RNG.html), if you want to write from scratch. A more detailed explanation of your goal, approach and problem will help me give some direct answers.
Let me know if this works.

zkingston, 2023-02-27:
As mentioned, see the [`setSeed()`](https://ompl.kavrakilab.org/classompl_1_1RNG.html#a5a2d71d1c12f13707457c9461807ee2c) function for `RNG`, if you are using the default sampler this should work.

closed 2023-02-27 by zkingston