ompl/ompl #933: What do I cite if I use PathSimplifier?

werner291, 2022-08-22:
Hello,

more of a citation-related question than anything implementation-related...

I noticed that the use of `PathSimplifier::perturbPath` and `PathSimplifier::shortcutPath` greatly improve the results of my experiments, but I'm not too sure about who or what to cite for the algorithms used in these methods.

Are these published in literature somewhere, or should I just cite OMPL itself for these?

zkingston, 2022-08-23:
I'd cite the following:
Geraerts, Roland, and Mark H. Overmars. "Creating high-quality paths for motion planning." The international journal of robotics research 26.8 (2007): 845-863.

closed 2022-08-23 by zkingston

werner291, 2022-08-24:
Actually, that one presents shortcutting only, not random perturbation... I'll see if I can find one for that too.

werner291, 2022-08-24:
There's a version of the random displacement algorithm published as part of _Task Constrained Motion Planning in Robot Joint Space_ by Mike Stilman too