ompl/ompl #944: Compilation failed with undefined reference

skywoodsz, 2022-09-24:
Hi, I wang to install OMPL on ubuntu2004. I  run the _install-ompl-ubuntu.sh_ according to the instructions on the official website. But I get error as follow:
`/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dBodyGetQuaternion(dxBody*)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dGeomGetAABB(dxGeom*, double*)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dBodyGetLinearVel(dxBody*)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dGeomGetBody(dxGeom*)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dBodySetPosition(dxBody*, double, double, double)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dJointCreateContact(dxWorld*, dxJointGroup*, dContact const*)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dAreConnectedExcluding(dxBody*, dxBody*, int)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dJointAttach(dxJoint*, dxBody*, dxBody*)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dBodySetLinearVel(dxBody*, double, double, double)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dCollide(dxGeom*, dxGeom*, int, dContactGeom*, int)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dBodyGetPosition(dxBody*)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dGeomIsSpace(dxGeom*)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dWorldQuickStep(dxWorld*, double)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dBodyGetAngularVel(dxBody*)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dSpaceCollide(dxSpace*, void*, void (*)(void*, dxGeom*, dxGeom*))'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dBodySetQuaternion(dxBody*, double const*)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dBodySetAngularVel(dxBody*, double, double, double)'
/usr/bin/ld: ../lib/libompl.so.1.5.2: undefined reference to `dJointGroupEmpty(dxJointGroup*)'
collect2: error: ld returned 1 exit status`

I tried both on 1.5.0, 1.4.2, but got the same error.
Can you give me some advice to solve this problem？

zkingston, 2023-01-19:
This appears to be an issue with OpenDE being missing from linking, can you verify if OpenDE is installed correctly on your machine?

zkingston, 2023-02-13:
Closing due to age - if you are still interested in this issue let us know.

dbdxnuliba, 2023-05-08:
@zkingston how to install OpenDE in ubuntu20.04