ompl/ompl #946: Problem with path planning using multiple goal state with RRTconnect in Dubin's state space

krakjack5, 2022-09-24:
I am experimenting with planning paths using RRTconnect in Dubin's state space, and have been able to plan paths from a given starting state to the goal in the presence of obstacles. Now I am trying to use the GoalStates class to specify multiple goals, but somehow it is not working as expected. I have attached the source code I used.

State-space: Dubin's
Planner: RRTconnect
Start: 1 start state
Goal: 3 goal states in GoalStates

This is how I have set up the problem. The first goal state is (x1, y1, yaw1), and there is an obstacle very near to (x1, y1) position such that approach at yaw1 is blocked. The two other states are (x1, y1, yaw2) and (x1, y1, yaw3). Thus all the goal states have the same x and y coordinates, but different yaw angles. The two yaw angles, yaw2 and yaw3, are such that it is possible to compute paths from the start state to (x1, y1, yaw2) and (x1, y1, yaw3). I use the GoalStates object to add the three goal states. All the setup is done using SimpleSetup.

I thought that the RRTconnect planner would sample all the three goal states and would be able to plan a path to one of the two feasible goal states. However, it seems that only the first goal state was ever sampled. In the debugger I checked the goal details in the SimpleSetup object. It showed three goal states added, but the samplePosition_ remained unchanged at 1, indicating that only the first goal state was sampled. The console output also shows that there ever was one goal state used in planning, which leads to the planner not being able to compute a path as the yaw angle of the first goal state makes any approach to its x-y possible impossible. In the console output below, we can see the tree growing only from the start state but not from the goal as only the infeasible goal state is sampled. The type of solution after each call to solve remains as APPROXIMATE solution.

I am not sure why I am seeing this behavior, and I will appreciate any input!

Source code used to reproduce the issue: [ompl_goal_states_test.cpp.txt](https://github.com/ompl/ompl/files/9639256/ompl_goal_states_test.cpp.txt)

====== PROGRAM OUTPUT START ======
3 goal states, threshold = 2.22045e-16, memory address = 0x7ff58d706e08
Compound state [
RealVectorState [50 50]
SO2State [0]
]

Compound state [
RealVectorState [50 50]
SO2State [1.0472]
]

Compound state [
RealVectorState [50 50]
SO2State [1.5708]
]

Debug:   RRTConnect: Planner range detected to be 28.598431
Info:    RRTConnect: Starting planning with 1 states already in datastructure
Info:    RRTConnect: Created 9347 states (9346 start + 1 goal)
Info:    ProblemDefinition: Adding approximate solution from planner RRTConnect
Info:    Solution found in 1.005351 seconds
Here is the code I used for testing planning with multiple goal states.
====== PROGRAM OUTPUT End ======

krakjack5, 2022-09-30:
I looked through RRTConnect code, and think that the if condition on line 240 of RRTConnect.cpp that is used to sample goal states is what is causing failure to sample all goal states.

`if (tGoal_->size() == 0 || pis_.getSampledGoalsCount() < tGoal_->size() / 2)`

[RRTConnect.cpp](https://github.com/ompl/ompl/blob/9c3a20faaddfcd7f58fce235495f043ebee3e735/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp)

That if condition means that unless the very first sampled goal state can be connected to some other state, no other goal states will be sampled. In the example I am working with, the first goal state is sitting extremely close to a large obstacle. Thus the likelihood of sampling a state between this goal state and the obstacle that can be connected to the the goal state is extremely low. As such, a tree cannot grow from the first goal state and no other goal states get sampled, and thus no solution is generated.

I changed the if condition to the if condition used in RRTstar and that made RRTConnect work. This is how my change looks like

`if (pis_.getSampledGoalsCount() < goal->maxSampleCount()) {`

I would appreciate any input on this possible solution. Thank you for your help!

zkingston, 2023-01-19:
The goal sampling conditional is to prevent adding too many goal samples to the goal tree - if the goal sampler is fast, it may pollute the goal tree and remove the benefit of using a goal tree in RRTConnect. Perhaps the solution would be to count the number of added goals, and check if `tGoal_->size() == # of goals sampled`, indicating if no other states have been added.