ompl/ompl #949: Segfault in ABIT*: graph reduced to zero or one vertex

andreas-botbuilt, 2022-10-11:
ABIT* sometimes segfaults because the graph is reduced to zero vertices (or one). Here's the terminal output for zero vertices (shortened down for clarity):

```
informedtrees/src/BITstar.cpp:993 - robot[ABITstar]: Finished with a solution of 
cost 6.7633 (2 vertices) found from 73 samples by processing 162 edges (135 collision checked) to 
create 18 vertices and perform 0 rewirings. The final graph has 0 vertices.
BITstar.cpp:753 - curVertex: vertex = goal, depth: 0
Stack trace (most recent call last) in thread 1374030:
#31   Object "", at 0xffffffffffffffff, in 
#30  Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x7f46cd0139ff]
```
It seems this is caused by pruning:
```
BITstar.cpp:993 - robot[ABITstar]: Finished with a solution of cost 6.7633 (2 vertices) found from 73 samples by processing 162 edges (135 collision checked) to create 18 vertices and perform 0 rewirings. The final graph has 0 vertices.
```

And it can be fixed by setting `use_graph_pruning` in ABIT* to `false`.

marlinstrub, 2022-10-11:
Thanks for reporting, I will look into this. What's the optimization objective and heuristic cost you are using when this occurs?

andreas-botbuilt, 2022-10-11:
Just the default values. So, the optimization objective is path length and the heuristic cost should be the zero-heuristic.

marlinstrub, 2022-10-22:
I can't reproduce this... Can you please share more info, ideally a minimal example of where this shows up?

marlinstrub, 2023-01-20:
@andreas-botbuilt I still have no idea what could cause this. If you cannot provide an MWE, could you tell me which version/commit you are using?

andreas-botbuilt, 2023-01-21:
Thank you for looking into this again. Unfortunately, I can't easily provide an MWE (because it's code used in a company) but I'll think about it and see if I can come up with something. 

This should be the commit we are using: https://github.com/ompl/ompl/commit/eac678788183a4c715d425c1b47da47ab6021c81.

Hugal31, 2023-07-05:
I have the same issue using ros-noetic-ompl 1.6.0-1focal.20230117.010101. Disabling pruning prevents the bug.

`drop_unconnected_samples_on_prune` is set to false.

```
[ INFO] [1688570593.835431040]: arm_group/arm_group[kABITstar]: Pruning disconnected 0 vertices from the tree and completely removed 100 samples.
[... same line around 100 times ...]
[ INFO] [1688570595.750460369]: arm_group/arm_group[kABITstar]: Pruning disconnected 0 vertices from the tree and completely removed 100 samples.
[ INFO] [1688570595.768262849]: arm_group/arm_group[kABITstar]: Pruning disconnected 0 vertices from the tree and completely removed 100 samples.
[ INFO] [1688570595.784840932]: arm_group/arm_group[kABITstar]: Finished with a solution of cost 2.9668 (2 vertices) found from 10365 samples by processing 3 edges (1 collision checked) to create 2 vertices and perform 0 rewirings. The final graph has 0 vertices.
[move_group-4] process has died [pid 131103, exit code -11, cmd /opt/ros/noetic/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/laloge_h/.ros/log/8c8716b2-1b04-11ee-8a21-97fbf7b3aad4/move_group-4.log].
```

Sorry, it's a bit difficult to have a minimum example, I am working with MoveIt with a proprietary robot. I can provide the backtrace though:

```
#0  0x00007f67a3cbcf84 in ompl::geometric::BITstar::Vertex::state() const () at ./src/ompl/geometric/planners/informedtrees/bitstar/src/Vertex.cpp:142
#1  0x00007f67a3cd07e0 in ompl::geometric::BITstar::bestPathFromGoalToStart (this=<optimized out>) at /usr/include/c++/9/bits/shared_ptr_base.h:1020
#2  0x00007f67a3cd17b5 in ompl::geometric::BITstar::publishSolution (this=0x7f679804bfa0) at ./src/ompl/geometric/planners/informedtrees/src/BITstar.cpp:712
#3  0x00007f67a3cd1ee8 in ompl::geometric::BITstar::solve (this=0x7f679804bfa0, ptc=...) at ./src/ompl/geometric/planners/informedtrees/src/BITstar.cpp:381
#4  0x00007f67a3e4b7b3 in ompl::geometric::SimpleSetup::solve (this=0x7f67980312e0, ptc=...) at /usr/include/c++/9/bits/shared_ptr_base.h:1020
#5  0x00007f67a83c7c12 in ompl_interface::ModelBasedPlanningContext::solve (this=0x7f6798035a70, timeout=2, count=0x1) at /usr/include/c++/9/bits/shared_ptr_base.h:1020
[...]
```

marlinstrub, 2023-07-05:
Thank you for confirming that this is still an issue. I'll try to find some time over the weekend to take another stab at it.

andreas-botbuilt, 2023-07-05:
We changed it so that [the condition for pruning a vertex](https://github.com/ompl/ompl/blob/main/src/ompl/geometric/planners/informedtrees/bitstar/src/ImplicitGraph.cpp#L1257-L1258)  in `canSampleBePruned` is a strict comparison: `isCostWorseThan` instead of `isCostWorseThanOrEqualTo`.

marlinstrub, 2023-07-10:
Thanks for sharing your fix @andreas-botbuilt, that's definitely better than segfaulting!

I think I have found the underlying issue, but I still can't easily reproduce the error. So if either (or both) of you could try #1070, that would be much appreciated!

Some background: In (A)BIT*-implementation-lingo, samples refer to sampled states that are not in the search tree. For these samples, `isCostWorseThanOrEquivalentTo` is the right criterium, as they should be pruned if they are not in the informed set. The problem is that `canSampleBePruned` is applied to goals even when they are in the search tree (and part of the current solution), if 1) there are multiple goals, 2) the direct connection between a start and that goal is valid, and 3) the edge-cost heuristic is perfectly accurate. All three conditions seem plausible for your use-cases, so I'm hoping this is the root cause for the error.

Thanks again for bringing this up.

Hugal31, 2023-07-10:
That seems to fix it. Thanks a lot!

andreas-botbuilt, 2023-07-10:
Thank for looking into this issue @marlinstrub. [1070](https://github.com/ompl/ompl/pull/1070) fixes it for me as well.

PR #1070 (2023-07-10): Only prune goal in BIT* if it is not part of the current best solution
This hopefully fixes #949.

The errors reported in #949 suggest that the start and goal of a trivial solution (direct start to goal connection) are sometimes pruned. If there are multiple goals and if the heuristic is perfect for a trivial solution (e.g., Euclidean distance heuristic for path length objective in an environment that allows for the trivial solution), then the current code will indeed prune the start and goal. The proposed change directly prevents this by checking the cost of the goal and by adding previously  [... cut]
closed 2023-07-10 by zkingston