ompl/ompl #951: Python Bindings for Ubuntu 22.04 not working

amberxie88, 2022-11-02:
Hi,

Running into many issues with installing python bindings in my conda env. 

**Attempt 1**: When I try running `install-ompl-ubuntu.sh`, I'm able to call `make update_bindings`, but I get errors like 
```sh 
pygccxml.declarations.runtime_errors.declaration_not_found_t: Unable to find declaration. Matcher: [(decl type==class_t) and (name==ve
ctor< std::string >)]     
 ``` 
in `pyplusplus_util.log` 

and thus later errors in `pyplusplus_geometric.log` : `FileNotFoundError: [Errno 2] No such file or directory: 'bindings/util'`

To try and address this issue, which I think is due to python3.10, I tried changing a few flags for a second attempt:

**Attempt 2**: `cmake ../.. -DOMPL_BUILD_PYBINDINGS=TRUE -DCMAKE_PREFIX_PATH=$CONDA_PREFIX  -DPYTHON_EXEC=/path/to/anaconda3/envs/envname/bin/python3.8`

leads to this error when I try to make bindings `make: *** No rule to make target 'update_bindings'.  Stop.`

Seems like it cannot find Boost for python3.8. 

Is there another flag I should use or another way I need to reinstall boost? I have attached both CMakeCache files.

Attempt 1 CMakeCache.txt:
```sh 
# This is the CMakeCache file.
# For build in directory: /home/usern/ompl-1.5.2/build/Release
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Boost filesystem library (debug)
Boost_FILESYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Boost filesystem library (release)
Boost_FILESYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/include

//Boost library directory DEBUG
Boost_LIBRARY_DIR_DEBUG:PATH=/usr/lib/x86_64-linux-gnu

//Boost library directory RELEASE
Boost_LIBRARY_DIR_RELEASE:PATH=/usr/lib/x86_64-linux-gnu

//Boost numpy310 library (debug)
Boost_NUMPY310_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_numpy310.so

//Boost numpy310 library (release)
Boost_NUMPY310_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_numpy310.so

//Boost numpy38 library (debug)
Boost_NUMPY38_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY38_LIBRARY_DEBUG-NOTFOUND

//Boost numpy38 library (release)
Boost_NUMPY38_LIBRARY_RELEASE:FILEPATH=Boost_NUMPY38_LIBRARY_RELEASE-NOTFOUND

//Boost numpy3 library (debug)
Boost_NUMPY3_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY3_LIBRARY_DEBUG-NOTFOUND
[... 812 lines cut ...]
_Boost_COMPONENTS_SEARCHED:INTERNAL=filesystem;numpy;numpy3;numpy310;numpy38;program_options;python;python-py310;python-py38;python3;python310;python38;serialization;system;unit_test_framework
//Last used Boost_INCLUDE_DIR value.
_Boost_INCLUDE_DIR_LAST:INTERNAL=/usr/include
//Last used Boost_LIBRARY_DIR_DEBUG value.
_Boost_LIBRARY_DIR_DEBUG_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_LIBRARY_DIR_RELEASE value.
_Boost_LIBRARY_DIR_RELEASE_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_NAMESPACE value.
_Boost_NAMESPACE_LAST:INTERNAL=boost
//Last used Boost_USE_MULTITHREADED value.
_Boost_USE_MULTITHREADED_LAST:INTERNAL=ON
//CMAKE_INSTALL_PREFIX during last run
_GNUInstallDirs_LAST_CMAKE_INSTALL_PREFIX:INTERNAL=/usr/local
__pkg_config_arguments_ODE:INTERNAL=ode
__pkg_config_arguments_YAMLCPP:INTERNAL=yaml-cpp
__pkg_config_checked_FLANN:INTERNAL=1
__pkg_config_checked_ODE:INTERNAL=1
__pkg_config_checked_SPOT:INTERNAL=1
__pkg_config_checked_YAMLCPP:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_ODE_ode
pkgcfg_lib_ODE_ode-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_YAMLCPP_yaml-cpp
pkgcfg_lib_YAMLCPP_yaml-cpp-ADVANCED:INTERNAL=1
prefix_result:INTERNAL=/usr/lib/x86_64-linux-gnu
```

amberxie88, 2022-11-02:
Attempt 2 CMakeCache.txt:
```sh # This is the CMakeCache file.
# For build in directory: /home/usern/ompl-1.5.2/build/Release
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Boost filesystem library (debug)
Boost_FILESYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Boost filesystem library (release)
Boost_FILESYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/include

//Boost library directory DEBUG
Boost_LIBRARY_DIR_DEBUG:PATH=/usr/lib/x86_64-linux-gnu

//Boost library directory RELEASE
Boost_LIBRARY_DIR_RELEASE:PATH=/usr/lib/x86_64-linux-gnu

//Boost numpy38 library (debug)
Boost_NUMPY38_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY38_LIBRARY_DEBUG-NOTFOUND

//Boost numpy38 library (release)
Boost_NUMPY38_LIBRARY_RELEASE:FILEPATH=Boost_NUMPY38_LIBRARY_RELEASE-NOTFOUND

//Boost numpy3 library (debug)
Boost_NUMPY3_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY3_LIBRARY_DEBUG-NOTFOUND

//Boost numpy3 library (release)
Boost_NUMPY3_LIBRARY_RELEASE:FILEPATH=Boost_NUMPY3_LIBRARY_RELEASE-NOTFOUND

//Boost numpy library (debug)
Boost_NUMPY_LIBRARY_DEBUG:FILEPATH=Boost_NUMPY_LIBRARY_DEBUG-NOTFOUND

//Boost numpy library (release)
Boost_NUMPY_LIBRARY_RELEASE:FILEPATH=Boost_NUMPY_LIBRARY_RELEASE-NOTFOUND

//Boost program_options library (debug)
Boost_PROGRAM_OPTIONS_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost program_options library (release)
Boost_PROGRAM_OPTIONS_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost python-py38 library (debug)
Boost_PYTHON-PY38_LIBRARY_DEBUG:FILEPATH=Boost_PYTHON-PY38_LIBRARY_DEBUG-NOTFOUND

//Boost python-py38 library (release)
Boost_PYTHON-PY38_LIBRARY_RELEASE:FILEPATH=Boost_PYTHON-PY38_LIBRARY_RELEASE-NOTFOUND

//Boost python38 library (debug)
Boost_PYTHON38_LIBRARY_DEBUG:FILEPATH=Boost_PYTHON38_LIBRARY_DEBUG-NOTFOUND

//Boost python38 library (release)
Boost_PYTHON38_LIBRARY_RELEASE:FILEPATH=Boost_PYTHON38_LIBRARY_RELEASE-NOTFOUND

//Boost python3 library (debug)
Boost_PYTHON3_LIBRARY_DEBUG:FILEPATH=Boost_PYTHON3_LIBRARY_DEBUG-NOTFOUND
[... 744 lines cut ...]
//Components requested for this build tree.
_Boost_COMPONENTS_SEARCHED:INTERNAL=filesystem;numpy;numpy3;numpy38;program_options;python;python-py38;python3;python38;serialization;system;unit_test_framework
//Last used Boost_INCLUDE_DIR value.
_Boost_INCLUDE_DIR_LAST:INTERNAL=/usr/include
//Last used Boost_LIBRARY_DIR_DEBUG value.
_Boost_LIBRARY_DIR_DEBUG_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_LIBRARY_DIR_RELEASE value.
_Boost_LIBRARY_DIR_RELEASE_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_NAMESPACE value.
_Boost_NAMESPACE_LAST:INTERNAL=boost
//Last used Boost_USE_MULTITHREADED value.
_Boost_USE_MULTITHREADED_LAST:INTERNAL=ON
//CMAKE_INSTALL_PREFIX during last run
_GNUInstallDirs_LAST_CMAKE_INSTALL_PREFIX:INTERNAL=/usr/local
__pkg_config_arguments_ODE:INTERNAL=ode
__pkg_config_arguments_YAMLCPP:INTERNAL=yaml-cpp
__pkg_config_checked_FLANN:INTERNAL=1
__pkg_config_checked_ODE:INTERNAL=1
__pkg_config_checked_SPOT:INTERNAL=1
__pkg_config_checked_YAMLCPP:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_ODE_ode
pkgcfg_lib_ODE_ode-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_YAMLCPP_yaml-cpp
pkgcfg_lib_YAMLCPP_yaml-cpp-ADVANCED:INTERNAL=1
prefix_result:INTERNAL=/usr/lib/x86_64-linux-gnu```

zkingston, 2022-11-02:
Please continue discussion at https://github.com/ompl/ompl/issues/913

closed 2022-11-02 by zkingston