ompl/ompl #952: OMPL.app create custom environments

gabrielpeixoto-cvai, 2022-11-07:
Hello all,

I started studying OMPL recently (I have been using ROS for over two years). I am trying to create simple 2D and 3D environments like the ones here: 
[image]
[image]
source [MPNet](https://sites.google.com/view/mpnet/home)

The setup is very simple:
- Randomly place squares in SE2 and plan with point robot;
- Randomly place cubes in SE3 and plan with the point robot;
- Create random feasible mazes and plan around them with a point robot.

However, as far as the [demo](https://ompl.kavrakilab.org/group__demos.html) and [OMPL GUI](https://ompl.kavrakilab.org/gui.html) pages go, the only way I can understand is to create a static scene in Collada and import it.

Is there a way to dynamically place primitive forms and add them as collision objects?

OMPL.app seems to be the easiest way to use OMPL for simple environments; if there is a simpler setup, I am open to suggestions.

Thanks in advance!

zkingston, 2022-11-08:
You are correct, OMPL.app does not support building arbitrary scenes. You would have to build your environments and export them as Collada files. There are many integrations that use OMPL (e.g., http://ompl.kavrakilab.org/integration.html). We do not plan to support building environments like this in OMPL.app, but would welcome contributions.

closed 2022-11-08 by zkingston

gabrielpeixoto-cvai, 2022-11-12:
@zkingston Thanks for the quick response. Indeed, I am creating my own environments based on these 2 repositories:

1. https://github.com/monica-ekal/ompl-R2-add-obstacles
2. https://github.com/lyfkyle/pybullet_ompl

I am working with python, so when the environments are ready I will link them in this issue so any other person might find them later.