ompl/ompl #955: Implementation of PrecomputedStateSampler

kkobashi2021, 2022-11-09:
Dear Developers,

I have a question about the implementation of PrecomputedStateSampler in python. In my understanding, the inputs of PreconputedStateSampler are a statespace object and states. I made the statespace object by using ompl.base.RealVectorStateSpace and made the states by using ompl.base.state(statespace). However, I do not know how we concatenate several states into a state object, so currently I use a list that each element is a different state object (ompl.base.state(statespace)). In this case, when I input this statespace and the list of states, I encounter an error when I make an object of PrecomputedStateSampler. Even when I input statespace and a state object, I encounter the same error.  The error says did not much C++ signature. I suspect that the cause of this error is that I input inappropriate types of objects to PrecomputedStateSampler, however, I do not know how to make a states object (not a state object). Would it be possible to tell me how to make a states object? Thank you for your time and consideration.

zkingston, 2022-11-15:
Could you please post a minimum example demonstrating this problem?

kkobashi2021, 2022-11-15:
`states = [ob.State(self.space) for _ in range(1000)]`
`sample_states = np.random.normal(-1, 1, size=(1000, 6)) # This is just an example. I do not use Gaussian samples.`
`for i in range(sample_states.shape[0]):`
   　 `for j in range(sample_states.shape[1]):`
       　　 `states[i][j] = sample_states[i, j]`
`precomputed_sampler = self.allocPrecomputedStateSampler(self.space, states[1]) # The error occurs this line.`
`self.space.setStateSamplerAllocator(ob.StateSamplerAllocator(precomputed_sampler))`

kkobashi2021, 2022-11-15:
This is an example. Even if I use just an ob.State object, I encountered the same error at the same line.
Thank you for responding.

zkingston, 2023-01-19:
It appears that you are giving `states[1]` to the signature, which would be a single state in this case, not a list.

zkingston, 2023-02-15:
Here is an example that uses OMPL's StateStorage class to create a precomputed sampler that works in Python.

```
from ompl import base as ob
from ompl import geometric as og
from ompl import util as ou

import numpy as np

space = ob.RealVectorStateSpace(3)
storage = ob.StateStorage(space)

sample_states = np.random.normal(
    -1, 1,
    size=(1000, 3))

states = []
for i in range(sample_states.shape[0]):
    state = space.allocState()
    for j in range(sample_states.shape[1]):
        state[j] = sample_states[i, j]
    storage.addState(state)
    space.freeState(state)

sampler_allocator = storage.getStateSamplerAllocator()
sampler = sampler_allocator(space)

state = space.allocState()
for i in range(100):
    sampler.sampleUniform(state)
    print(state[0], state[1], state[2])
```

closed 2023-02-15 by zkingston