ompl/ompl #957: Not able to find OMPL app header files

dt1729, 2022-11-15:
Hi all, I am trying to compile this example: [QuadrotorPlanning Cpp](https://ompl.kavrakilab.org/demos_2SE3RigidBodyPlanning_2QuadrotorPlanning_8cpp_source.html) from scratch using my own CMakeLists.txt, however, the compiler is failing to find omplapp folder itself from where the files are to be imported, how should I go about solving this problem, I see that there is an OMPLAPP.so file so there should be a way to include and link the file in the system, could anyone please guide me with this problem? Here is my CMakeLists.txt and CMakeCache.txt:

```
cmake_minimum_required(VERSION 3.10)

project(StepOneMotionPlanning)

set(CMAKE_CXX_FLAGS "-Wall -std=c++11")

set(CMAKE_CXX_FLAGS_RELEASE "-O3")

set(CMAKE_CXX_FLAGS_DEBUG "-g -O0 -std=c++11")

set(CMAKE_BUILD_TYPE Debug)

find_package(ompl REQUIRED)
include_directories(${OMPL_INCLUDE_DIRS})
link_directories(${OMPL_LIBRARY_DIRS} /usr/local/include/ompl-1.5)
add_definitions(${OMPL_DEFINITIONS})

find_package(Boost COMPONENTS system REQUIRED)

add_executable(OMPL_SE3 src/simpleSE3.cpp)
target_link_libraries(OMPL_SE3 ${OMPL_LIBRARIES})
```

```
# This is the CMakeCache file.
# For build in directory: /home/dt/Classes-Fall22/Motion-Planning-Classes/Project/build
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Path to a program.
CMAKE_ADDR2LINE:FILEPATH=/usr/bin/addr2line

//Path to a program.
CMAKE_AR:FILEPATH=/usr/bin/ar

//Choose the type of build, options are: None Debug Release RelWithDebInfo
// MinSizeRel ...
CMAKE_BUILD_TYPE:STRING=

//Enable/Disable color output during build.
CMAKE_COLOR_MAKEFILE:BOOL=ON

//CXX compiler
CMAKE_CXX_COMPILER:FILEPATH=/usr/bin/c++

//A wrapper around 'ar' adding the appropriate '--plugin' option
// for the GCC compiler
CMAKE_CXX_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-9

//A wrapper around 'ranlib' adding the appropriate '--plugin' option
// for the GCC compiler
CMAKE_CXX_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-9

//Flags used by the CXX compiler during all build types.
CMAKE_CXX_FLAGS:STRING=

[... 320 lines cut ...]
//ADVANCED property for variable: CMAKE_SHARED_LINKER_FLAGS_RELWITHDEBINFO
CMAKE_SHARED_LINKER_FLAGS_RELWITHDEBINFO-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_SKIP_INSTALL_RPATH
CMAKE_SKIP_INSTALL_RPATH-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_SKIP_RPATH
CMAKE_SKIP_RPATH-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_STATIC_LINKER_FLAGS
CMAKE_STATIC_LINKER_FLAGS-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_STATIC_LINKER_FLAGS_DEBUG
CMAKE_STATIC_LINKER_FLAGS_DEBUG-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_STATIC_LINKER_FLAGS_MINSIZEREL
CMAKE_STATIC_LINKER_FLAGS_MINSIZEREL-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_STATIC_LINKER_FLAGS_RELEASE
CMAKE_STATIC_LINKER_FLAGS_RELEASE-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_STATIC_LINKER_FLAGS_RELWITHDEBINFO
CMAKE_STATIC_LINKER_FLAGS_RELWITHDEBINFO-ADVANCED:INTERNAL=1
//ADVANCED property for variable: CMAKE_STRIP
CMAKE_STRIP-ADVANCED:INTERNAL=1
//uname command
CMAKE_UNAME:INTERNAL=/usr/bin/uname
//ADVANCED property for variable: CMAKE_VERBOSE_MAKEFILE
CMAKE_VERBOSE_MAKEFILE-ADVANCED:INTERNAL=1
//Details about finding ompl
FIND_PACKAGE_MESSAGE_DETAILS_ompl:INTERNAL=[/opt/ros/noetic/include/ompl-1.5;/usr/include;/usr/include/eigen3][/opt/ros/noetic/lib/x86_64-linux-gnu;/usr/lib/x86_64-linux-gnu][/opt/ros/noetic/lib/x86_64-linux-gnu/libompl.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_system.so][v()]
```

zkingston, 2022-11-15:
It appears that CMake is finding your ROS installed OMPL rather than the locally installed OMPL. Try running CMake without your ROS environment sourced.

dt1729, 2022-11-18:
No, it does not find the files still.

dt1729, 2022-11-18:
Update on this: I tried uninstalling and reinstalling OMPL, however for some reason during installation procedure it mentioned: 
``` -- Set runtime path of "/usr/lib/python3/dist-packages/ompl/app/_app.so" to "/usr/local/lib:/opt/ros/noetic/lib:/opt/ros/noetic/lib/x86_64-linux-gnu"
``` 
Why does it set runtime path to /opt/ros/noetic/lib/x86_64-linux-gnu instead of /usr/lib/x86_64-linux-gnu? 

A temporary workaround is updating the cmake file which searces for .so files, some comments on this are appreciated.

zkingston, 2023-01-18:
I am not quite sure of what your precise issue is, but this appears to be an issue with you still having ROS sourced in your environment while compiling & using OMPL. Try to build and install OMPL without ROS sourced at all - not just uninstalling the ROS binaries.

zkingston, 2023-02-13:
Closing due to age - if you are still interested in this issue let me know.

closed 2023-02-13 by zkingston