ompl/ompl #958: Could not build ompl.Dockerfile on M1 Macbook

patrick-5546, 2022-11-22:
I am trying to build ompl.Dockerfile on my M1 Macbook. Here is what I have done so far.

### Unable to install libspot-dev

#### Error message
```
The following packages have unmet dependencies:
#5 82.89  libspot-dev : Depends: spot (>= 1.99.1a.3-1) but it is not installable
#5 82.89                Depends: spot (< 1.99.1a.3-1.1~) but it is not installable
#5 82.90 E: Unable to correct problems, you have held broken packages.
```

#### Solution
Use an alternative installation method
```
RUN apt-get update && \
    apt-get install -y \
        software-properties-common && \
    add-apt-repository ppa:asiffer/libspot
RUN apt-get update && \
    apt-get install -y \
        build-essential \
        ...
        libspot-dev \
        ...
        wget \
```

### Unable to build

#### Error message
```
#11 1.221 ninja: error: unknown target 'update_bindings'
```

#### What I have tried
- Seems like a similar issue to #725, so I tried the [solution in the comments](https://github.com/ompl/ompl/issues/725#issuecomment-658925712), but it didn't work

#### Ideas
- Perhaps this could be solved by doing something similar to #940?

Any help would be much appreciated, thanks in advance.

patrick-5546, 2022-11-23:
`./install-ompl-ubuntu.sh.in --github --python` ran successfully on my x86 laptop running Ubuntu, which I think narrows down this problem to an ARM issue

patrick-5546, 2022-11-23:
Looks like my alternative method of install `libspot-dev` didn't work, here is a snippet of my build output:
```
-- Checking for module 'libspot'
--   No package 'libspot' found
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS)
```

zkingston, 2023-01-18:
You appear to be installing the wrong `libspot`  - OMPL uses https://spot.lre.epita.fr/, not https://asiffer.github.io/libspot/, which appears to be the PPA you are using. I believe that the alternative installation method just disables `libspot`, which is optional to compile OMPL. Unless you need the LTL planning capabilities, you can remove `libspot`  from the Dockerfile entirely.

patrick-5546, 2023-01-19:
Is https://asiffer.github.io/libspot/ not equivalent to https://spot.lre.epita.fr/? I tried a different PPA because the original does not have ARM packages.

I tried removing libspot from my Dockerfile but I got the following error:
```
 > [ompl-builder 5/5] RUN cmake         -DPYTHON_EXEC=/usr/bin/python3         -DOMPL_REGISTRATION=OFF         -DCMAKE_INSTALL_PREFIX=/usr         -G Ninja         /ompl     && ninja update_bindings -j `nproc`     && ninja -j `nproc`     && ninja install:                                         
#17 0.374 CMake Error: CMake was unable to find a build program corresponding to "Ninja".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.                                                                                                                          
#17 0.374 CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage                                                                                                                                                                                                                                 
#17 0.374 -- Configuring incomplete, errors occurred!                                                                                                                                                                                                                                                   
#17 0.374 See also "/build/CMakeFiles/CMakeOutput.log". 
```
Do I need to do something to remove the libspot dependency from OMPL compilation?

zkingston, 2023-01-19:
Seems now like there is some error with the Ninja build tool - does `ninja` exist for ARM?

patrick-5546, 2023-01-19:
I got this error when compiling on my x86 computer. The only things I changed were removing the libspot dependency and libspot-dev installation:
```
RUN apt-get update \
    && apt-get install -y software-properties-common \
    && add-apt-repository ppa:asiffer/libspot
RUN apt-get update && apt-get install -y libspot-dev
```

patrick-5546, 2023-01-19:
I'm encountering another error when compiling ompl.Dockerfile on my x86 computer, I've opened a new issue: https://github.com/ompl/ompl/issues/974

patrick-5546, 2023-01-19:
> I got this error when compiling on my x86 computer. The only things I changed were removing the libspot dependency and libspot-dev installation

Never mind I deleted another dependency by mistake. I was able to remove the libspot dependency and rebuild without error on my x86 computer.

Do you have any idea on how I can resolve the `unkown target 'update_bindings'` error when compiling OMPL on an ARM computer?

zkingston, 2023-01-19:
I would guess that it is an issue with CMake not finding either pyplusplus, pygccxml, or castxml and is not producing the target. Can you check whether these exist in your build? #725 had issues with castxml

patrick-5546, 2023-01-19:
All I had to do was add the arm ubuntu triplet to `PATH_SUFFIXES` as described in https://github.com/ompl/ompl/issues/725#issuecomment-638421658, thanks for the help!

Should I create a PR for this?

closed 2023-01-20 by zkingston