ompl/ompl #959: No rule to make target 'update_bindings' (ompl 1.5.2, --python)

im-renpei, 2022-11-27:
```
root@ubuntu:/home/pren/Project/Robot_v04# ./install-ompl-ubuntu.sh --python
Reading package lists... Done
Building dependency tree       
Reading state information... Done
lsb-release is already the newest version (11.1.0ubuntu2).
The following packages were automatically installed and are no longer required:
  libavfilter7 libmysofa1 libpostproc55 librubberband2 libvidstab1.1
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Hit:1 https://mirrors.aliyun.com/ubuntu-ports focal InRelease
Hit:2 https://mirrors.aliyun.com/ubuntu-ports focal-updates InRelease                                     
Hit:3 https://mirrors.aliyun.com/ubuntu-ports focal-backports InRelease                                   
Hit:4 https://mirrors.aliyun.com/ubuntu-ports focal-security InRelease                                                                            
Hit:5 http://repo.ros2.org/ubuntu/main focal InRelease                                                                                            
Hit:6 https://repo.download.nvidia.cn/jetson/common r35.1 InRelease              
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  libavfilter7 libmysofa1 libpostproc55 librubberband2 libvidstab1.1
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
g++ is already the newest version (4:9.3.0-1ubuntu2).
libboost-filesystem-dev is already the newest version (1.71.0.0ubuntu2).
libboost-program-options-dev is already the newest version (1.71.0.0ubuntu2).
libboost-system-dev is already the newest version (1.71.0.0ubuntu2).
libyaml-cpp-dev is already the newest version (0.6.2-4ubuntu1).
pkg-config is already the newest version (0.29.1-0ubuntu4).
libboost-serialization-dev is already the newest version (1.71.0.0ubuntu2).
libboost-test-dev is already the newest version (1.71.0.0ubuntu2).
libeigen3-dev is already the newest version (3.3.7-2).
libode-dev is already the newest version (2:0.16-3build1).
cmake is already the newest version (3.16.3-1ubuntu1.20.04.1).
wget is already the newest version (1.20.3-1ubuntu2).
The following packages were automatically installed and are no longer required:
  libavfilter7 libmysofa1 libpostproc55 librubberband2 libvidstab1.1
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-dev is already the newest version (3.8.2-0ubuntu2).
python3-pip is already the newest version (20.0.2-5ubuntu1.6).
The following packages were automatically installed and are no longer required:
  libavfilter7 libmysofa1 libpostproc55 librubberband2 libvidstab1.1
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Using pip 22.3.1 from /usr/local/lib/python3.8/dist-packages/pip (python 3.8)
Looking in indexes: https://mirrors.bfsu.edu.cn/pypi/web/simple/
Collecting https://github.com/CastXML/pygccxml/archive/develop.zip
  Using cached https://github.com/CastXML/pygccxml/archive/develop.zip
  Running command python setup.py egg_info
  running egg_info
  creating /tmp/pip-pip-egg-info-37u8k8cc/pygccxml.egg-info
  writing /tmp/pip-pip-egg-info-37u8k8cc/pygccxml.egg-info/PKG-INFO
  writing dependency_links to /tmp/pip-pip-egg-info-37u8k8cc/pygccxml.egg-info/dependency_links.txt
  writing requirements to /tmp/pip-pip-egg-info-37u8k8cc/pygccxml.egg-info/requires.txt
  writing top-level names to /tmp/pip-pip-egg-info-37u8k8cc/pygccxml.egg-info/top_level.txt
  writing manifest file '/tmp/pip-pip-egg-info-37u8k8cc/pygccxml.egg-info/SOURCES.txt'
  reading manifest file '/tmp/pip-pip-egg-info-37u8k8cc/pygccxml.egg-info/SOURCES.txt'
  adding license file 'LICENSE.rst'
  adding license file 'NOTICE'
  writing manifest file '/tmp/pip-pip-egg-info-37u8k8cc/pygccxml.egg-info/SOURCES.txt'
  Preparing metadata (setup.py) ... done
[... 86 lines cut ...]
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * spot, <http://spot.lrde.epita.fr/>
   Used for constructing finite automata from LTL formulae.
 * Doxygen, <http://doxygen.org/>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org/).
 * Python, <https://www.python.org/>
   Used for python bindings.

-- The following Python modules were found:

 * pyplusplus pygccxml

-- Configuring done
-- Generating done
-- Build files have been written to: /home/pren/Project/Robot_v04/ompl-1.5.2/build/Release
make: *** No rule to make target 'update_bindings'.  Stop.
root@ubuntu:/home/pren/Project/Robot_v04#
```

im-renpei, 2022-12-08:
With the help of cmake-gui, this problem that Python package cannot find in the process of cmake is solved.

closed 2022-12-08 by im-renpei

patrick-5546, 2022-12-08:
How did you use cmake-gui to solve this problem?
- I believe I am facing the same issue in #958