ompl/ompl #967: NearestNeighborsGNATNoThreadSafety With Rebalance is not working

joaquimortizdeharo, 2023-01-13:
Hello, 
It seems there is a bug in NearestNeighborsGNATNoThreadSafety if we use the argument rebalance=True. 
The rebuildSize_ is never increased in practice because the call to rebuildDataStructure will reset rebuildSize to the default value. 

https://github.com/ompl/ompl/blob/e2994e580fcba0eb117e38bd1dc7439c4beb35ef/src/ompl/datastructures/NearestNeighborsGNATNoThreadSafety.h#L446

A possible solution is:
```
                            int previous = gnat.rebuildSize_;
                                    std::cout << "rebuild_size:" << gnat.rebuildSize_ << std::endl;
                            gnat.rebuildDataStructure();
                            gnat.rebuildSize_ = 2 * previous;
```

Thanks!

mamoll, 2023-01-13:
Thanks for reporting the bug and fix. I have created PR #968 to fix this.

zkingston, 2023-01-18:
Closed by #968

closed 2023-01-18 by zkingston