ompl/ompl #974: Error when building ompl.Dockerfile

patrick-5546, 2023-01-19:
### Error
```
#10 440.6 [415/600] Building CXX object py-bindings/CMakeFiles/py_ompl_geometric.dir/bindings/geometric/BFMT.pypp.cpp.o
#10 440.6 In file included from /usr/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
#10 440.6                  from /usr/include/boost/smart_ptr/detail/yield_k.hpp:23,
#10 440.6                  from /usr/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
#10 440.6                  from /usr/include/boost/smart_ptr/detail/spinlock.hpp:42,
#10 440.6                  from /usr/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
#10 440.6                  from /usr/include/boost/smart_ptr/shared_ptr.hpp:29,
#10 440.6                  from /usr/include/boost/shared_ptr.hpp:17,
#10 440.6                  from /usr/include/boost/python/converter/shared_ptr_to_python.hpp:12,
#10 440.6                  from /usr/include/boost/python/converter/arg_to_python.hpp:15,
#10 440.6                  from /usr/include/boost/python/call.hpp:15,
#10 440.6                  from /usr/include/boost/python/object_core.hpp:14,
#10 440.6                  from /usr/include/boost/python/args.hpp:22,
#10 440.6                  from /usr/include/boost/python.hpp:11,
#10 440.6                  from /build/py-bindings/bindings/geometric/BFMT.pypp.cpp:3:
#10 440.6 /usr/include/boost/bind.hpp:36:1: note: '#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.'
#10 440.6    36 | BOOST_PRAGMA_MESSAGE(
#10 440.6       | ^~~~~~~~~~~~~~~~~~~~
#10 440.6 /usr/include/boost/detail/iterator.hpp:13:1: note: '#pragma message: This header is deprecated. Use <iterator> instead.'
#10 440.6    13 | BOOST_HEADER_DEPRECATED("<iterator>")
#10 440.6       | ^~~~~~~~~~~~~~~~~~~~~~~
#10 440.6 ninja: build stopped: subcommand failed.
```

Full output: https://drive.google.com/file/d/1snWjh4c8Y9xTLQVRNwZw5wpf_Bw-xPK-/view?usp=sharing

### Steps to Reproduce
1. `git clone https://github.com/ompl/ompl.git`
2. `cd ompl`
3. `docker build . --file scripts/docker/ompl.Dockerfile --tag ompl`

patrick-5546, 2023-01-19:
Actually I get the same deprecation messages when running install-ompl-ubuntu.sh.in. However, the script runs without error.

zkingston, 2023-01-19:
Yes, deprecation messages are from Boost.Python itself, see https://github.com/boostorg/python/issues/359. I was not able to reproduce the issue - the docker image builds on my machine.

zkingston, 2023-01-19:
Looking at your output, it seems like you might be running out of RAM while compiling, and getting your process killed.

patrick-5546, 2023-01-19:
That was it thanks for the help

closed 2023-01-19 by patrick-5546