ompl/ompl #977: No module named 'ompl'

tarunc20, 2023-01-21:
Hi, I followed the steps in https://ompl.kavrakilab.org/installation.html to install the python bindings - however, when I try to import ompl, it gives me a module not found error. In addition, in the terminal, when I add the py-bindings folder to the pythonpath, it's able to import ompl, but when I try to import something specific from ompl like base, or geometric, it gives me the following error:
`>>> import sys
>>> sys.path.insert(0, '/home/tarunc/ompl-1.6.0/py-bindings')
>>> import ompl
>>> from ompl import base
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tarunc/ompl-1.6.0/py-bindings/ompl/base/__init__.py", line 2, in <module>
    from ompl.base._base import *
ModuleNotFoundError: No module named 'ompl.base._base'`
If you have any suggestions as to how I could fix this, that would be greatly appreciated - thanks!

zkingston, 2023-01-21:
Please post your CMakeCache.txt. Also, verify if `make update_bindings` ran properly (run it again manually yourself) - what has probably happened is that the Python bindings did not actually get generated and built.

tarunc20, 2023-01-21:
Here are the contents of my CMakeCache.txt
`//Boost filesystem library (debug)
Boost_FILESYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Boost filesystem library (release)
Boost_FILESYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/include

//Boost library directory DEBUG
Boost_LIBRARY_DIR_DEBUG:PATH=/usr/lib/x86_64-linux-gnu

//Boost library directory RELEASE
Boost_LIBRARY_DIR_RELEASE:PATH=/usr/lib/x86_64-linux-gnu

//Boost numpy38 library (debug)
Boost_NUMPY38_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_numpy38.so

//Boost numpy38 library (release)
Boost_NUMPY38_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_numpy38.so

//Boost program_options library (debug)
Boost_PROGRAM_OPTIONS_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost program_options library (release)
Boost_PROGRAM_OPTIONS_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_program_options.so

//Boost python-py38 library (debug)
Boost_PYTHON-PY38_LIBRARY_DEBUG:FILEPATH=Boost_PYTHON-PY38_LIBRARY_DEBUG-NOTFOUND

//Boost python-py38 library (release)
Boost_PYTHON-PY38_LIBRARY_RELEASE:FILEPATH=Boost_PYTHON-PY38_LIBRARY_RELEASE-NOTFOUND

//Boost python38 library (debug)
Boost_PYTHON38_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_python38.so

//Boost python38 library (release)
Boost_PYTHON38_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_python38.so

//Boost serialization library (debug)
Boost_SERIALIZATION_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_serialization.so

//Boost serialization library (release)
Boost_SERIALIZATION_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_serialization.so

//Boost system library (debug)
Boost_SYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so

//Boost system library (release)
Boost_SYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so

//Boost unit_test_framework library (debug)
Boost_UNIT_TEST_FRAMEWORK_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so

//Boost unit_test_framework library (release)
Boost_UNIT_TEST_FRAMEWORK_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so

//Path to a program.
CASTXML:FILEPATH=/usr/bin/castxml

//Path to a program.
CMAKE_ADDR2LINE:FILEPATH=/usr/bin/addr2line

//Path to a program.
CMAKE_AR:FILEPATH=/usr/bin/ar

//Type of build
CMAKE_BUILD_TYPE:STRING=Release

[... 706 lines cut ...]
//Last used Boost_LIBRARY_DIR_RELEASE value.
_Boost_LIBRARY_DIR_RELEASE_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_NAMESPACE value.
_Boost_NAMESPACE_LAST:INTERNAL=boost
//Last used Boost_USE_MULTITHREADED value.
_Boost_USE_MULTITHREADED_LAST:INTERNAL=ON
//CMAKE_INSTALL_PREFIX during last run
_GNUInstallDirs_LAST_CMAKE_INSTALL_PREFIX:INTERNAL=/usr/local
__pkg_config_arguments_ODE:INTERNAL=ode
__pkg_config_arguments_YAMLCPP:INTERNAL=yaml-cpp
__pkg_config_checked_FLANN:INTERNAL=1
__pkg_config_checked_ODE:INTERNAL=1
__pkg_config_checked_SPOT:INTERNAL=1
__pkg_config_checked_YAMLCPP:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_ODE_ode
pkgcfg_lib_ODE_ode-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_YAMLCPP_yaml-cpp
pkgcfg_lib_YAMLCPP_yaml-cpp-ADVANCED:INTERNAL=1
prefix_result:INTERNAL=/usr//usr/lib/x86_64-linux-gnu`

and when I ran make update_bindings, it gives the following at the end:
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tarunc/ompl-1.6.0
so I'm not sure whether the bindings have been generated or not.  Moreover, I can see the py-bindings folder in the ompl-1.6.0 directory, but each of the ompl directories within (base, geometric, etc) only contain one __init__.py file.

zkingston, 2023-01-22:
Verify that `make update_bindings` or `ninja update_bindings` complete successfully - you should see the `*.pypp.cpp` objects being compiled when building the entire project afterwards.

tarunc20, 2023-01-22:
After running make update_bindings and then running make, I see the following types of messages 
`Building CXX object py-bindings/CMakeFiles/py_ompl_util.dir/bindings/util/Exception.pypp.cpp.o`
and neither of those commands gives me an error, but the original error is still there.

zkingston, 2023-01-23:
Ah, you are only adding the `./py-bindings` directory to the search path. For local use, I believe you need to add the directory to the built shared objects, which would be in your `./build/lib` directory as well. Append this to the search path as well.

tarunc20, 2023-01-23:
Actually, I just reran make update_bindings and saw a couple things that popped up as tracebacks while running the command - maybe these might be more insightful?
`::ompl::control::ODESolver::getStatePropagator
Traceback (most recent call last):
  File "/home/tarunc/ompl-1.6.0/py-bindings/generate_bindings.py", line 963, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/home/tarunc/ompl-1.6.0/py-bindings/generate_bindings.py", line 461, in __init__
    ['bindings/util', 'bindings/base', 'bindings/geometric'], replacement)
  File "/home/tarunc/ompl-1.6.0/py-bindings/ompl/bindings_generator.py", line 154, in __init__
    self.mb.register_module_dependency(dep)
  File "/usr/local/lib/python3.8/dist-packages/pyplusplus/module_builder/boost_python_builder.py", line 142, in register_module_dependency
    db.load( other_module_generated_code_dir )
  File "/usr/local/lib/python3.8/dist-packages/pyplusplus/utils/__init__.py", line 129, in load
    f = open( fpath, 'r+b' )
FileNotFoundError: [Errno 2] No such file or directory: 'bindings/base'`

`Traceback (most recent call last):
  File "/home/tarunc/ompl-1.6.0/py-bindings/generate_bindings.py", line 963, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/home/tarunc/ompl-1.6.0/py-bindings/generate_bindings.py", line 154, in __init__
    code_generator_t.__init__(self, 'base', ['bindings/util'], replacement)
  File "/home/tarunc/ompl-1.6.0/py-bindings/ompl/bindings_generator.py", line 151, in __init__
    self.filter_declarations()
  File "/home/tarunc/ompl-1.6.0/py-bindings/generate_bindings.py", line 194, in filter_declarations
    self.ompl_ns.class_(f'SpecificParam< {self.string_decl} >').rename('SpecificParamString')
  File "/usr/local/lib/python3.8/dist-packages/pygccxml/declarations/scopedef.py", line 554, in class_
    recursive=recursive)
  File "/usr/local/lib/python3.8/dist-packages/pygccxml/declarations/scopedef.py", line 464, in _find_single
    found = matcher.get_single(decl_matcher, decls, False)
  File "/usr/local/lib/python3.8/dist-packages/pygccxml/declarations/scopedef.py", line 90, in get_single
    raise runtime_errors.declaration_not_found_t(decl_matcher)
pygccxml.declarations.runtime_errors.declaration_not_found_t: Unable to find declaration. Matcher: [(decl type==class_t) and (name==SpecificParam< std::string >)]`

zkingston, 2023-01-23:
Can you try a clean build of the bindings? Also, you should see shared library objects `_base.so`, etc. in the `build/lib` directory if the bindings were built successfully - these are the `ompl.base._base` names being failed to import in your case.

tarunc20, 2023-01-23:
I tried reinstalling the package from the start using https://ompl.kavrakilab.org/core/installation.html and then ran make update_bindings in the build/Release folder, but within build/release/lib, I don't see the _base.so object - only _util.so.

zkingston, 2023-01-23:
Seems like `update_bindings` failed, can you post the full output of that here?

tarunc20, 2023-01-23:
I've attached here the output of the command.
[ub.txt](https://github.com/ompl/ompl/files/10483369/ub.txt)

zkingston, 2023-01-23:
Seems similar to #836. What version of pygccxml, castxml, and pypplusplus are you using?

tarunc20, 2023-01-23:
I'm using 2.2.1, 0.4.5, and 1.8.5, respectively.

zkingston, 2023-01-23:
Could you try updating your castxml to 0.5 if possible? Also, what distro are you using?

tarunc20, 2023-01-23:
From my end, it seems like 0.4.5 is the latest version through pip - also is this distro of ompl?

zkingston, 2023-01-23:
Linux distro in this case. Try installing from their github with `pip install https://github.com/CastXML/CastXML/archive/refs/tags/v0.5.1.zip`

tarunc20, 2023-01-23:
Still working on getting the castxml installation to work, but my linux distro is ubuntu 20.04.5.

tarunc20, 2023-01-24:
I tried using castxml superbuild since I had issues using the regular github, and I was unable to get that to resolve the issue either for some reason. If I were to borrow a folder with the correct build files from someone else, would this be able to fix it? Or is there something system and user specific happening during the build process?

zkingston, 2023-01-25:
Can you try #980 for me? Didn't have time to track down root cause, but with this patch I was able to get this to work in a docker container.

zkingston, 2023-02-13:
Closing given age and merge of #980 / ongoing merge #979

closed 2023-02-13 by zkingston