ompl/ompl #981: cannot build OMPL Python bindings

gongchengli, 2023-02-01:
Hi all, recently I tried to build OMPL Python bindings on Nvidia Jetson Orin but faild. 
I have tried OMPL 1.5.2, OMPL 1.6.0 and the main branch which updated recently, but still get the same error while compiling: 
[image]
I have tried the provied script **install-ompl-ubuntu.sh** and also the Dockerfile, but all ended in that error. 

I tried Ubuntu 20.04 and 22.04 both in docker, and also, 
Castxml 0.5.0, 0.3.6

I notice the similar issues and tried, but still failed and ended with that error. 

thx for reading and hope your reply

zkingston, 2023-02-01:
How much RAM does your platform have? Building the Python bindings can be rather RAM hungry - at least 6GB is recommended.

im-renpei, 2023-02-02:
> How much RAM does your platform have? Building the Python bindings can be rather RAM hungry - at least 6GB is recommended.

Hi Zachary @zkingston , appreciate your timely reply. Our target platform is Nvidia Jetson AGX Orin Developer Kit, the memory is 32 GB 256-bit LPDDR5.

gongchengli, 2023-02-02:
> 

Thx for your reply. 
As for the RAM, I think that's quite enough as shown below, 
[image]

tuanhe, 2023-02-03:
> Hi all, recently I tried to build OMPL Python bindings on Nvidia Jetson Orin but faild. I have tried OMPL 1.5.2, OMPL 1.6.0 and the [main(d9c9566)](https://github.com/ompl/ompl/commit/d9c9566b43e67e6a2eb46888a95613b605ed7bf9) branch which updated recently, but still get the same error while compiling: [image] I have tried the provied script [**install-ompl-ubuntu.sh**](https://github.com/ompl/ompl/blob/main/install-ompl-ubuntu.sh.in) and the [**Dockerfile**](https://github.com/ompl/ompl/blob/main/scripts/docker/ompl.Dockerfile), respectly under Ubuntu 20.04 and 22.04 in docker, and also tried different version of **Castxml** (the latest **0.5.0** and **0.3.6**).
> 
> I notice the similar issues and tried, but still met that error.
> 
> thx for reading and hope your reply

oops, I also met the exactly same issue ! It f**ked me for nearly a month.
Quite strange, that Cannot build on ubuntu Arm platform.
@zkingston , could you help to check it ?
Thanks

zkingston, 2023-02-03:
Can you please post the full output of trying to compile the docker image on your platform? #958 also had issues compiling on ARM.

gongchengli, 2023-02-04:
> Can you post the full output of the build of the docker container? `docker build . -t ompl -f scripts/docker/ompl.Dockerfile` in the top level of the OMPL repository.

Here is the output of building the docker container, 

[log.out.txt](https://github.com/ompl/ompl/files/10607935/log.out.txt)

zkingston, 2023-02-05:
The error appears to be with CastXML on ARM, a number of these errors appear, indicating that it failed to generate output, and thus the rest of the Python binding pipeline failed:
```
In file included from /tmp/tmpapu0i3zy.h:1:
In file included from ./bindings/base.h:1145:
In file included from /usr/include/eigen3/Eigen/Core:214:
/usr/include/eigen3/Eigen/src/Core/arch/NEON/PacketMath.h:3096:10: error: use of undeclared identifier 'vrndn_f32'
{ return vrndn_f32(a); }
```

I'm surprised this error occurred with the latest CastXML - it appears to have been fixed in the past: https://github.com/CastXML/CastXML/issues/223
Can you verify that if you install the latest CastXML in Docker if things work?

gongchengli, 2023-02-06:
> The error appears to be with CastXML on ARM, a number of these errors appear, indicating that it failed to generate output, and thus the rest of the Python binding pipeline failed:
> 
> ```
> In file included from /tmp/tmpapu0i3zy.h:1:
> In file included from ./bindings/base.h:1145:
> In file included from /usr/include/eigen3/Eigen/Core:214:
> /usr/include/eigen3/Eigen/src/Core/arch/NEON/PacketMath.h:3096:10: error: use of undeclared identifier 'vrndn_f32'
> { return vrndn_f32(a); }
> ```
> 
> I'm surprised this error occurred with the latest CastXML - it appears to have been fixed in the past: [CastXML/CastXML#223](https://github.com/CastXML/CastXML/issues/223) Can you verify that if you install the latest CastXML in Docker if things work?

@zkingston I've just verified castxml 0.5.0 from the official  [https://data.kitware.com/#collection/57b5c9e58d777f126827f5a1/folder/63bed6c86d3fc641a02d7e96](url) (the "castxml_v0.5.0-linux-aarch64.tar.gz"), 

and updated the Dockerfile (only comment out the castxml and use COPY command), 
[ompl.Dockerfile.txt](https://github.com/ompl/ompl/files/10644041/ompl.Dockerfile.txt)

After all, the output shows like, 
[log0206.out.txt](https://github.com/ompl/ompl/files/10644717/log0206.out.txt)

zkingston, 2023-02-06:
I believe this is now an issue with pygccxml (https://github.com/CastXML/pygccxml) - please make sure the most recent version is installed in the docker as well. Also take a look at the discussion in https://github.com/ompl/ompl/issues/725 - the same error message appears there.

tuanhe, 2023-02-07:
HI, just followed the issue these days and tried your tips , but all failed! All failed. 
We had absolutely 100% sure that we use the latset pygccxml(it's 2.2.1) and the latest commit of the pygccxml  is May.25.2021
And
#725 is the same issue , but no one can solve it!!!!

gongchengli, 2023-02-07:
> I believe this is now an issue with pygccxml (https://github.com/CastXML/pygccxml) - please make sure the most recent version is installed in the docker as well. Also take a look at the discussion in #725 - the same error message appears there.

@zkingston I am sure that the latest pygccxml is installed in the docker. 
Also, I tried the solutions mentioned in #725 , but unfortunately they doesn't work.

zkingston, 2023-02-07:
I do not have an ARM platform to test on to replicate this, could you insert `traceback.print_exc()` at the bottom of `generate_bindings.py` to see where exactly the error is occuring?

zkingston, 2023-11-28:
Should be fixed by the recently merged #1100 - let me know if that isn't the case.

closed 2023-11-28 by zkingston