ompl/ompl #982: Moveit workspace wouldn't compile after OMPL update

vivekcdavid, 2023-02-02:
Today my Ubuntu Today morning Ubuntu automatically updated my ROS OMPL from 1.5 to version 1.6 and now my moveit workspace wouldn't compile and i am getting this error.

/home/sawyer/ws_moveit/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/model_based_planning_context.h:46:10: fatal error: ompl/base/State.h : No such file or directory
   46 | #include <ompl/base/State.h >

Do you know why it is happening?
Any help is appreciated.

zkingston, 2023-02-02:
Can you clean and build your environment? It might be looking for versioned headers.

vivekcdavid, 2023-02-02:
I tried it multiple times but it was not working, in the end, I had to build a new moveit workspace from scratch and then copy the different packages from the old workspace skipping the packages that were created from the moveit . Now it is compiling.

closed 2023-02-02 by zkingston