ompl/ompl #983: New Way of Checking Python Bindings Incompatible with Python < 3.8

fishbotics, 2023-02-03:
For various (annoying) reasons, I have to build OMPL using Python 3.7. A change introduced a month ago breaks the dependency checking when building the Python bindings in 3.7. Here is the change:
https://github.com/ompl/ompl/commit/30f3a5bb90587c133973c42c2b6f26f252e7baea#diff-812c4ae87316942a30217e350bf50f289a827f09b952507df19832d0a9556a1bR128

Note the use of `importlib.metadata`. In Python 3.7, this is an elective package called `importlib_metadata` which must be installed manually. The install script should probably be modified to install this package if necessary and the FindPython.cmake file could use a try/except one-liner to try both packages in order to work with both Python versions. 

I'm happy to make these changes if they seem reasonable. As far as I can tell, the current release (1.6.0) is broken with Python 3.7.

zkingston, 2023-02-03:
Yes, this is a reasonable change - thanks for reporting this issue. Would installing `importlib_metadata` affect more recent Python versions? If not, we can just add it to the install script and dependencies.

fishbotics, 2023-02-03:
Unfortunately, I don't believe that package is available for more recent Python versions. There probably needs to be a version check before installing. My understanding is that they moved that package into the built-in package `importlib` starting in Python 3.8 (according to the docs here: https://docs.python.org/3/library/importlib.metadata.html#module-importlib.metadata)

zkingston, 2023-02-16:
Take a look at #992 and see if that resolves the issue.

PR #992 (2023-02-16): Fix Python versions <= 3.7 finding module path
Fixes #983
closed 2023-02-16 by zkingston