ompl/ompl #984: No rule for update_bindings

fishbotics, 2023-02-03:
Hi,

I am trying to install OMPL version 1.6.0 and when I try to build the Python bindings, I get an error saying there is no rule to update bindings. As far as I can tell, I have all the necessary dependencies installed. 

Thanks for your help!

Here is the output of my CMake.

```
-- The CXX compiler identification is GNU 9.4.0                                                                                                                                                                                                        
-- Check for working CXX compiler: /usr/bin/g++                                                                                                                                                                                                        
-- Check for working CXX compiler: /usr/bin/g++ -- works                                                                                                                                                                                               
-- Detecting CXX compiler ABI info                                                                                                                                                                                                                     
-- Detecting CXX compiler ABI info - done                                                                                                                                                                                                              
-- Detecting CXX compile features                                                                                                                                                                                                                      
-- Detecting CXX compile features - done                                                                                                                                                                                                               
-- Build type: Release                                                                                                                                                                                                                                 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")                                                                                                                                                                                       
-- Checking for module 'ode'                                                                                                                                                                                                                           
--   Found ode, version 0.16                                                                                                                                                                                                                           
-- Checking for module 'flann'                                                                                                                                                                                                                         
--   No package 'flann' found                                                                                                                                                                                                                          
-- Checking for module 'libspot'                                                                                                                                                                                                                       
--   No package 'libspot' found                                                                                                                                                                                                                        
-- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS)                                                                                                                                                                                     
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)                                                                                                                                                                                                
-- Found PY_numpy: /usr/local/lib/python3.7/dist-packages/numpy                                                                                                                                                                                        
-- Found PY_pyplusplus: /usr/local/lib/python3.7/dist-packages/pyplusplus                                                                                                                                                                              
-- Found PY_pygccxml: /usr/local/lib/python3.7/dist-packages/pygccxml                                                                                                                                                                                  
-- Found castxml: /usr/bin/castxml                                                                                                                                                                                                                     
-- Checking for module 'yaml-cpp'
--   Found yaml-cpp, version 0.6.2
-- Found yaml-cpp: yaml-cpp
-- The following features have been enabled:

 * OMPL_VERSIONED_INSTALL, Whether to install header files in
   <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
   version numbers
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
 * OMPL_BUILD_TESTS, Whether to build the unit tests.
 * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.

-- The following OPTIONAL packages have been found:

 * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
   Pthreads is sometimes needed, depending on OS / compiler.
 * pypy, <https://pypy.org>
   Used to speed up the generation of python bindings.
 * ODE, <http://ode.org>
   The Open Dynamics Engine can be used as a "black box" for kinodynamic planning.
 * castxml, <https://github.com/CastXML/CastXML>
   Used to generate Python bindings.
 * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
   Used to find (compilation flags for) dependencies.
 * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
   Used for parsing input data for the PlanarManipulator demo.
 * Python, <https://www.python.org>
   Used for python bindings.

-- The following REQUIRED packages have been found:

 * Eigen3, <http://eigen.tuxfamily.org>
   A linear algebra library used throughout OMPL.
 * Boost (required version >= 1.58), <https://www.boost.org>
   Used throughout OMPL for data serialization, graphs, etc.

-- The following OPTIONAL packages have not been found:

 * MORSE, <https://www.openrobots.org/wiki/morse>
   OMPL includes a plugin for the MORSE Robot Simulation engine.
 * Triangle, <http://www.cs.cmu.edu/~quake/triangle.html>
   Used to create triangular decompositions of polygonal 2D environments.
 * flann (required version >= 1.8.3), <https://github.com/mariusmuja/flann>
   If detetected, FLANN can be used for nearest neighbor queries by OMPL.
 * spot, <http://spot.lrde.epita.fr>
   Used for constructing finite automata from LTL formulae.
 * Doxygen, <http://doxygen.org>
   Used to create the OMPL documentation (i.e., https://ompl.kavrakilab.org).

-- The following Python modules were found:

 * numpy pyplusplus pygccxml
```

zkingston, 2023-02-03:
To be clear, this is after installation of `importlib_metadata` as mentioned in #983?

Could you post your CMakeCache.txt from the build directory?

fishbotics, 2023-02-03:
Thanks @zkingston  for your help. Yes it was after. Before I made that change, these two lines (copied from the CMake output above) said not found:
```
-- Found PY_pyplusplus: /usr/local/lib/python3.7/dist-packages/pyplusplus                                                                                                                                                                              
-- Found PY_pygccxml: /usr/local/lib/python3.7/dist-packages/pygccxml        
```
and these lines said that it was not going to build the bindings:

```
 * OMPL_BUILD_PYBINDINGS, Whether to build the Python bindings.
 * OMPL_BUILD_PYTESTS, Whether to build/run the python binding unit tests.
```

Here's the CMakeCache file: 
[CMakeCache.txt](https://github.com/ompl/ompl/files/10581261/CMakeCache.txt)

zkingston, 2023-02-03:
It seems like CMake was unable to find Boost Python/numpy for Python 3.7 - what version of boost do you have installed?

fishbotics, 2023-02-03:
Ah I see that now. I did install it, but CMake can't seem to find it...

```
root@bionic-fish:~/ompl-1.6.0/build/Release# apt install libboost-python-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libboost-python-dev is already the newest version (1.71.0.0ubuntu2).
0 upgraded, 0 newly installed, 0 to remove and 56 not upgraded.
```

zkingston, 2023-02-03:
I believe that 1.71 is going to have the `.so` for Python 3.8, double check that with `ls /usr/lib/x86_64-linux-gnu/libboost_python*` - this is going to be the issue most likely. I think you might need to compile boost with your specific Python support in mind.

fishbotics, 2023-02-03:
Gah, you're right. It must be because this is from the Ubuntu 20.04 aptitude, which means that it's only built for the system Python. I will go manually compile Boost now 😭

fishbotics, 2023-02-03:
@zkingston, if I have Boost Python living somewhere else (it's coincidentally already installed in my Docker), how would I point the build to it? 

For example, if I have it here:
```
/isaac-sim/kit/extscore/omni.usd.libs/bin/libboost_python37.so.1.68.0
/isaac-sim/kit/extscore/omni.usd.libs/bin/libboost_python37.so
```

zkingston, 2023-02-03:
Try the advice in [this post](https://scaron.info/blog/cmake-can-use-a-local-boost.html) - I haven't tried this myself.

fishbotics, 2023-02-03:
I managed to make everything work by building a new binary from Boost. The solution is hacky, but I'll post my process here for posterity.

I had to clone the Boost Python repo, pip install `faber`, simlink `/usr/bin/python3.7` to `/usr/bin/python` (otherwise `faber` breaks), built the Boost Python libraries with `faber`, and then putting the shared objects into the directory that @zkingston mentioned above. Perhaps there's a nicer way to do the system installation with `faber`. Anyone struggling with this in the future can hopefully find that here: https://github.com/boostorg/python/issues/411

After putting the shared objects in the right place, I blew out the build directory, reran CMake, and the `update_bindings` command worked.

zkingston, 2023-02-03:
Great! Thanks for reporting how to reproduce this.

closed 2023-02-03 by zkingston

amberxie88, 2023-02-03:
I installed Boost somewhat like this: https://stackoverflow.com/questions/12578499/how-to-install-boost-on-ubuntu/24086375#24086375

My issue was how to link Boost in the cmake process. I had to manually remove the apt installed Boost (though I'm not sure if that was necessary) and added `include_directories(/path/to/boost_1_81_0)` and `link_directories(/path/to/boost_1_81_0/stage/lib`. I also updated `find_package Boost 1.81 QUIET COMPONENTS ...)` to search for the specific version of boost. This seemed to have fixed it for me, though it could have also been other things at the time.

For whatever reason, `make update_bindings` was not building the .so files in `build/Release/lib`, so I tried `make` and it installed the Python bindings. Strange but it worked.

zkingston, 2023-02-03:
`make update_bindings` generates the code for the Python bindings, but `make` is still required to build the shared objects.

ccm010101, 2024-11-13:
> castxml

I already installed castxml as
```
Warning: castxml 0.6.8_2 is already installed and up-to-date.
To reinstall 0.6.8_2, run:
  brew reinstall castxml
```

But I still see

```
(ompl) Doonams-MacBook-Pro:ompl chanmichoi$ make -j 8 update_bindings
make: *** No rule to make target `update_bindings'.  Stop.
```

ccm010101, 2024-11-13:
> Hi @zkingston
> 
> I met the same error when I built python binding for conda env in my new Ubuntu 20.04 system.
> 
> I referenced [this issue](https://github.com/ompl/ompl/issues/1059#issue-1753041441) and added the cmake option:
> 
> ```
> cmake ../.. -DCMAKE_INSTALL_PREFIX="$CONDA_PREFIX" -DBOOST_ROOT="$CONDA_PREFIX" -DPYTHON_EXEC=/home/xzc/anaconda3/envs/ompl_env/bin/python3.7 -DPYTHONPATH=/home/xzc/anaconda3/envs/ompl_env/lib/python3.7/site-packages
> ```
> 
> Here is the cmake output:
> 
> ```
> -- The CXX compiler identification is GNU 9.4.0
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Build type: Release
> -- Found Boost: /home/xzc/anaconda3/envs/ompl_env/lib/cmake/Boost-1.73.0/BoostConfig.cmake (found suitable version "1.73.0", minimum required is "1.58") found components: serialization filesystem system program_options 
> -- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
> -- Checking for module 'flann'
> --   Found flann, version 1.9.1
> -- Checking for module 'libspot'
> --   No package 'libspot' found
> -- Could NOT find spot (missing: SPOT_LIBRARIES SPOT_INCLUDE_DIRS) 
> -- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
> -- Found PY_numpy: /home/xzc/anaconda3/envs/ompl_env/lib/python3.7/site-packages/numpy  
> -- Found PY_pyplusplus: /home/xzc/anaconda3/envs/ompl_env/lib/python3.7/site-packages/pyplusplus
> -- Found PY_pygccxml: /home/xzc/anaconda3/envs/ompl_env/lib/python3.7/site-packages/pygccxml  
> -- Could NOT find castxml (missing: CASTXML) 
> -- Checking for module 'yaml-cpp'
> --   Found yaml-cpp, version 0.6.2
> -- Found yaml-cpp: yaml-cpp  
> -- The following features have been enabled:
> 
>  * OMPL_VERSIONED_INSTALL, Whether to install header files in
>    <prefix>/include/ompl-X.Y/ompl, where X and Y are the major and minor
>    version numbers
>  * OMPL_BUILD_TESTS, Whether to build the unit tests.
>  * OMPL_BUILD_DEMOS, Whether to build the OMPL demo programs.
> 
> -- The following OPTIONAL packages have been found:
> 
>  * Threads, <https://en.wikipedia.org/wiki/POSIX_Threads>
>    Pthreads is sometimes needed, depending on OS / compiler.
>  * boost_numpy (required version == 1.73.0)
>  * Python, <https://www.python.org>
>    Used for python bindings.
>  * boost_python (required version == 1.73.0)
> * boost_unit_test_framework (required version == 1.73.0)
>  * PkgConfig, <https://www.freedesktop.org/wiki/Software/pkg-config/>
>    Used to find (compilation flags for) dependencies.
>  * yaml-cpp, <https://github.com/jbeder/yaml-cpp>
>    Used for parsing input data for the PlanarManipulator demo.
> 
> -- The following REQUIRED packages have been found:
> 
>  * boost_serialization (required version == 1.73.0)
>  * boost_filesystem (required version == 1.73.0)
>  * boost_system (required version == 1.73.0)
>  * boost_program_options (required version == 1.73.0)
>  * Eigen3, <http://eigen.tuxfamily.org>
>    A linear algebra library used throughout OMPL.
>  * boost_headers (required version == 1.73.0)
>  * Boost (required version >= 1.58), <https://www.boost.org>
>    Used throughout OMPL for data serialization, graphs, etc.
> 
[... 62 lines cut ...]
My
`make update_bindings`
ran well as 
```
...
-- The following Python modules were found:

 * numpy pyplusplus pygccxml

-- Configuring done (0.7s)
-- Generating done (0.3s)
-- Build files have been written to: /Users/chanmichoi/ompl
[100%] Built target update_tools_bindings
[100%] Built target update_bindings

```
However, still I see 
```
    from ompl import base as ob
ImportError: cannot import name 'base' from 'ompl' (unknown location)
```
when I try to import base from ompl.

`import ompl`
ran fine.