ompl/ompl #986: update_bindings failed

tuanhe, 2023-02-07:
I met the issue  on ubuntu 20.04 and python3.8 
And met the issue:
aceback (most recent call last):
  File "/root/ompl/py-bindings/generate_bindings.py", line 967, in <module>
    globals()['ompl_'+module+'_generator_t']()
  File "/root/ompl/py-bindings/generate_bindings.py", line 626, in __init__
    replacement)
  File "/root/ompl/py-bindings/ompl/bindings_generator.py", line 142, in __init__
    self.std_ns = self.mb.namespace('std')
  File "/usr/local/lib/python3.8/dist-packages/pyplusplus/module_builder/module_builder.py", line 267, in namespace
    , recursive=recursive )
  File "/usr/local/lib/python3.8/dist-packages/pygccxml-2.2.1-py3.8.egg/pygccxml/declarations/namespace.py", line 121, in namespace
    recursive=recursive)
  File "/usr/local/lib/python3.8/dist-packages/pygccxml-2.2.1-py3.8.egg/pygccxml/declarations/scopedef.py", line 464, in _find_single
    found = matcher.get_single(decl_matcher, decls, False)
  File "/usr/local/lib/python3.8/dist-packages/pygccxml-2.2.1-py3.8.egg/pygccxml/declarations/scopedef.py", line 92, in get_single
    raise runtime_errors.multiple_declarations_found_t(decl_matcher)
pygccxml.declarations.runtime_errors.multiple_declarations_found_t: Multiple declarations have been found. Matcher: [(decl type==namespace_t) and (name==std)]

Is there anyone that can help me to solve it ?
Words fail me to express the heartfelt gratitude

tuanhe, 2023-02-07:
Some updated:
pip3 list
Package    Version
---------- -------
numpy      1.17.4 
pip        20.0.2 
pygccxml   2.2.1  
pyplusplus 1.8.5  

 $castxml --version
castxml version 0.5.0

CastXML project maintained and supported by Kitware (kitware.com).

clang version 13.0.0
Target: aarch64-unknown-linux-gnu
Thread model: posix

zkingston, 2023-02-07:
This is the same issue as reported in #981, please keep discussion there.

closed 2023-02-07 by zkingston